PowerBroker2/ELMduino #158: Issue with connecting to ELM327

hawkeye1103, 2023-03-03:
Hello,

I have 3 elm327 modules with the same issue. I'm guessing i'm doing something wrong and I need your help to find out what. My code is compiling correctly.
Two of the modules I have bought recently because I suspected my old module is not working ok. So I tested them with bluetooth terminal from my phone and I got the following result from them. 
[image]
[image]

I'm guessing the one that doesn't have the J is the one that looks ok. 

I also added my code below maybe you can spot an issue with it .... already tried on the first elm327 to use the code you provided and I was not able to connect to it either way, also tried a different esp32 module same issue. Now I'm fresh out of ideas and don't know how to solve it. Maybe you guys can shed some new light on the problem. 

The code is a from a shiftlight module called Chippernut and was originally designed for an Arduino Nano and I made it work for ESP32. If you need further clarification I'm here to try anything:). 

Thanks in advance for all your help.

` /* 
   ________    _                                   __ 
  / ____/ /_  (_)___  ____  ___  _________  __  __/ /_
 / /   / __ \/ / __ \/ __ \/ _ \/ ___/ __ \/ / / / __/
/ /___/ / / / / /_/ / /_/ /  __/ /  / / / / /_/ / /_  
\____/_/ /_/_/ .___/ .___/\___/_/  /_/ /_/\__,_/\__/  
            /_/   /_/         
Visit us: www.chippernut.com 
ARDUINO RPM TACHOMETER MULTI-DISPLAY 
Written by Jonduino (Chippernut)
03-20-2016
Updated 08-09-2017
********************* Version NOTES ******************** 
/* v1.0 BETA 11/17/2013 -- Initial Release 
** v1.1 03/09/2014 - 
** Fixed bug with flasher that didn't correspond to brightness value 
** Improved sleep function, now it shuts off after 5-seconds of engine OFF 
** Other minor improvements. 
** v2.1 08/10/2015 -- Too many to list here. 
** v2.2 01/17/2016 -- 
** New Display Support
** v2.4 03/20/2016 --
** Added dimmer support 
** Fixed array mathematics (bug)
** Improved rotary encoder response
** Finer PPr Control (0.1)
** v3.0 08/08/2017
* New display support
* New LEDS (APA102c)
* New Animations
* General code improvements and optimization
*/ 

// Include these libraries 
#include <Wire.h>
#include <Adafruit_NeoPixel.h> 
#include <EEPROM.h> 
#include <EEPROMAnything.h> 
#include "SSD1306Ascii.h"
#include "SSD1306AsciiWire.h"
#include "BluetoothSerial.h"
#include "ELMduino.h"

//Uncomment for ELM327 connection >>>

#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif
BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
ELM327 myELM327;
String MACadd = "11:22:33:AA:BB:CC";
uint8_t address[6] = {0x11, 0x22, 0x33, 0xAA, 0xBB, 0xCC};; //MAC Address
//uint32_t rpm;

[... 1850 lines cut ...]
  
          if (flashbool == true)
           for(int i=0; i<NUMPIXELS; i++) { 
              strip.setPixelColor(i, flclr1); 
           }
          else
           for(int i=0; i<NUMPIXELS; i++) { 
               strip.setPixelColor(i, strip.Color(0, 0, 0)); 
            }      
        }  
  }

  strip.show(); 

  
    //Poll the Button, if pushed, cue animation and enter menu subroutine 
    if (digitalRead(button_pin) == LOW){ 
      delay(250); 
      strip.clear(); 
      oled.clear();
      entermenu();  
      menuvar=1; 
      menu(); 
    } 
} `

PowerBroker2, 2023-03-03:
`"Couldn't connect to OBD scanner - Phase 1"` means your Arduino can't pair/connect to the ELM327. You need to make sure you use the correct Bluetooth device name in your sketch that corresponds to your ELM327. You also need to make sure the ELM327 isn't connected to another device (i.e. your phone).

**Also, I am NOT taking a look at your 1 million lines of unformatted code.**

hawkeye1103, 2023-03-04:
Already done all of that including deleting the history of connected devices of my esp32. It's a esp32, not arduino.

hawkeye1103, 2023-03-04:
Fixed by connecting via MAC Address although it doesn't have the standard AA:BB:CC:11:22:33 mac address it seems to work... it was bench tested for now, I'm going to try it on the car later. Tested very quickly and easily with the BluetoothSerial example: SerialtoserialBTM

closed 2023-03-04 by PowerBroker2