PowerBroker2/ELMduino #160: Write value to PCM

maciekelga, 2023-03-26:
Hi,

Is there any option to write a value or modify in some how? I have one non standard PID which I can read but I want to modify from time to time. For example if is more than 5 i want to write 4 etc.

Is there any option like this?

PowerBroker2, 2023-03-26:
I'm unaware of PIDs that you can "write" to. There are certain configuration commands, but I don't think that's what you're asking about. You might want to ask on car forums for how to use specific PIDs if that's what you're looking for.

closed 2023-03-26 by PowerBroker2

patfelst, 2023-03-26:
As powerBroker2 said, and as far as I'm aware, you can only read from PIDs. Please check https://en.wikipedia.org/wiki/OBD-II_PIDs. 

Also, check the original ELM327 chip user manual from page 33 "Talking to the vehicle" https://www.elmelectronics.com/wp-content/uploads/2020/05/ELM327DSL.pdf, you might find what you're looking for there.