PowerBroker2/ELMduino #162: Vin Number Question

adolfie450, 2023-03-29:
thanks for great library, its helping a lot.

I have a ESp32 Wroom using Bluetooth Serial. 

I have all the PIds in the example Working, Except for VIN Number. I see its blocking function and you must pass a Char array to it. 

however it gives me Memory dump when it is executed. 

Can someone please show me an example code snippet on how to Retrieve it.

thank you

G

patfelst, 2023-03-29:
Yes I wrote that function. I’ll post an example calling it later today.

patfelst, 2023-03-29:
Will still post an example later, but in the mean time you could try this. The default ELM327 object payload length is 40 bytes as per this line (payloadLen = 40)
https://github.com/PowerBroker2/ELMduino/blob/eabbb1a2e8e82d8fed1b758c0d893bc46efbe171/src/ELMduino.h#L310

As the VIN payload is so long (49 character for my car), you'll need to increase this to at least 50. To be safe for your car, increase it to 60. This can be done when creating the ELMDuino object, e.g. 
`myELM327.begin(SerialBT, false, 1000, '0', 60);`

patfelst, 2023-03-29:
well I looked in my local code repositories, but it seems I didn't keep any code that queries the vin. I'll have to do this on the weekend to make sure it actually works. Should be as simple ad creating a character array of 20 bytes (the VIN is 18 characters) that you pass into the function. e.g. (from memory):
```C++
char vin_string[20] = "";
myELM327.get_vin_blocking(char vin_string);
printf("Vehicle ID number is: %s\n", vin_string);
```

patfelst, 2023-04-01:
Here's some how to read vehicle VIN from the `setup()` function. 

```C++
#include <ELMduino.h>
#include "BluetoothSerial.h"

ELM327 myELM327;
BluetoothSerial SerialBT;

char bluetooth_mac_str[] = "00:1d:a5:22:5f:f6";  // VGate iCar Pro black and gold
uint8_t bt_mac[6] = {0x00};
char txt[80] = "";
char car_vin[18] = {0}; // string storage for VIN
bool bt_connected = false;

void setup() {

  // Start ESP32 bluetooth
  SerialBT.begin("ELM_NonBlock", true);

  // Convert bluetooth MAC address string into integer array[6]
  sscanf(bluetooth_mac_str, "%02hhx:%02hhx:%02hhx:%02hhx:%02hhx:%02hhx",
         bt_mac, bt_mac + 1, bt_mac + 2,
         bt_mac + 3, bt_mac + 4, bt_mac + 5);
  sprintf(txt, "%s = %s", "MAC address", bluetooth_mac_str);
  Serial.println(txt);

  // Attempt to connect ESP32 to bluetooth OBDII scanner
  bt_connected = SerialBT.connect(bt_mac);

  if (!bt_connected) {
    sprintf(txt, "%s", "Can't connect OBD - Phase 1");
    Serial.println(txt);
    while (1) {}
  }

  sprintf(txt, "%s", "Phase 1 ok");
  Serial.println(txt);

  sprintf(txt, "%s", "Trying to start ELM327");
  Serial.println(txt);

  // Initialise ELM327 with larger receive buffer "Payload" of 60 characters to enable read of VIN
  bt_connected = myELM327.begin(SerialBT, true, 1000, '0', 60);

  if (!bt_connected) {
    sprintf(txt, "%s", "Can't connect OBD - Phase 2");
    Serial.println(txt);
    while (1) {}
  }

  sprintf(txt, "%s", "Phase 2 ok");
  Serial.println(txt);

  // Get VIN
  myELM327.get_vin_blocking(car_vin);
  if (strlen(car_vin) > 0) {
    Serial.printf("VIN: %s\n", car_vin);
  } else {
    Serial.println("VIN not avaialable");
  }

}
```

patfelst, 2023-04-20:
Hey @adolfie450 did you ever get this to work?

PowerBroker2, 2023-12-21:
Stale issue

closed 2023-12-21 by PowerBroker2