PowerBroker2/ELMduino #163: Problem getting data from the ELM327

miguelgarcia42, 2023-03-30:
Hey guys , im getting always the same error using ELMduino library , right now im using OBDsim to simulate the elm327 on my PC (i get the exact same response when using it in my car) .
I can make it work with the softwareSerial library but im having trouble converting the rpm value from hexadecimal to decimal , thats why im trying ELMduino.

The code im using:

```
#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial mySerial(8, 9); // RX, TX
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(115200);
  myELM327.begin(mySerial, true, 2000);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

The serial monitor loops this and the rpm is always 884:
[image]

This is the OBDSim log , it is getting the commands to reset and the queries.

[image]

Hope someone can help me fix this , thanks :)

PowerBroker2, 2023-03-30:
If the query is `010C1`, then the response header should be `410C1`. However, it seems the sim is sending a response header of  `410C0` for some reason. This means ELMduino will not recognize the header and eventually time out.

miguelgarcia42, 2023-03-30:
> If the query is `010C1`, then the response header should be `410C1`. However, it seems the sim is sending a response header of `410C0` for some reason. This means ELMduino will not recognize the header and eventually time out.

Any sugestion on how i can fix it? I get the same error with the elm327 plugged to my car

PowerBroker2, 2023-03-30:
Seems like both your car and sim are screwed up, not sure how to fix that. Perhaps you could jerry-rig the lib src files to work around the issue.

What is your car's make/model/year?

miguelgarcia42, 2023-03-30:
> Seems like both your car and sim are screwed up, not sure how to fix that. Perhaps you could jerry-rig the lib src files to work around the issue.
> 
> What is your car's make/model/year?

Oh ok :/ i will try to find a way to fix it , it is a 2017 honda civic hatchback

PowerBroker2, 2023-03-30:
Interesting, I was expecting a pre-2008 car since it doesn't seem to be adhering to the OBD2 standard

closed 2023-03-30 by PowerBroker2

vladislav-n-dario, 2023-05-24:
Hi @PowerBroker2 ! Currently having timeout stuff with my Civic 8th gen Hatchback 2007, can you please give more info on those issues you've mentioned with pre-2008 Civics?