PowerBroker2/ELMduino #165: no work with teensy 4.0

cruz177, 2023-04-10:
Good morning

I'm trying to get an elm327 1.5v to work with your code on a teensy 4.0 and I can't get the code to work.

I have made sure that elm327 was real (4hz oscillator, original integrated,...) and I have made it work by bt with the android mobile

I'm sure I have everything connected correctly on RX and TX to TX and RX of my teensy on serial1 (pin 0,1)

The code runs and it seems to return data, but all the time I get the error message error: elm_timeout and it seems that if it connects to teensy, but it does not open the obd2 port (or so I think).

Could it be a problem of having a 16-bit AVR processor?

Thank you

PowerBroker2, 2023-04-10:
I use the latest release of ELMduino on my personal car HUD that has a Teensy 4.1 and have had pretty much 0 issues with it.

Can you post the debug printouts?

cruz177, 2023-04-21:
Thanks for the answer, I've been trying to do my research but I'm the same.

This continuously appears on the serial monitor, I begin to think that it could be the same elm327?

imeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT Z
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT E0
	Received char: A
	Received char: T
	Received char: _
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT S0
	Received char: A
	Received char: T
	Received char: _
	Received char: S
	Received char: 0
	Received char: \r
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT AL
	Received char: A
	Received char: T
	Received char: _
	Received char: A
	Received char: L
	Received char: \r
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT ST 00
	Received char: A
	Received char: T
	Received char: _
	Received char: S
	Received char: T
	Received char: _
	Received char: 0
	Received char: 0
	Received char: \r
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT TP A0
	Received char: A
	Received char: T
	Received char: _
	Received char: T
	Received char: P
	Received char: _
	Received char: A
	Received char: 0
	Received char: \r
Timeout detected with overflow of 0ms
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
	Received char: A
	Received char: T
	Received char: _
	Received char: S
	Received char: P
	Received char: _
	Received char: 0
	Received char: \r
Timeout detected with overflow of 0ms
Setting protocol via AT SP %c did not work
Couldn't connect to OBD scanner
Connected to ELM327
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: \r
Timeout detected with overflow of 0ms
Received: 010C1
ERROR: ELM_TIMEOUT
Service: 1
PID: 13
Normal length query detected
Query string: 010D1
Clearing input serial buffer
Sending the following command/query: 010D1
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: D
	Received char: 1
	Received char: \r
Timeout detected with overflow of 0ms
Received: 010D1
ERROR: ELM_TIMEOUT
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: \r
Timeout detected with overflow of 0ms
Received: 010C1
ERROR: ELM_TIMEOUT
Service: 1
PID: 13
Normal length query detected
Query string: 010D1
Clearing input serial buffer
Sending the following command/query: 010D1
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: D
	Received char: 1
	Received char: \r
Timeout detected with overflow of 0ms
Received: 010D1
ERROR: ELM_TIMEOUT
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
	Received char: 0
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 1
	Received char: \r

Could you tell me where to buy one that works 100% to discard?

Thank you

PowerBroker2, 2023-05-04:
Sorry, didn't notice your previous post. It seems your OBD2 scanner isn't replying with the correct responses. You'll probably need to get a better/different one.

jimwhitelaw, 2023-12-12:
I don't think this is related to the teensy board, it looks like another instance of this issue:
https://github.com/PowerBroker2/ELMduino/issues/146#issuecomment-1851132513

**TL;DR** - try the latest version of ELMduino.

closed 2023-12-21 by PowerBroker2