PowerBroker2/ELMduino #166: NODATA for fuelRate() and mafRate()

nvdias0, 2023-04-14:
**Describe the bug**
NODATA for fuelRate() and mafRate()

**To Reproduce**
All major pids (rpm, speed, voltage, throttle) reported ok.
Tested on Opel Corsa C 2003, Renault Clio 3 2008, Mercedes A180 2020, Ford Fusion 2007

**Expected behavior**
Getting some values for those PIDS

**Equipment**
Arduino Uno R3, 2 lines LCD, ELM327 obd scanner cable connected directly to Arduino Data lines.

**Code**
Should not be a code issue as the pids referred above are reporting ok.

**Wiring**
Should not be a code issue as the pids referred above are reporting ok.

PowerBroker2, 2023-04-14:
`NO DATA` means the PIDs you're requesting are unsupported and not an issue with the library. This makes sense considering all of your cars are 2008 or earlier (I can only guarantee this lib works on cars past 2008)

enslinm123, 2023-04-17:
@nvdias0 if you compile a sketch to query each PID separately, so a sketch per PID, do you receive responses or still get the ```NO DATA``` response?

My vehicles are all 2007 and I get the ```NO DATA``` response when I query those PID's separately so it makes sense what @PowerBroker2 said, the lib fully supports higher then 2008 vehicles, however, the torque pro app can collect and display that data somehow.

nvdias0, 2023-04-17:
Thank you for your suggestions,
In the 4 cars i've tested (see above) I do not receive information regarding a standard pid as the MAF Rate.
I've tried just reading this PID, and also tried the multi pid algorithm (as per the examples in this project - the example with the "case switch" and the "obd_state" var).

With the opel corsa, I've queried the "supported pids" pid sending 41 00.
The car returned:
BE 1F B8 11
Knowing that Maf Rate is the pid 0x10 (or 16 decimal), it would be 16th bit of the above DWORD - last bit of the 2nd byte:
1F --> 0001 1111 --> last bit 1 --> MAFRate should be reported ....

Please not that MAFRate() if not returned by neither of the reported cars above, including the 2020 Mercedes.

Could it be the protocol ? what is the best way to force other protocols ?

Could it be my ELM327 device ? it is an old cable based reade - but oddly other softwares like PCMSCAN running in windows are able to read those pids ....

Thank you once more.

closed 2025-02-09 by PowerBroker2