PowerBroker2/ELMduino #167: ESP32_test.ino does not work well

NickEsColR, 2023-04-29:
I am trying to get some data from a car Renault Logan 2018. I buy an ELM327 adapter has the one in the following image
[image]
I already check if the adapter can read data using an app from play store, has can be seen in the following 2 images it shows some data and can be sure at least rpm is the real data.
[image]
[image]
But when using an ESP32 and connect via bluetooth using MAC address with the example code and sent the AT commands it respond the following
[image]

I also check using a serial monitor app and it respond to some but not all AT commands. For example, the AT Z respond with ELM327v2.1, and ATSP0 ok. But some others like ATSP6, answer with question mark '?'
Also tried to send obd commands like 0100 or 010C but every one return '?'

I appreciate if anyone know why the some AT commands dont work and how i can get at least 1 data that is the requirement of my project.

PowerBroker2, 2023-04-29:
What happens if you do CR and NL line ending?

NickEsColR, 2023-04-29:
> What happens if you do CR and NL line ending?

[image]
answer looks almost the same the only difference is this time doesn't repeat the command

PowerBroker2, 2023-04-30:
Ok, in that case, go back to only CR line ending.

I'd say ensure no other bluetooth device is connected to the ELM327 at the same time. Also ensure your car is on and running. Lastly try all different bauds.

If none of these fix it, use a better OBD2 scanner.

NickEsColR, 2023-04-30:
I got news, after trying every baud without success and currently in 19200 i decide to enter '>ATZ' instead of 'AT Z' and it succesfully answer. also want to add using space dont respond has expected
[image]
I also try using the AT commands suggested in the test script and all respond OK, but using OBD pids still dont respond as you can see in the following image
[image]
I am curious if maybe exist more than one way to ask for them has the AT commands situation, also i would add that the app i previously use to verify data can be retrieve has 2 step connection, first connect to ELM327 and after connect to ECU.

PowerBroker2, 2023-04-30:
Idk what your scanner is doing, I think you should use a better one

closed 2023-12-21 by PowerBroker2