PowerBroker2/ELMduino #168: ELM_TIMEOUT issue with Civic 8th gen

subwayaddicted, 2023-05-24:
Hi @PowerBroker2 !

So I'm facing similar ELM_TIMEOUT with my 8th gen 2007 Hatchback Civic. You've mentioned in [this issue](https://github.com/PowerBroker2/ELMduino/issues/163) that pre-2008 is not completely compliant with OBDII standards, can you please give more info on that, if possible?

PowerBroker2, 2023-05-24:
OBD2 was standardized for cars starting in 2008, so anything at or before 2008 I can't guarantee will work with this lib or ELM327's in general. You should post your (formatted) sketch and (formatted) debug prints where the car goes from a good reading to a timeout

vladislav-n-dario, 2023-05-24:
Sure, thanks for the swift answer! I will prepare everything and post tomorrow. Thanks a lot!

subwayaddicted, 2023-05-24:
Sure, thanks for the swift answer! I will prepare everything and post tomorrow. Thanks a lot!

subwayaddicted, 2023-05-25:
Code:

```
#include <BluetoothSerial.h>
#include <ELMduino.h>
#include <atomic>

// const uint8_t led = 2;

ELM327 obd;
BluetoothSerial SerialBT;
TaskHandle_t rpm_task;

std::atomic<short> rpm{ 0 };

const char* pin = "1234";
uint8_t address[6] = { 0xB1, 0x00, 0x27, 0x36, 0x9D, 0xED };

void setup(void) {
  Serial.begin(115200);

  connect_to_obd();

  // pinMode(led, OUTPUT);

  // Do blocking OBD call on one core
  xTaskCreatePinnedToCore(
    get_rpm_task,  // Task function.
    "Task1",       // Name of task.
    10000,         // Stack size of task
    NULL,          // Parameter of the task
    0,             // Priority of the task
    &rpm_task,     // Task handle to keep track of created task
    0);            // Pin task to core 0

  delay(500);
}

void loop() {
  vTaskDelete(NULL);
}

void get_rpm_task(void* parameters) {
  for (;;) {
    read_rpm();
    // Sadly you need this or you will starve the IDLE task
    delay(1);
  }
}

void connect_to_obd() {
  SerialBT.begin("auto_friend", true);
  SerialBT.setPin(pin);

  bool connected = SerialBT.connect(address);

  if (connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while (!SerialBT.connected(10000)) {
      Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app.");
    }
  }

  while (!obd.begin(SerialBT, false, 1000, 3)) {
    Serial.println("Couldn't connect to OBD scanner");
  }

  Serial.println("Connected to ELM327");
}

void read_rpm() {
  static unsigned long current_time{ 0 };
  static unsigned long previous_rpm_time{ 0 };
  static const int rpm_interval{ 52 };

  current_time = millis();

  if (current_time - previous_rpm_time >= rpm_interval) {
    previous_rpm_time = current_time;

    if (obd.nb_rx_state != ELM_GETTING_MSG && obd.nb_rx_state != ELM_SUCCESS) {
      obd.printError();
    }

    rpm = static_cast<short>(obd.rpm());
    // throttle = static_cast<short>(obd.throttle());

    if (rpm != 0) {
      Serial.print("RPM: ");
      Serial.println(rpm);
      // Serial.print("THROTTLE: "); Serial.println(rpm);
      if (rpm >= 1500) {
        Serial.print("RPM above 1500: ");
        Serial.println(rpm);
      }
      if (rpm >= 1000 && rpm <= 1500) {
        Serial.print("RPM above 1500: ");
        Serial.println(rpm);
      }
      if (rpm <= 1000) {
        Serial.print("RPM below 1500: ");
        Serial.println(rpm);
      }
    }
  }
}
```

subwayaddicted, 2023-05-25:
Output is from slightly different project, differs only in missing Serials for RPM above\below:
```
16:52:03.238 -> RPM: 758
16:52:03.270 -> Service: 1
16:52:03.270 -> PID: 12
16:52:03.270 -> Normal length query detected
16:52:03.270 -> Query string: 010C1
16:52:03.270 -> Clearing input serial buffer
16:52:03.270 -> Sending the following command/query: 010C1
16:52:03.367 -> 	Received char: 0
16:52:03.433 -> 	Received char: 1
16:52:03.498 -> 	Received char: 0
16:52:03.531 -> 	Received char: C
16:52:03.597 -> 	Received char: 1
16:52:03.630 -> 	Received char: \r
16:52:03.696 -> 	Received char: 4
16:52:03.728 -> 	Received char: 1
16:52:03.793 -> 	Received char: _
16:52:03.859 -> 	Received char: 0
16:52:03.891 -> 	Received char: C
16:52:03.957 -> 	Received char: _
16:52:03.989 -> 	Received char: 0
16:52:04.055 -> 	Received char: B
16:52:04.121 -> 	Received char: _
16:52:04.153 -> 	Received char: E
16:52:04.217 -> 	Received char: C
16:52:04.283 -> 	Received char: _
16:52:04.314 -> 	Received char: \r
16:52:04.380 -> 	Received char: \n
16:52:04.413 -> 	Received char: \r
16:52:04.479 -> 	Received char: \n
16:52:04.511 -> 	Received char: >
16:52:04.511 -> Delimiter found.
16:52:04.544 -> All chars received: 010C1410C0BEC
16:52:04.544 -> Expected response header: 410C
16:52:04.544 -> Single response detected
16:52:04.544 -> 64-bit response: 
16:52:04.544 -> 	responseByte_0: 236
16:52:04.544 -> 	responseByte_1: 11
16:52:04.544 -> 	responseByte_2: 0
16:52:04.544 -> 	responseByte_3: 0
16:52:04.544 -> 	responseByte_4: 0
16:52:04.544 -> 	responseByte_5: 0
16:52:04.544 -> 	responseByte_6: 0
16:52:04.544 -> 	responseByte_7: 0
16:52:04.544 -> RPM: 763
16:52:04.576 -> Service: 1
16:52:04.576 -> PID: 12
16:52:04.576 -> Normal length query detected
16:52:04.609 -> Query string: 010C1
16:52:04.609 -> Clearing input serial buffer
16:52:04.609 -> Sending the following command/query: 010C1
16:52:05.630 -> Timeout detected with overflow of 27ms
16:52:05.696 -> Received: 
16:52:05.696 -> ERROR: ELM_TIMEOUT
16:52:05.799 -> Service: 1
16:52:05.799 -> PID: 12
16:52:06.987 -> Clearing input serial buffer
16:52:07.019 -> Sending the following command/query: 010C1
16:52:08.041 -> Timeout detected with overflow of 41ms
16:52:08.106 -> Received: 
16:52:08.106 -> ERROR: ELM_TIMEOUT
16:52:08.206 -> Service: 1
16:52:08.206 -> PID: 12
16:52:08.206 -> Normal length query detected
16:52:08.206 -> Query string: 010C1
16:52:08.206 -> Clearing input serial buffer
16:52:08.206 -> Sending the following command/query: 010C1
16:52:09.261 -> Timeout detected with overflow of 41ms
16:52:09.293 -> Received: 
16:52:09.326 -> ERROR: ELM_TIMEOUT
16:52:09.425 -> Service: 1
16:52:09.425 -> PID: 12
```

And continues with timeouts

subwayaddicted, 2023-05-25:
Also, will test on Accord 10 2018, had a small one time test, worked pretty stable, but need to do long time run.

PowerBroker2, 2023-05-25:
You might want to try without specifying the protocol. Other than that, I'm not sure why the ELM327 or diagnostics computer is going non-responsive

closed 2023-12-21 by PowerBroker2