PowerBroker2/ELMduino #169: Cant establish a bluetooth connection

amerr97, 2023-06-05:
Hi,
I'm trying to connect to the elm327 dongle but with no success. 
What I have tried.

-Making sure the bluetooth is turned off on my phone ( I've used the dongle with car scanner app), and off on my laptop.
-Changing all types of baud rates that are less than 115200.
-running bt_remove_paired devices to remove any paired devices with esp32
-running Discover Connect 
It finds my elm327 device, name is OBDII 
It sends the default message hello this is esp32... 

Then I tried running the 
ESP32_test
I get the compiling error for
`LED_BUILTIN
`
I set #define LED_BUILTIN 2
It compiles. But again It doesn't want to connect with my ELM327 ( tried setting the baudrades no luck)

Any tips, do i have to do something differently in Discover Connect ?

By the way I tried running ESP32_bluetooth_serial before all of this, and I wasn't able to connect.

PowerBroker2, 2023-06-05:
What MCU are you using? Is it an ESP32? If so, what kind? Also, when you run the sketch that can compile, what are the debug messages you see in the serial monitor?

amerr97, 2023-06-05:
> What MCU are you using? Is it an ESP32? If so, what kind? Also, when you run the sketch that can compile, what are the debug messages you see in the serial monitor?

Hi, thank you for the reply. Yes, I'am using ESP32 Dev module. 
I get: "Attempting to connect to ELM327...
Couldn't connect to OBD scanner

I ran, the remove_bonded_devices  in 0 when I came back, it didn't bond with ELM327 adapter.

PowerBroker2, 2023-06-05:
If it's failing on "phase 1" of connecting, that's an issue with your ESP32 and OBD2 scanner. Maybe try using the OBD2 mac address or using a "1234" pin? I'm not sure - maybe ask in an Espressif or Arduino forum

closed 2023-06-05 by PowerBroker2