PowerBroker2/ELMduino #171: Improve data transfer speed

DanteCecchetti, 2023-07-02:
Hello, I am using your library to develop a shift light identical to the one that is put on car steering wheels. I currently have a rate of 22 data per second. I wanted to ask you if there is any configuration to further improve the speed of data transfer, or what parameters should I vary to try to improve the transfer. I think I use the same obd that you used to develop the library, the BAFX obd 2.

PowerBroker2, 2023-07-02:
I think the default parameters enable as fast of transfer as you can expect. I'm not aware of anything further you can do. Perhaps there's some optimization you can do in your sketch?

DanteCecchetti, 2023-07-03:
In reality I can't modify the program much more and I optimized it to more I couldn't, besides I'm using an esp 32 at 240 mhz. Did you ever try to wire the obd to bypass the bluetooth connection. Because capable bluetooth can be one of the limiting factors.
And I ask you something else, could you develop a function to clean the check engine? If so, how do I do it?

closed 2023-11-28 by PowerBroker2