PowerBroker2/ELMduino #172: Questions from a beginner

mateusdealencar, 2023-07-04:
Hey, congrats on the library, it's great.

But I can't connect. I'm a beginner in arduino, so I apologize in advance if I'm missing something basic.

I'll describe step by step what I did:

1. I connected the HC-05's RX to port 3 and the TX to port 2, as shown in the diagram below:

[image]

2. I loaded the example code on the arduino, changing the ports to RX on 3 and TX on 2, exactly like this:

#include <SoftwareSerial.h>
#include "ELMduino.h"

SoftwareSerial mySerial(3, 2); // RX, TX
ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
  Serial.begin(115200);
  mySerial.begin(115200);
  myELM327.begin(mySerial, true, 2000);
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}

3. I connected the ELM327 to my car and stayed inside the car waiting for it to pair with the arduino + HC-05 turned on. But nothing. I tried the baud like 38400, 57600 and 115200. But nothing.

The question is: what procedure is this? Did I make the right connections? Do I really have to just wait to connect?

If everything is right, where could I be going wrong?

enslinm123, 2023-07-04:
Which Arduino are you using specifically ?

If you are using a Arduino Uno, this might solve your problem. Connect your RX from HC05 to TX pin of Arduino, TX from HC05 to RX pin of Arduino, adapt your sketch to reflect the correct pins and try it again. I've seen a lot of people make the mistake of connecting RX from HC05 to RX of the Arduino and TX from HC05 to TX of the Arduino.

For more details on the pinouts for different Ardiono's: [https://www.arduino.cc/reference/en/language/functions/communication/serial/](url)`

mateusdealencar, 2023-07-04:
I tried connecting the HC-05's RX to the Arduino Uno's TX and the HC-05's TX to the Arduino Uno's RX (HC-05's RX to pin 1 and the TX to pin 0). However, the arduino gives a generic error that does not allow the code to be uploaded.

I was able to upload the code with the pins disconnected, but it still didn't connect with the ELM.

Additionally the output on the serial monitor is setting to a data type that I cannot interpret if there is any sign of what is going on.

[image]

enslinm123, 2023-07-04:
Ok so first off, I'm by no mens an arduino expert and still a noob myself, however with PB's help I got my el cheapo Bluetooth ELM327 working with the ESP32's Bluetooth and communicating with my ecu, although its intermittently...but that's a bug in my code which I've not figured out yet, but I digress...

Secondly, reconnect the wires the same way you had them initially (HC05 RX pin 3 and HC05 TX to pin 2). Then swop the RX and TX (pin 3 with 2 and 2 with 3) wires on the arduino's side so that the HC05 RX goes to whichever pin you assign for TX on the arduino and the same with the TX of the HC05 to the pin you assign for RX. 

Think of RX and TX like a 2 way radio. If both people press the button to transmit and speak, you hear nothing on either end, but if one transmits (TX) the other end receives (RX) the signal. To transmit something back it uses its own TX to send and the other end uses its RX to receive, hope it makes sense😁

Thirdly, set your baud rate on serial monitor to 11500 (you have it at 9600, which will for sure give you gibberish) like you have it in your sketch 😉

And lastly, let me know how it goes. 

If I'm not mistaken you have to set the HC05 in master mode or something for it to accept the BT comms from your elm327. I have a bare hc05 I still need to play with, so I cannot say for sure.

mateusdealencar, 2023-07-04:
Many thanks for the suggestions!

I think I'm missing something in the HC-05 configuration.

I made the following settings on my HC-05:

AT+ROLE=1 (Set as Master)
AT+CMODE=0 (Connect to a single device)
AT+UART:115200,1,0 (115200 baud)
AT+BIND=0000,00,333333 (My ELM address is 00:00:00:33:33:33)

I then connected the RX and TX pins the way you indicated, but still nothing. I really wanted to be able to connect.

@PowerBroker2, do you have any suggestions where I'm going wrong?

PowerBroker2, 2023-07-05:
From the screenshot it looks like a bad baud. Perhaps trying different bauds will help (you will have to reconfigure both your HC-05 and sketch for each baud)

closed 2023-12-21 by PowerBroker2