PowerBroker2/ELMduino #173: Odometer value

vizziniroberto71, 2023-07-16:
HI,
I'm using your library for a project and it's been very useful, really nice work!
I also wanted to enter the value of the kilometers traveled (odometer), but I can't find the entry in the library. Am I looking wrong or hasn't been implemented? I checked on Wikipedia the formula for the calculation (PID 166) but not being an expert I don't know how to implement it. Is there a way to extract the value using the library, or is this a function you would have to implement from scratch?
Thank you.
Robert.

patfelst, 2023-07-16:
you'll have to code it yourself. Just copy one of the other PID functions, and replace the PID with 166. Something like this (I've not tested it):
```c++
uint8_t ELM327::odometer()
{
	return (uint8_t)processPID(SERVICE_01, 166, 1, 4, 0.1);

}
```

Note that it returns 4 bytes, and you need to divide the result by 10 (that's what the 0.1 scale factor is for)

vizziniroberto71, 2023-07-16:
Hi Patrick,
thanks for your suggestion, I modified the library to also read the odometer value, but when compiling it returns the following error:
c:\Users\Administrator\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp: In member function 'uint32_t ELM327::supportedPIDs_161_192()':
c:\Users\Administrator\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:2261:42: error: 'SUPPORTED_PIDS_161_192' was not declared in this scope
  return (uint32_t)processPID(SERVICE_01, SUPPORTED_PIDS_161_192, 1, 4);
                                          ^~~~~~~~~~~~~~~~~~~~~~
c:\Users\Administrator\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:2261:42: note: suggested alternative: 'SUPPORTED_PIDS_61_80'
  return (uint32_t)processPID(SERVICE_01, SUPPORTED_PIDS_161_192, 1, 4);
                                          ^~~~~~~~~~~~~~~~~~~~~~
                                          SUPPORTED_PIDS_61_80
c:\Users\Administrator\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp: In member function 'uint32_t ELM327::odometer()':
c:\Users\Administrator\Documents\Arduino\libraries\ELMDuino\src\ELMduino.cpp:2284:42: error: 'ODOMETER' was not declared in this scope
  return (uint32_t)processPID(SERVICE_01, ODOMETER, 1, 4, 1.0 / 10.0);

this error is returned at compile time. The changes made on the two library files are:
`uint32_t supportedPIDs_161_192();
	
 uint32_t odometer();`
these lines have been inserted in the elmduino.h file!

In the elmduino.cpp file i inserted:
/*
 uint32_t ELM327::supportedPIDs_161_192()

 Description:
 ------------
  * Determine which of PIDs 0xA1 through 0xC0 are supported (bit encoded)

 Inputs:
 -------
  * void

 Return:
 -------
  * uint32_t - Bit encoded booleans of supported PIDs 0xA1-0xC0
*/

uint32_t ELM327::supportedPIDs_161_192()
{
	return (uint32_t)processPID(SERVICE_01, SUPPORTED_PIDS_161_192, 1, 4);
}

/*
 uint32_t ELM327::odometer()

 Description:
 ------------
  *  Queries and parses received message for/returns Total Kilometers Traveled (km)

 Inputs:
 -------
  * void

 Return:
 -------
  * uint32_t - Total kilometers traveled
*/

uint32_t ELM327::odometer()
{
	return (uint32_t)processPID(SERVICE_01, ODOMETER, 1, 4, 1.0 / 10.0);
}

I'm sorry if I bother you for this triviality, but I can't understand the error and it is the last step of the project that is still missing.
Thank you.

Roberto.

patfelst, 2023-07-17:
I wouldn't worry about the `uint32_t supportedPIDs_161_192()`, it's not needed, for the moment, lets just assume your vehicle supports reporting of odometer.

Also you've referenced `ODOMETER` when it hasn't been defined in the .h file. To start with, just hard code everything as per my example, if it works, then maybe it's worth defining ODOMETER constant etc. I assumed you knew you needed to add the function prototype to the class.

In the .h file where all the other functions are defined - it doesn't really matter where - but I suggest here
https://github.com/PowerBroker2/ELMduino/blob/eabbb1a2e8e82d8fed1b758c0d893bc46efbe171/src/ELMduino.h#L410
add `uint8_t odometer();`

and paste in the original function I provided in the .cpp file.

that should be all you need to do.

vizziniroberto71, 2023-07-20:
Hi Patrick, thank you very much for your support, I compiled the sketch
according to your suggestions and the error disappeared, I can't see the
data on the display but there must be some error in the Nextion code. Thank
you. Roberto.

patfelst, 2023-07-26:
try turning debug on to see what your vehicle is reporting - this will be independent of your Nextion display.

have a look at the multiple PIDs example for how to turn debug on
https://github.com/PowerBroker2/ELMduino/blob/master/examples/multiple_pids/multiple_pids.ino

vizziniroberto71, 2023-07-26:
Thanks again Patrick, I will try your suggestion these days.
See you soon.

Roberto.

jimwhitelaw, 2023-12-17:
I have created a new method to get the odometer reading and an example program to run it. It should be working (works with an emulator) but I'm unable to properly test it as I don't have a vehicle that supports the OBDII spec PID ("A6") to test it with. If someone wants to have a go at it, it's on the "odometer" branch of my ELMduino fork. https://github.com/jimwhitelaw/ELMduino.git

closed 2025-02-09 by PowerBroker2