PowerBroker2/ELMduino #175: ELM327 to HC-05 Module

hammikb, 2023-08-03:
Hello i am trying to connect my device (Teensy with an HC-05 module) to an elm327 . I have done all the commands you recommended by it is failing at the LINK stage in the code. I have tried resetting and re running all the commands again on the hc-05 module also have tried restarting the elm327 multiple times to see if it was something with it but i was able to connect to it with another device. Thank you in advance

hammikb, 2023-08-03:
Ran all over these (+INQ:1D:A5:3A741,1F00,7FFF):

 
AT+RESET
AT+ORGL
AT+ROLE=1
AT+CMODE=0
AT+NAME=HC05
AT+INIT
AT+PAIR=1D,A5,3A741,20
AT+BIND=1D,A5,3A741
AT+CMODE=1
AT+LINK=1D,A5,3A741

hammikb, 2023-08-07:
Am I using the wrong hc-05 module? Or do they not work well with Esp32? I'm just confused why it can't link

PowerBroker2, 2023-08-07:
Why are you using an HC-05 with an ESP32? Does your ESP32 not have built-in bluetooth?

hammikb, 2023-08-07:
> Why are you using an HC-05 with an ESP32? Does your ESP32 not have built-in bluetooth?

I can be completely wrong but the board shares the same antenna as WiFi, so you can only fully use one at a time. I want to be able to use Bluetooth connection with the obd2 and use the WiFi to set up an ap web server

PowerBroker2, 2023-08-07:
That's true, you can only use WiFi or BT at one time. Sometimes HC-05s are tricky when programming and some require you to reboot in AT mode (with the tiny button) after certain commands and some commands require you to specifically hold the button during the execution of said commands. You might get better help in the Arduino forum or some forum that deals with HC devices.

closed 2023-08-07 by PowerBroker2

hammikb, 2023-08-16:
@PowerBroker2 By chance do you have a link to the one you used for both the module and OBD2 Scanner? Also curious if my OBD2 Bluetooth Scanner i have no doesnt have a pin would that be what is messing me up?

PowerBroker2, 2023-08-16:
I don't have a link to the HC-05, but [this is the OBD2 scanner I use](https://www.amazon.com/gp/product/B005NLQAHS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&th=1).

As for the pin question, you might try uncommenting this line and see if it works:

https://github.com/PowerBroker2/ELMduino/blob/eabbb1a2e8e82d8fed1b758c0d893bc46efbe171/examples/ESP32_test/ESP32_test.ino#L27

I vaguely remember various issues discussing support for scanners with pins - you might try searching them and see if their inputs are any help.

Also note that the one I use (the one I linked above) does not need a pin.

PowerBroker2, 2023-08-16:
Wait, I just realized I think you were talking about the pin of the HC-05. In this case, I'm referring to the pin button (physical button). These are two different things. If your unit does not have a programming mode pin, you do not have an HC-05 and that might be why it isn't working.

hammikb, 2023-08-16:
@PowerBroker2 i just meant if my obd2 doesn't have a pin would I still be able to pair it with an hc-05 and I get all the commands to work on my hc-05 but pair and link fail so I'm thinking maybe it's my obd2 scanner that doesn't want to link. Now after all this my original script with my Esp32 won't directly connect to the obd2 anymore. (But my computer can connect to it , so idk what's going on anymore)

hammikb, 2023-08-17:
Also these things are just so weird i just bought another OBD2 dongle of the same exact one. The new one connects to both the computer and the esp32 but the old one can now only connect to the computer no matter how i change the code and i have4 tried to both connect by name and mac address but only one works with esp32

hammikb, 2023-08-18:
So for some reason when i pair the HC-05 to the OBD2 ELM327 adapter it passes OK but then Fails to link. Once i try to go back to the esp32 it fails entirely and wont work after that anymore regardless of what i do. Its like it paired to the hc05 adapter and wont unpair yet my computer is able to connect