PowerBroker2/ELMduino #176: Why does it take so long to connect?

edo44, 2023-08-05:
Via bluetooth the esp takes 27 seconds to connect. Even if I specify the protocol in ELM327::begin() . Why is that?
Is it maybe the ELM327 the cause?

PowerBroker2, 2023-08-07:
No idea. Maybe try a different OBD2 scanner and/or arduino?

edo44, 2023-08-07:
> No idea. Maybe try a different OBD2 scanner and/or arduino?

How long does it usually take to connect to you?

> Try connecting by BT MAC address instead of BT name. It is faster in my experience. There are plenty of closed issues that describe how to do this.

I'm already conecting using the MAC address of the device.

patfelst, 2023-08-07:
I don't know then. Try compiling with verbose logging turned on, you'll be see the whole BT connection process, it may give some clues.

edo44, 2023-08-07:
> I don't know then. Try compiling with verbose logging turned on, you'll be see the whole BT connection process, it may give some clues.

How do I activate the verbose logging?

patfelst, 2023-08-07:
I only use PlatformIO and from there you set the log level using a build flag in the `platform.io` file as follows:
```c++
build_flags = 
  ; -DCORE_DEBUG_LEVEL=0 ; None
  ; -DCORE_DEBUG_LEVEL=1 ; Error
  ; -DCORE_DEBUG_LEVEL=2 ; Warn
  ; -DCORE_DEBUG_LEVEL=3 ; Info
  ; -DCORE_DEBUG_LEVEL=4 ; Debug
  -DCORE_DEBUG_LEVEL=5 ; Verbose
```
If you use Arduino IDE, I assume you can add this line to the top of your .ino file:
```c++
#define CORE_DEBUG_LEVEL 5 // Turn on verbose logging
```

patfelst, 2023-08-07:
apparently you can set the log level from the Arduino IDE tools menu instead of using the #define method
[image]

patfelst, 2023-08-07:
oh great, glad you found the problem!

closed 2023-08-07 by PowerBroker2