PowerBroker2/ELMduino #177: Trying to get to work monitorStatus();

bensonOBD, 2023-08-07:
I have been trying to get Number of active DTCs from the ELM327. So I guess the best way is use the function monitorStatus();
Does anyone know what does it return?
I tried Serial.println(myELM327.monitorStatus()); and got output as "24576". 
I cant find anything that corresponds.

From the ELM datasheet, i understood that sending a request 0101 would give some values and second byte holds the info on number of DTCs.
Is myELM327.sendCommand("01 01"); the proper way ?
And how can i Read the output of these commands?

PowerBroker2, 2023-08-07:
Might look into this later, but I've been super busy and will be for the foreseeable future

bensonOBD, 2023-08-15:
Thank you @PowerBroker2 for taking the time for this library.
I have changed another code for my use and it works for me.

Linking the arduino forum post here, 
[https://forum.arduino.cc/t/trying-to-detect-dtc-error-codes-with-obd2-and-hc05/1153446/12?u=maverick8086]

It checks whether DTCs are present and works just like a check engine light.

closed 2023-08-15 by bensonOBD