PowerBroker2/ELMduino #178: "Guru Meditation Error: Core  1 panic'ed (StoreProhibited)" error when using a function, example 'myELM327.rpm()'

penright, 2023-08-18:
I am in the "hello world" stage. I am using VScode with PlatformIO as the development platform. I did test in the Arduino environment, with the same issue. 

I am connecting via Bluetooth. The first test was connecting and testing the AT commands. This worked as expected. I was able to send an '01 00' and received the expected response. 

I am using the ELM-emulator for testing.  https://github.com/Ircama/ELM327-emulator. I tested with 'SerialBluetooth' on my Andriod phone. At commands and PID's responses were as expected. Again, the echo commands of the demo code worked as well. Just when trying to do a 'myELM327.rpm()" or 'myELM327.EngineCoolantTemp() do I get the fatal error.

Using the "esp32_exception_decoder" this is more what the issue is:

> Start of poll
> Guru Meditation Error: Core  1 panic'ed (StoreProhibited). Exception was unhandled.
> 
> Core  1 register dump:
> PC      : 0x4009024f  PS      : 0x00060a30  A0      : 0x800d256d  A1      : 0x3ffca230
> A2      : 0x00000000  A3      : 0x00000000  A4      : 0x00000001  A5      : 0x00000000  
> A6      : 0x3ffca56c  A7      : 0x00000000  A8      : 0x00000004  A9      : 0x00000004
> A10     : 0x3ffc3849  A11     : 0x00000002  A12     : 0x0000005a  A13     : 0x00000000  
> A14     : 0x3ffca56c  A15     : 0x00000000  SAR     : 0x00000016  EXCCAUSE: 0x0000001d
> EXCVADDR: 0x00000000  LBEG    : 0x40090228  LEND    : 0x40090233  LCOUNT  : 0xffffffff  
> 
> 
> Backtrace: 0x4009024c:0x3ffca230 0x400d256a:0x3ffca240 0x400d25c8:0x3ffca260 0x400d2a37:0x3ffca280 0x400d2ab1:0x3ffca2b0 0x400d2377:0x3ffca2f0 0x400d6171:0x3ffca320
> 
>   #0  0x4009024c:0x3ffca230 in memset at /builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/newlib/newlib/libc/machine/xtensa/memset.S:182
>   #1  0x400d256a:0x3ffca240 in ELM327::sendCommand(char const*) at .pio/libdeps/wemos_d1_mini32/ELMDuino/src/ELMduino.cpp:2218
>   #2  0x400d25c8:0x3ffca260 in ELM327::queryPID(unsigned char const&, unsigned short const&, unsigned char const&) at .pio/libdeps/wemos_d1_mini32/ELMDuino/src/ELMduino.cpp:434
>   #3  0x400d2a37:0x3ffca280 in ELM327::processPID(unsigned char const&, unsigned short const&, unsigned char const&, unsigned char const&, float const&, float const&) at .pio/libdeps/wemos_d1_mini32/ELMDuino/src/ELMduino.cpp:499
>   #4  0x400d2ab1:0x3ffca2b0 in ELM327::rpm() at .pio/libdeps/wemos_d1_mini32/ELMDuino/src/ELMduino.cpp:814
>   #5  0x400d2377:0x3ffca2f0 in loop() at src/main.cpp:56
>   #6  0x400d6171:0x3ffca320 in loopTask(void*) at C:/Users/paul/.platformio/packages/framework-arduinoespressif32/cores/esp32/main.cpp:50
> 
> 
> 
> 
> ELF file SHA256: c3f3c942989aa7d2
> 
> Rebooting...

In my search, the closest I could find was someone using memset() and not declaring the int type. I know this has got to be working so, it has to be something I am doing and not the code.

The code is basically the demo, with adding the test with a predefined function. 
Any thoughts?

### PlatformIO.ini

```

[env:wemos_d1_mini32]
platform = espressif32
board = wemos_d1_mini32
framework = arduino

monitor_port = COM22
upload_port = COM22
monitor_speed=115200
monitor_filters = esp32_exception_decoder

lib_deps = powerbroker2/ELMDuino@^3.0.2

```

### Code

```

#include <Arduino.h>
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT SerialBT
ELM327 myELM327;
uint32_t EngineCoolantTemp = 0;

void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);
  delay(10000);
  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("MIS-PENRIGHT-PC"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while (1)
      ;
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if (DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if (ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if (c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
  DEBUG_PORT.println("Start of poll");
  float tmpEngineCoolantTemp = myELM327.rpm();

  while (myELM327.nb_rx_state != ELM_SUCCESS)
  {
    DEBUG_PORT.print("RPM: ");
    if (myELM327.nb_rx_state != ELM_SUCCESS)
    {
      EngineCoolantTemp = (uint32_t)tmpEngineCoolantTemp;
      DEBUG_PORT.println(EngineCoolantTemp);
      break;
    }
    else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    {
      myELM327.printError();
      break;
    }
    delay(1000);
  }
}

```

PowerBroker2, 2023-08-18:
I'm not sure what's causing this issue. Perhaps try changing the third argument in `sendCommand()`'s `memset()` call?

penright, 2023-08-18:
Give me some time, I have always just pointed to a lib, using PlatformIO, and never have tried to modify one. I want to learn, let me do some searching on what to do and I will get back. I tried to patch it to turn on debug, but that did not work.  
Once I get it figured out, What should I change it to?
Screenshot
[image]

Is there a way from the main setup to enable debugging? I tried myELM327.debugMode = True. 
I also tried to hardcode it in the formatQueryArray, which is why I don't think I have to modify libs down. :-) 
[image]
I was expecting this to dump to the serial monitor.

penright, 2023-08-18:
Should PAYLOAD_LEN be 0 at this point?

This is the only place I found where PAYLOAD was set

```
bool ELM327::begin(Stream &stream, const bool &debug, const uint16_t &timeout, const char &protocol, const uint16_t &payloadLen, const byte &dataTimeout)
{
	elm_port = &stream;
	PAYLOAD_LEN = payloadLen;
	debugMode = debug;
	timeout_ms = timeout;

	Serial.println("begin - payloadLen: " + payloadLen);
	payload = (char *)malloc(PAYLOAD_LEN + 1); // allow for terminating '\0'

	// test if serial port is connected
	if (!elm_port)
		return false;

	// try to connect
	if (!initializeELM(protocol, dataTimeout))
		return false;

	return true;
}
```

As you can see, I added a Serial.println() in that function and I did not see it executed.
Is that a difference between Bluetooth and Serial?

The bottom line, I don't think the memory is being allocated.  

I added a Serial.println above the memset to see and I was getting the error on strlen(). 
[image]
[image]

animevietsub, 2023-08-20:
Make sure that you have `ELM327::begin` in the `void setup()`.
In your code that should be something like:

```
void setup()
{
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);
  delay(10000);
  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("MIS-PENRIGHT-PC"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while (1);
  }

  if (!myELM327.begin(Serial1, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner");
    while (1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}
```

penright, 2023-08-21:
I deleted my previous post, it was way off although it did point me in the right direction. It caused me to go back to the example "https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino" 

I am guessing where I got confused was between learning to connect to the Bluetooth and ELMduino library. Caught my mistake, @animevietsub  was right, in cutting and pasting I missed the myELM327.begin. 

Most of my class experience was with .NET. After searching and learning more about C++ class constructor/destructor and typing the deleted response. I saw the errors of my way. I figured it had to be a dumb mistake which is why it was a question and not a bug. :-) Sorry for the confusion. Phase 2 errored when setting the SP (AT SP0). That could be an issue with my emulator, I will need to do some digging. I am going to close this issue. Thanks all.

closed 2023-08-21 by penright