PowerBroker2/ELMduino #179: M5STACK with ELMduino not working

zejulio, 2023-08-22:
Hello,

First of all many thanks for this awesome lib, really want this to work on my ELM327 (I think it's a clone, not original, but works great with Torque).

I am trying to reproduce this example:
https://embedscope.com/125

Changed some stuff, since I think it's using an old version of the lib and myELM327.status no longer exist so replace with myELM327.nb_rx_state.
I cannot display any data, I've even removed all other and tried only RPM, but even then is not working, can anyone help me out?

I attach the debug log:
`
16:12:01.452 == Start Up == 
16:12:27.116 Clearing input serial buffer 
16:12:27.116 Sending the following command/query: AT D 
16:12:27.129 ^IReceived char: ?
16:12:27.129 ^IReceived char: \r
16:12:27.129 ^IReceived char: \r
16:12:27.129 ^IReceived char: > 
16:12:27.134 Delimiter found. 
16:12:27.134 All chars received: 
16:12:27.229 Clearing input serial buffer 
16:12:27.229 Sending the following command/query: AT Z 
16:12:27.262 ^IReceived char: A
16:12:27.262 ^IReceived char: T
16:12:27.262 ^IReceived char: _
16:12:27.262 ^IReceived char: Z 
16:12:27.265 ^IReceived char: \r
16:12:28.042 ^IReceived char: \r
16:12:28.042 ^IReceived char: \r
16:12:28.042 ^IReceived char: E
16:12:28.042 ^IReceived char: L 
16:12:28.048 ^IReceived char: M
16:12:28.048 ^IReceived char: 3
16:12:28.048 ^IReceived char: 2 
16:12:28.053 ^IReceived char: 7
16:12:28.053 ^IReceived char: _
16:12:28.053 ^IReceived char: v
16:12:28.059 ^IReceived char: 1
16:12:28.059 ^IReceived char: .
16:12:28.059 ^IReceived char: 5
16:12:28.059 ^IReceived char: \r 
16:12:28.064 ^IReceived char: \r
16:12:28.064 ^IReceived char: >
16:12:28.064 Delimiter found. 
16:12:28.068 All chars received: ATZELM327v1.5 
16:12:28.163 Clearing input serial buffer 
16:12:28.163 Sending the following command/query: AT E0 
16:12:28.202 ^IReceived char: A
16:12:28.202 ^IReceived char: T
16:12:28.202 ^IReceived char: _
16:12:28.202 ^IReceived char: E 
16:12:28.206 ^IReceived char: 0
16:12:28.206 ^IReceived char: \r
16:12:28.206 ^IReceived char: O 
16:12:28.212 ^IReceived char: K
16:12:28.212 ^IReceived char: \r
16:12:28.212 ^IReceived char: \r
16:12:28.217 ^IReceived char: >
16:12:28.217 Delimiter found. 
16:12:28.217 All chars received: ATE0OK 
16:12:28.315 Clearing input serial buffer 
16:12:28.315 Sending the following command/query: AT S0 
16:12:28.347 ^IReceived char: O
16:12:28.347 ^IReceived char: K
16:12:28.347 ^IReceived char: \r
16:12:28.347 ^IReceived char: \r 
16:12:28.353 ^IReceived char: >
16:12:28.353 Delimiter found. 
16:12:28.353 All chars received: OK 
16:12:28.457 Clearing input serial buffer 
16:12:28.457 Sending the following command/query: AT AL 
16:12:28.490 ^IReceived char: O
16:12:28.490 ^IReceived char: K
16:12:28.490 ^IReceived char: \r
16:12:28.490 ^IReceived char: \r 
16:12:28.495 ^IReceived char: >
16:12:28.495 Delimiter found. 
16:12:28.495 All chars received: OK 
16:12:28.601 Clearing input serial buffer 
16:12:28.601 Sending the following command/query: AT ST 00 
16:12:28.637 ^IReceived char: O
16:12:28.637 ^IReceived char: K
16:12:28.637 ^IReceived char: \r
16:12:28.637 ^IReceived char: \r 
16:12:28.641 ^IReceived char: >
16:12:28.641 Delimiter found. 
16:12:28.641 All chars received: OK 
16:12:28.747 Clearing input serial buffer 
16:12:28.747 Sending the following command/query: AT TP A0 
16:12:28.780 ^IReceived char: O
16:12:28.780 ^IReceived char: K
16:12:28.780 ^IReceived char: \r
16:12:28.780 ^IReceived char: \r 
16:12:28.785 ^IReceived char: >
16:12:28.785 Delimiter found. 
16:12:28.785 All chars received: OK 
16:12:28.788 Connected to ELM327 
16:12:28.824 Service: 1 
16:12:28.824 PID: 12 
16:12:28.824 Normal length query detected 
16:12:28.824 Query string: 010C1 
16:12:28.829 Clearing input serial buffer 
16:12:28.829 Sending the following command/query: 010C1 
16:12:28.833 Received: 
16:12:28.833 No error detected 
16:12:28.933 ^IReceived char: S
16:12:28.933 Received: S 
16:12:28.933 No error detected 
16:12:29.033 ^IReceived char: E
16:12:29.033 Received: SE 
16:12:29.033 No error detected 
16:12:29.133 ^IReceived char: A
16:12:29.133 Received: SEA 
16:12:29.133 No error detected 
16:12:29.233 ^IReceived char: R
16:12:29.233 Received: SEAR 
16:12:29.233 No error detected 
16:12:29.333 ^IReceived char: C
16:12:29.333 Received: SEARC 
16:12:29.333 No error detected 
16:12:29.434 ^IReceived char: H
16:12:29.434 Received: SEARCH 
16:12:29.434 No error detected 
16:12:29.534 ^IReceived char: I
16:12:29.534 Received: SEARCHI 
16:12:29.534 No error detected 
16:12:29.634 ^IReceived char: N
16:12:29.634 Received: SEARCHIN 
16:12:29.634 No error detected 
16:12:29.734 ^IReceived char: G
16:12:29.734 Received: SEARCHING 
16:12:29.734 No error detected 
16:12:29.834 ^IReceived char: .
16:12:29.834 Received: SEARCHING. 
16:12:29.834 No error detected 
16:12:29.934 ^IReceived char: .
16:12:29.934 Received: SEARCHING.. 
16:12:29.934 No error detected 
16:12:30.034 ^IReceived char: .
16:12:30.034 Received: SEARCHING... 
16:12:30.034 No error detected 
16:12:30.134 ^IReceived char: \r
16:12:30.134 Received: SEARCHING... 
16:12:30.134 No error detected 
16:12:30.233 Received: SEARCHING... 
16:12:30.233 No error detected 
16:12:30.333 Received: SEARCHING... 
16:12:30.333 No error detected 
16:12:30.432 Received: SEARCHING... 
16:12:30.432 No error detected 
16:12:30.533 Received: SEARCHING... 
16:12:30.533 No error detected 
16:12:30.633 Received: SEARCHING... 
16:12:30.633 No error detected 
16:12:30.732 Received: SEARCHING... 
16:12:30.732 No error detected 
16:12:30.834 Timeout detected with overflow of 0ms 
16:12:30.834 Received: SEARCHING... 
16:12:30.834 ERROR: ELM_TIMEOUT 
16:12:30.934 Service: 1 
16:12:30.934 PID: 12 
16:12:30.934 Normal length query detected 
16:12:30.934 Query string: 010C1 
16:12:30.940 Clearing input serial buffer 
16:12:30.940 Sending the following command/query: 010C1 
16:12:30.944 Received: 
16:12:30.944 No error detected 
16:12:31.043 ^IReceived char: S
16:12:31.043 Received: S 
16:12:31.043 No error detected 
16:12:31.143 ^IReceived char: T
16:12:31.143 Received: ST 
16:12:31.143 No error detected 
16:12:31.243 ^IReceived char: O
16:12:31.243 Received: STO 
16:12:31.243 No error detected 
16:12:31.343 ^IReceived char: P
16:12:31.343 Received: STOP 
16:12:31.343 No error detected 
16:12:31.444 ^IReceived char: P
16:12:31.444 Received: STOPP 
16:12:31.444 No error detected 
16:12:31.543 ^IReceived char: E
16:12:31.543 Received: STOPPE 
16:12:31.543 No error detected 
16:12:31.644 ^IReceived char: D
16:12:31.644 Received: STOPPED 
16:12:31.644 No error detected 
16:12:31.744 ^IReceived char: \r
16:12:31.744 Received: STOPPED 
16:12:31.744 No error detected 
16:12:31.844 ^IReceived char: \r
16:12:31.844 Received: STOPPED 
16:12:31.844 No error detected 
16:12:31.944 ^IReceived char: >
16:12:31.944 Delimiter found. 
16:12:31.944 All chars received: STOPPED 
16:12:31.950 ELM responded with errror "STOPPED" 
16:12:31.950 Received: STOPPED 
16:12:31.950 ERROR: ELM_STOPPED 
`

zejulio, 2023-08-22:
Just compiled the ESP32_Bluetooth_Serial.ino example and it's the same output.
BUT, if I shut down the car, and turn ON again, WITH the BT connected, I can get the RPM values....
Whats wrong?

PowerBroker2, 2023-08-22:
Sorry, I can't speak Korean, but it should be relatively easy to update the example sketch in the linked website using [the up-to-date examples given in the library](https://github.com/PowerBroker2/ELMduino/tree/master/examples). Best advice is to start with a working example from the library and work up to the level of functionality that you need for your project slowly.

zejulio, 2023-08-22:
Neither do I 😅.
Just used the code.
But as stated it also dont work with the esp32 bluetooth serial example...
Its always searching and stopping...
If I turn off the engine and turn on again sometimes it work...
Any idea what can it be?

PowerBroker2, 2023-08-22:
Ah, I see. The only idea I can come up with is maybe the ELM327 is connecting not to your ESP32, but with another BT device (laptop, phone, etc) before the ESP32 can. Be sure to unpair every other device except for the ESP32 before turning either the ESP32 or the ELM327 or car on. Might even try a different OBD2 scanner. I've had ones go bad on me before.

zejulio, 2023-08-22:
Its strange that I can use torque without any issue, and dont work with ESP32...

zejulio, 2023-08-22:
Also when connecting it takes more than 10seconds to connect ESP to ELM, is this normal?

hawkeye1103, 2023-09-03:
Getting the same issue with the elm327 wired with the esp32 hardware serial. It flicks on searching and stopped and so on if i disconnect the elm327 from the car and plug it back in I get data instantly. Otherwise if I leave the elm327 working and unplug the microcontroller and plug it back in i'm back to searching and stopped. I am using it on a Honda civic 10th gen.

hawkeye1103, 2023-10-19:
any updates on this issue... is there a way to bypass it ?

DerKleinePunk, 2023-11-02:
@hawkeye1103 i understand that right the Problem ist disconnect and reconnect ?

hawkeye1103, 2023-11-27:
> @hawkeye1103 i understand that right the Problem ist disconnect and reconnect ?

The problem is it doesn't interpret the information until i unplug and replug the elm327 with the microcontroller still powered .... so it misses the connection start up protocol where it gives the commands: AT D, AT Z .... please tell me if I didn't explain clear I will try to explain differently.

closed 2025-02-09 by PowerBroker2