PowerBroker2/ELMduino #182: Pulling PID 119

themagicm, 2023-10-01:
I'm trying to pull pid 119 Absolute Barometric Pressure but get gibberish.

I've updated the .h file with:
const uint8_t INTAKE_AIR_TEMP                  = 119;

And left the rest of the code the same:
uint16_t ELM327::intakeAirTemp()
{
	return (uint16_t)processPID(SERVICE_01, INTAKE_AIR_TEMP, 1, 5);
}

but I get gibberish back.  What am I doing wrong?

PowerBroker2, 2023-10-02:
I don't know what your code is doing or what your car is responding with, so I can't help much.

themagicm, 2023-10-02:
Same here.  I dont know what its doing either.  LOL!    I get a value that looks out of range.  I'll run it in debug and see what it displays.

themagicm, 2023-10-02:
> I don't know what your code is doing or what your car is responding with, so I can't help much.

Here you go.  I pulled PID 119:

Service: 1
PID: 119
Normal length query detected
Query string: 01771
Clearing input serial buffer
Sending the following command/query: 01771
        Received char: 4
        Received char: 1
        Received char: 7
        Received char: 7
        Received char: 0
        Received char: 3
        Received char: 4
        Received char: F
        Received char: 4
        Received char: F
        Received char: 0
        Received char: 0
        Received char: 0
        Received char: 0
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 4177034F4F0000
Expected response header: 4177
Single response detected
64-bit response: 
        responseByte_0: 3
        responseByte_1: 0
        responseByte_2: 79
        responseByte_3: 79
        responseByte_4: 0
        responseByte_5: 0
        responseByte_6: 0
        responseByte_7: 0
=========================================================

and also 15 (default in elmduino).  I switched to 119 because I got a no data and assumed CACT was the correct PID.

Service: 1
PID: 15
Normal length query detected
Query string: 010F1
Clearing input serial buffer
Sending the following command/query: 010F1
        Received char: N
        Received char: O
        Received char: _
        Received char: D
        Received char: A
        Received char: T
        Received char: A
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: NODATA
ELM responded with errror "NO DATA"
Service: 1
PID: 11
Normal length query detected
Query string: 010B1
Clearing input serial buffer
Sending the following command/query: 010B1
        Received char: N
        Received char: O
        Received char: _
        Received char: D
        Received char: A
        Received char: T
        Received char: A
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: NODATA
ELM responded with errror "NO DATA"

PowerBroker2, 2023-10-02:
Why are you using 119 for intake air temp? Intake air temp is a standard PID and it's 15. If it returns "NO DATA", that means your car's diagnostics computer doesn't support the PID.

themagicm, 2023-10-02:
Using Torque Pro, my truck has IAT labelled as CACT which is Pid 119. Pid 15

-------- Original Message --------

PowerBroker2, 2023-10-03:
PID 119 isn't natively supported by this lib. You might be able to use some combination of `responseByte_0` through `responseByte_7` to reconstruct the desired output. I think you might be having the same issue as those in #141. I tried to do an update to fix that issue, but never tagged that version out due to lack of testing. The main branch has the updates that might help you. Try cloning main to your libraries folder and see if that fixes things. If not, you can mess with the responseBytes to get the right output if you want.

themagicm, 2023-10-03:
So you're saying I can force it to try 119 using main and changing INTAKE_AIR_TEMP to 119 to see if it works?

PowerBroker2, 2023-10-03:
I wouldn't do it that way, but maybe it would work

themagicm, 2023-10-03:
No it didnt work.  I copied over whats in src from main and it still didnt read 119 CACT.     I did make the change that was suggested by others that fixed their problem.

Data looked a tiny bit different.. instead of seeing 79 79 I get 79 89.    Is there something other than the wiki for OBD2 that gives more of a description of what 119 should return that way I can either write something myself or fix your code?

Service: 1
PID: 119
Normal length query detected
Query string: 01771
Clearing input serial buffer
Sending the following command/query: 01771
        Received char: 4
        Received char: 1
        Received char: 7
        Received char: 7
        Received char: 0
        Received char: 3
        Received char: 5
        Received char: 9
        Received char: 4
        Received char: F
        Received char: 0
        Received char: 0
        Received char: 0
        Received char: 0
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 417703594F0000
Expected response header: 4177
Single response detected
64-bit response: 
        responseByte_0: 3
        responseByte_1: 0
        responseByte_2: 79
        responseByte_3: 89
        responseByte_4: 0
        responseByte_5: 0
        responseByte_6: 0
        responseByte_7: 0

PowerBroker2, 2023-10-03:
There are apparently two sensors where the formula for each sensor is `value - 40 = temp` in degrees (celcius?). The value for the first sensor is the value of `responseByte_2` and the value of the other is the value of `responseByte_3`. [Source](https://racechrono.com/forum/discussion/2090/charge-air-cooler-temperature-cact)

themagicm, 2023-10-03:
Hmm.. so 79 to decimal is 121 - 40 = 81.  89 to decimal is 97.  I'll go see that torque pro reads on that.

PowerBroker2, 2023-10-03:
The 79 is decimal, not hex

themagicm, 2023-10-03:
Am I just missing it?

Sensor 1: t1 = B - 40 = 79 - 40 = 39°C

Sensor 2: t2 = C - 40 = 89 - 40 = 49°C

Torque Pro says 54*C

PowerBroker2, 2023-10-03:
Maybe your car is different and has a different temperature formula. I don't know what formula to use if it's not standardized. At this point you'll need to talk to someone in an automotive forum - I can't be of much more help

closed 2023-10-03 by PowerBroker2