PowerBroker2/ELMduino #184: Repeating pid request ( stopped car just turn on, esp32, elm327)

ham8962, 2023-10-03:
I am using your library with great appreciation (truly, thank you). However, as a beginner, I'm currently facing some difficulties.

My setup consists of an ESP32, ELM327, and a 2018 Hyundai Kona. The code I'm using is as follows:
`#include "BluetoothSerial.h"
#include "ELMduino.h"
/*
#if !defined(CONFIG_BT_SPP_ENABLED)
#error Serial Bluetooth not available or not enabled. It is only available for the ESP32 chip.
#endif
*/
BluetoothSerial SerialBT;
ELM327 myELM327;

uint32_t rpm = 0;
char CONVERTED_RPM[32];
char CONVERTED_KPH[32];
char CONVERTED_ENGINETEMP[32];
char CONVERTED_INTAKETEMP[32];

/*
String MACadd = "AA:BB:CC:11:22:33";
uint8_t address[6]  = {0xAA, 0xBB, 0xCC, 0x11, 0x22, 0x33};
//uint8_t address[6]  = {0x00, 0x1D, 0xA5, 0x02, 0xC3, 0x22};
*/
String name = "OBDII";
const char *pin = "1234"; //<- standard pin would be provided by default
bool connected;

void setup() {
Serial.begin(115200);
//SerialBT.setPin(pin);
SerialBT.begin("ESP32-test", true);
myELM327.begin(SerialBT, true, 2000);
//SerialBT.setPin(pin);
Serial.println("The device started in master mode, make sure remote BT device is on!");

// connect(address) is fast (upto 10 secs max), connect(name) is slow (upto 30 secs max) as it needs
// to resolve name to address first, but it allows to connect to different devices with the same name.
// Set CoreDebugLevel to Info to view devices bluetooth address and device names
connected = SerialBT.connect(name);
//connected = SerialBT.connect(address);

if(connected) {
Serial.println("Connected Succesfully!");
} else {
while(!SerialBT.connected(10000)) {
Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app.");
}
}
// disconnect() may take upto 10 secs max
/*
if (SerialBT.disconnect()) {
Serial.println("Disconnected Succesfully!");
}
*/
// this would reconnect to the name(will use address, if resolved) or address used with connect(name/address).
//SerialBT.connect();
}

void loop()
{

  float tempRPM = myELM327.rpm();
  
  if (myELM327.nb_rx_state == ELM_SUCCESS) 
  { 
    rpm = (uint32_t)tempRPM;
    sprintf(CONVERTED_RPM, "%d", rpm);
    Serial.print("RPM: "); Serial.println(CONVERTED_RPM);
  }
[... 234 lines cut ...]
04:41:09.944 -> Single response detected
04:41:09.944 -> 64-bit response:
04:41:09.944 -> 	responseByte_0: 0
04:41:09.944 -> 	responseByte_1: 0
04:41:09.944 -> 	responseByte_2: 0
04:41:09.944 -> 	responseByte_3: 0
04:41:09.944 -> 	responseByte_4: 0
04:41:09.944 -> 	responseByte_5: 0
04:41:09.944 -> 	responseByte_6: 0
04:41:09.944 -> 	responseByte_7: 0
04:41:09.944 -> Service: 1
04:41:09.944 -> PID: 13
04:41:09.944 -> Normal length query detected
04:41:09.990 -> Query string: 010D1
04:41:09.990 -> Clearing input serial buffer
04:41:09.990 -> Sending the following command/query: 010D1
04:41:09.990 -> RPM: 0
04:41:09.990 -> Km/h: 0
`

Although I understand that the km/h should read 0 since the vehicle is stationary, I don't understand why the PID 010D keeps repeating. Is this an issue with the ELM327?

I would appreciate it if you could explain the cause and provide a solution. Once again, thank you for creating this library and for your diligent responses.

(I am a student from Korea. Please excuse any awkwardness in my English expressions.)

PowerBroker2, 2023-10-03:
Firstly, your code isn't formatted correctly. Secondly, from what I can tell, your code isn't structured properly. [See the multiple pids example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/multiple_pids/multiple_pids.ino)

ham8962, 2023-10-03:
Thank you for your quick response. I received your answer just before falling asleep. (I didn't know there were multi-turn examples and ended up going to bed while contemplating other people's questions. It's 6 a.m. in Korea now, haha.) As soon as I wake up, I will try running the multi-turn request example you provided. I will share the results afterwards. Thank you sincerely.

ham8962, 2023-10-04:
I apologize for the delayed response.
Thanks to the example you provided, I successfully solved my problem.

One question I have is regarding the similarity between my initial incorrect code and some successful multi-PID implementations by other users. whether it's implemented or not Is it because of the difference in library versions?
Code posted by a user with the nickname neces0325 on page 4 of closed (sorry for the poor cause analysis since it's a beginner) (https://github.com/PowerBroker2/ELMduino/issues/53)

And is there a method that requires odometer?
If there is an odometer method, should I add Switch - CASE to the example of the module you gave me?
In the conditional statement outputting RPM, insert speed into the obd_state variable
In the conditional statement outputting SPEED, insert the odometer into the obd_state variable
In the  conditional statement outputting ODOMETER, insert the rpm into the  obd_state variable, like this way.

PowerBroker2, 2023-10-04:
> One question I have is regarding the similarity between my initial incorrect code and some successful multi-PID implementations by other users. whether it's implemented or not Is it because of the difference in library versions?

Yes, there was a large version update (I forget exactly when) that converted the PID query functions from synchronous to asynchronous and that changed the behavior/structure of the multi-PID example.

> And is there a method that requires odometer?
> I refer to the official document of wikipidia
odometer pid
> When requested PID by the Serial Bluetooth Terminal application, NODATA was output as a result.
> If there is an odometer method, should I add Switch - CASE to the example of the module you gave me?
> In the conditional statement outputting RPM, insert speed into the obd_state variable
> In the conditional statement outputting SPEED, insert the odometer into the obd_state variable
> In the conditional statement outputting ODOMETER, insert the rpm into the obd_state variable, like this way.

That sounds correct to me - sounds like a cool project - I hope you can get it working!

ham8962, 2023-10-04:
Thank you for answering to my question!!
I was waiting for your reply.(really thank you)

I'll try and find the odometer pid and share the results with you. 
Thank you for praising my project initiative. 
If you have any technical advice on my project, please let me know.

PowerBroker2, 2023-10-04:
No problem, I don't have any advice at the moment, but if you have questions or come up against roadblocks, the Arduino forum is really helpful if you post your entire code (in tags) along with a full description of your issue.

closed 2023-10-04 by PowerBroker2

DerKleinePunk, 2023-10-06:
@ham8962 If you have interest work together. I want save and show data on Rasperry PI with MQTT Protokoll.
https://github.com/DerKleinePunk/ODB-ESP32-Reader

ham8962, 2023-10-06:
@DerKleinePunk 
Thank you!
@PowerBroker2 
I received a "No DATA" response when requesting the Odometer PID through a Serial Terminal application on my phone 
Does it means that the Odometer data is not available for my car?

Regarding alternative methods, one idea I think is using a GPS module to track and record the distance traveled during each driving cycle

Is this a good method?

PowerBroker2, 2023-10-06:
"NO DATA" means your car doesn't support that pid. You could use a GPS module. Depending on how accurate you want the odometer function to be, you might consider using a Pixhawk controller to get more accurate displacement than a GPS module by itself.

ham8962, 2023-10-08:
@PowerBroker2 
Thank you!

DerKleinePunk, 2023-10-09:
@PowerBroker2 

Can you please explain what meen with Pixhawk controller. My Internet Search show Drone Controller.