PowerBroker2/ELMduino #186: How to use PIDs for EV

gonzob, 2023-10-23:
Hi. I have a MG ZS EV and I have the PID codes which I am successfully using in Torque. This shows that the codes are correct, and that the adapter works.

I want to build a ESP32 based device to communicate with the ECU via a AUTOHIL Bluetooth adapter. I have successfully communicated with the ECU using the ESP Bluetooth Serial example. Thanks for that!

I have non-standard PID codes like 0x22b041 (MG DC Bus Voltage), but I can't see how to use them. Can you help?

Gonzo

DerKleinePunk, 2023-11-02:
use queryPID(const uint8_t& service, const uint16_t& pid, const uint8_t& num_responses) ?

gonzob, 2023-11-02:
OK. Thanks
So, for 0x22b041, the service is 22, which I assume is hex, so I need to use 34 decimal for the service number. Yes?
The PID in hex is b041, so that's 45121 in decimal.Yes?
I have no idea how many responses I will get - if I over-specify will it time out?
Thanks in advance
Gonzo