PowerBroker2/ELMduino #187: Issue connecting at myELM327.begin

Hugues862, 2023-10-27:
Hello guys, I have a esp32, a elm327 and my car uses ISO 9141 10.4k baud (ISO_9141_5_BAUD_INIT).

`#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  // SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while (1)
      ;
  }

  if (!myELM327.begin(ELM_PORT, true, ISO_9141_5_BAUD_INIT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1)
      ;
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.throttle();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: ");
    Serial.println(rpm);
  }
  // delay(2000)
  // else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
  // myELM327.printError();
}`

When using ELM_PORT.begin, there isn't any problems and I can connect easily but then I can't connect using myELM327.begin.

`Clearing input serial buffer
Sending the following command/query: AT D
        Received char: \r
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
        Received char: A
        Received char: T
        Received char: _
        Received char: Z
        Received char: \r
        Received char: \r
        Received char: \r
        Received char: E
        Received char: L
        Received char: M
        Received char: 3
        Received char: 2
        Received char: 7
        Received char: _
        Received char: v
        Received char: 1
        Received char: .
        Received char: 5
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
        Received char: A
        Received char: T
        Received char: _
        Received char: E
        Received char: 0
        Received char: \r
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A3
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Connected to ELM327
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED`

Continuous STOPPED message.

Hugues862, 2023-10-27:
Response:
```Clearing input serial buffer
Sending the following command/query: AT D
        Received char: \r
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
        Received char: A
        Received char: T
        Received char: _
        Received char: Z
        Received char: \r
        Received char: \r
        Received char: \r
        Received char: E
        Received char: L
        Received char: M
        Received char: 3
        Received char: 2
        Received char: 7
        Received char: _
        Received char: v
        Received char: 1
        Received char: .
        Received char: 5
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
        Received char: A
        Received char: T
        Received char: _
        Received char: E
        Received char: 0
        Received char: \r
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A3
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Connected to ELM327
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED

PowerBroker2, 2023-10-27:
I'm not sure what's going wrong, but it seems like an ELM327 or car diagnostics computer issue. Was your car made 2008 or prior? That might be an issue. Also, what happens if you use [this example sketch](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino) to implement the PID queries manually?

Hugues862, 2023-10-27:
My car is a 2002 Mazda Miata that does not have can bus, but only iso9141 (Kline). When using the sketch provided I can get without an issue everything. 
Everything returns OK and when I query 0111 I can see the different hex values depending on the throttle position. So this works.

PowerBroker2, 2023-10-27:
If the manual program works ok, then that means main program might be initializing the ELM327 in a way that your car doesn't like. Mess with various initialization settings to see what works and what doesn't.

Hugues862, 2023-10-28:
Okay so I tried using the following:

This results in continuous STOPPED message
`myELM327.begin(ELM_PORT, true, ISO_9141_5_BAUD_INIT)`
`myELM327.begin(ELM_PORT, true, 2000, ISO_9141_5_BAUD_INIT)`

This results in alternating STOPPED  and SEARCHING... message
`myELM327.begin(ELM_PORT, true)`

Is there any other settings I could try ?

PowerBroker2, 2023-10-28:
What settings did you use with the manual input code?

Hugues862, 2023-10-29:
Here is my manual input code [esp32_test](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino)
As for the results here :
> AT  Z
ELM327 v1.5
> AT  E0
OK
> AT S0
OK
AT SP0
OK
>010C
410C11B7G...

Everything looks fine I think.

PowerBroker2, 2023-10-29:
Try implementing those commands in that order in your sketch after calling `begin()` using:

```C++
	sendCommand_Blocking(<see header file for argument to put here>);
	delay(100);
```

Hugues862, 2023-10-29:
So I guess this should be implemented in the first code sent on this issue ? 
Between these two begins ? 
`if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while (1)
      ;
  }

sendCommand_Blocking(/*see header file for argument to put here*/);
delay(100);

  if (!myELM327.begin(ELM_PORT, true, ISO_9141_5_BAUD_INIT))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1)
      ;
  }`

Hugues862, 2023-10-29:
Here is the current code:
```c++
void loop()
{
  myELM327.sendCommand_Blocking("0111");
  delay(200);
  float tempThrottle = myELM327.throttle();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    throttle = (uint32_t)tempThrottle;
    Serial.print("Throttle: ");
    Serial.println(throttle);
  }
  // delay(2000)
  // else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
  // myELM327.printError();
}
```
And the response:
```
Clearing input serial buffer
Sending the following command/query: AT D
        Received char: \r
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT Z
        Received char: A
        Received char: T
        Received char: _
        Received char: Z
        Received char: \r
        Received char: \r
        Received char: \r
        Received char: E
        Received char: L
        Received char: M
        Received char: 3
        Received char: 2
        Received char: 7
        Received char: _
        Received char: v
        Received char: 1
        Received char: .
        Received char: 5
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATZELM327v1.5
Clearing input serial buffer
Sending the following command/query: AT E0
        Received char: A
        Received char: T
        Received char: _
        Received char: E
        Received char: 0
        Received char: \r
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATE0OK
Clearing input serial buffer
Sending the following command/query: AT S0
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT AL
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT TP A3
        Received char: O
        Received char: K
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: OK
Connected to ELM327
Clearing input serial buffer
Sending the following command/query: 0111
Timeout detected with overflow of 0ms
Service: 1
PID: 17
Normal length query detected
Query string: 01111
Clearing input serial buffer
Sending the following command/query: 01111
Clearing input serial buffer
Sending the following command/query: 0111
        Received char: S
        Received char: T
        Received char: O
        Received char: P
        Received char: P
        Received char: E
        Received char: D
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: STOPPED
ELM responded with errror "STOPPED"
Clearing input serial buffer
Sending the following command/query: 0111
        Received char: 4
        Received char: 1
        Received char: 1
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 411100
Clearing input serial buffer
Sending the following command/query: 0111
        Received char: 4
        Received char: 1
        Received char: 1
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 411100
Clearing input serial buffer
Sending the following command/query: 0111
        Received char: 4
        Received char: 1
        Received char: 1
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 411100
Clearing input serial buffer
Sending the following command/query: 0111
        Received char: 4
        Received char: 1
        Received char: 1
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 411100
Clearing input serial buffer
Sending the following command/query: 0111
        Received char: 4
        Received char: 1
        Received char: 1
        Received char: 1
        Received char: 0
        Received char: 0
        Received char: \r
```

Hugues862, 2023-10-29:
Also what I find weird is that when querying throttle using .throttle() the query is `01111` (there's a extra 1)

PowerBroker2, 2023-10-29:
Only use the blocking commands for initialization within `setup()`, not for querying in `loop()`. You can use hardcoded strings as arguments if you'd like, but you can also use these premade commands (might need `sprintf()` to fill some fields, but still):

https://github.com/PowerBroker2/ELMduino/blob/636aa957f6f7d6ff01bf41fbfce004c1efc5883c/src/ELMduino.h#L145-L253

Hugues862, 2023-10-29:
So basically with the setup part I'm not having any issue, I can connect without an issue, but when using the .throttle() or .rpm. I have misunderstood the error, if you look at the first logs I sent, we can see that myELM327.begin works, but then I receive STOPPED messages when querying .throttle and .rpm in the loop.
I will rename this thread, to prevent more confusion.

closed 2024-01-15 by Hugues862

Bakar002, 2024-04-15:
does. your error solved?

Bakar002, 2024-04-15:
I have the same issue like when i use multiple PID's it shows the same error