PowerBroker2/ELMduino #188: ELM327 Bluetooth don't connection

AndrePelle, 2023-10-30:
HI,
I'm trying to connect my esp32 with an ELM327 but I can't, I tried with the two example codes, both ESP32_Bluetooth_Serial and ESP32_test but it always gives me "Couldn't connect to OBD scanner". I unpaired my phone with the ELM 327 and I also tried with 38400 baud instead. Can you help me?
Thank you!

[image]

PowerBroker2, 2023-10-30:
I can't help without knowing your code and debug prints. Either way, using the Arduino Forum is a much better place for questions. GitHub issues should be used for bugs or code enhancements. You'll get faster and better help on the forum.

AndrePelle, 2023-11-04:
My code i this: 

```c
/*
To test connection, type the following into the serial monitor:
  AT Z
  AT E0
  AT S0
  AT AL
  AT TP A0
*/

#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

#define DEBUG_PORT Serial
#define ELM_PORT   SerialBT

void setup()
{
  //pinMode(LED_BUILTIN, OUTPUT);
  //digitalWrite(LED_BUILTIN, HIGH);

  DEBUG_PORT.begin(115200);
  ELM_PORT.begin("ESP32test", true);
  //ELM_PORT.setPin("1234");

  DEBUG_PORT.println("Attempting to connect to ELM327...");

  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner");
    while(1);
  }

  DEBUG_PORT.println("Connected to ELM327");
  DEBUG_PORT.println("Ensure your serial monitor line ending is set to 'Carriage Return'");
  DEBUG_PORT.println("Type and send commands/queries to your ELM327 through the serial monitor");
  DEBUG_PORT.println();
}

void loop()
{
  if(DEBUG_PORT.available())
  {
    char c = DEBUG_PORT.read();

    DEBUG_PORT.write(c);
    ELM_PORT.write(c);
  }

  if(ELM_PORT.available())
  {
    char c = ELM_PORT.read();

    if(c == '>')
      DEBUG_PORT.println();

    DEBUG_PORT.write(c);
  }
}
``` 
My board is: ESP-WROOM-32
When I press the reset button my debug prints is this:
[image]

PowerBroker2, 2023-11-04:
Your ESP32 is not connecting to the bluetooth of your ELM327. What is your ELM327 device name? Set that name in your sketch for `ELM_PORT.connect()`

AndrePelle, 2023-11-04:
Whenever I search for my device using the phone, it's named OBDII, as in my sketch. However, when I connect with the phone, I need to enter a password (1234). I tried enabling `ELM_PORT.setPin("1234")` but it doesn't connect.

PowerBroker2, 2023-11-04:
If the set pin function doesn't work, I'm not sure how to help

AndrePelle, 2023-11-04:
Do you have any other tips to check why I can't connect?

jimwhitelaw, 2023-12-03:
You might just be timing out making the connection. Perhaps you could try using your phone app to get the BT address of your ELM device and call ELM_PORT.connect() using that instead of the device name "OBDII". Connecting via address usually takes < 5 sec. 

The version string you got back from the device is the ELM software version, not OBDII version. That's a legit version number, but with all the clones out there, it's impossible to know if that's the actual version of the firmware; odds are it's a v1.0 clone falsely reporting v2.1. It will probably still work fine for most things.

jimwhitelaw, 2023-12-04:
OK, so that seems like some progress, you're connecting to the ELM device now. The ELM327 data sheet (p. 72) says this about the ? response: _"This is the standard response to a misunderstood command on the RS232 input."_ So it might be possible that your adapter is not supporting the AT commands. I would try some other commands, like the "0100" that should be supported on all vehicles and see if you get any valid responses. If that's a no-go, double check the serial connection parameters like the line end setting etc.

AndrePelle, 2023-12-05:
Hi, Thank you! 
I checked with '0100' and the response is 'UNABLE TO CONNECT' it seems that the connection message is false, do you know what this could be due to?

[image]

jimwhitelaw, 2023-12-05:
That message means that the ELM327 can't establish a connection with the ECU - it can't determine the correct protocol. The "SEARCHING..." message is what ELM returns while it is doing a protocol search. Do you know what protocol you need? If so, you could try issuing command "AT SP Ah'"  where _h_ is the number for your desired protocol. (See p.25 of the data sheet for the list). My guess is that probably won't work since your device isn't responding to AT commands. Do you have a second adapter you can try? 

At this point, since you have a valid BluetoothSerial connection to your device, it might be easier to switch to using the ELMduino methods to query your ECU. Use code to send commands and use the serial monitor to view the debug output from ELMduino (enable debug info in your _begin()_ method).

AndrePelle, 2023-12-06:
Hi, sorry but I don't know ELM327 that well, what do you mean which protocol should I communicate with?
However, I will explain to you what I would like to do, my goal is to understand when my car is regenerating the DPF, so it could be clearer what I want to do.

Thanks for your patience and help

jimwhitelaw, 2023-12-06:
The protocol I'm referring to is the protocol that your vehicle uses. The OBD standard allows for several different vehicle protocols as does the ELM327. Usually, having the ELM327 do an auto search for the correct protocol works fine but it's not working for you. It appears that it will be beneficial to know what protocol your vehicle uses.

Here's a good primer on OBDII systems and terminology: [https://www.csselectronics.com/pages/obd2-explained-simple-intro](url)

You should get familiar with the [ELM327 data sheet](https://github.com/PowerBroker2/ELMduino/blob/master/reference/ELM327DS.pdf). Besides being the canonical reference for ELM327 devices, it goes into a lot of detail about OBD, CAN and the various inter-communications. 

As for your specific use case, I am not aware of any standard OBD commands (PIDs) related to DPF generation; they are very likely specific to your vehicle/manufacturer. You will have to find a reference to those commands (mode and PID) and then write your own custom code to use them specifically. Here is info on how to query custom PIDs: [https://github.com/PowerBroker2/ELMduino/issues/145#issuecomment-1289350326](url)

closed 2023-12-21 by PowerBroker2