PowerBroker2/ELMduino #189: Arduino obd monitoring Opel Insignia

MadaStanescu, 2023-10-31:
Hello, I also want to make a module like the one in the attached video( https://jakubdziworski.github.io/obd/bluetooth/arduino/car/2020/07/13/arduino-obd-dpf-monitor-insignia.html) I have the same components as in the arduino nano video presentation, BT module, IC2 display, I have a vgate vLink MC+ interface, but I encountered a problem. I downloaded the code from linck (  which is for opel insignia) but it gives me error of ic2, elm etc. I am not an expert in arduino but I have written codes and connected modules before. My question is what am I doing wrong?  If you can help me. Thanks in advance.

MadaStanescu, 2023-11-04:
When I want to write the code, it gives me the error below.

C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino: In function 'void connect()':
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:51:37: error: invalid operands of types 'void' and 'const int8_t {aka const signed char}' to binary 'operator!='
   if(vgate.sendCommand("AT SH 7E0") != ELM_SUCCESS) {
      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino: In function 'int32_t queryVgate(uint8_t, uint16_t)':
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:110:15: error: 'class ELM327' has no member named 'status'
     if (vgate.status == ELM_SUCCESS)
               ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:115:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_NO_RESPONSE)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:117:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_BUFFER_OVERFLOW)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:119:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_GARBAGE)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:121:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_UNABLE_TO_CONNECT)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:123:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_NO_DATA)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:125:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_STOPPED)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:127:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_TIMEOUT)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:129:20: error: 'class ELM327' has no member named 'status'
     else if (vgate.status == ELM_GENERAL_ERROR)
                    ^~~~~~
C:\Users\Eagle\Desktop\insignia-dpf-monitor-master\insignia-dpf-monitor-master\insignia-dpf-monitor\insignia-dpf-monitor.ino:133:28: error: 'class ELM327' has no member named 'status'
       Serial.println(vgate.status);
                            ^~~~~~

exit status 1

Compilation error: invalid operands of types 'void' and 'const int8_t {aka const signed char}' to binary 'operator!='

MadaStanescu, 2023-11-04:
Code
#include <Wire.h>
#include <LiquidCrystal_I2C.h>
#include <SoftwareSerial.h>
#include "ELMduino.h"

LiquidCrystal_I2C lcd(0x27,20,4);  // set the LCD address to 0x27 for a 16 chars and 2 line display
ELM327 vgate;
SoftwareSerial vgateSerial(10, 7);

const long debugRate = 115200;

void setup()
{
  Serial.begin(debugRate);
  Serial.println("START");
  setupBt();
  lcd.init();
  lcd.backlight();
}

void loop()
{
  if(isBtConfBtnPressed()) {
    pairWithVgate();
  }
  if(!vgate.connected) {
    connect();
    delay(1000);
    return;
  }
  int32_t regenStatus = getRegenerationStatus();
  if(regenStatus > 0) {
    printRegenerating(regenStatus);
    return;
  }
  printDpfStatus();
  delay(1000);
}

void connect() {
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("CONNECTING...");
  lcd.setCursor(0,1);
  if (!vgate.begin(vgateSerial))
  {
    Serial.println("Couldn't connect to OBD scanner");
    lcd.print("VGATE UNCONNECTED");
    return;
  }
  if(vgate.sendCommand("AT SH 7E0") != ELM_SUCCESS) {
    Serial.println("Unable to set header to 7E0");
    lcd.print("UNABLE TO SET AT SH 7E0");
    return;
  }
  lcd.print("CONNECTED");
  Serial.println("Connected to OBD scanner");
}

void printRegenerating(int32_t regenStatus) {
  int32_t percentRegenerated = map(regenStatus, 0, 255, 0, 100);
  String statusMessage = "      ";
  statusMessage = statusMessage + percentRegenerated + "%      ";
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print("!!REGENERATING!!");
  lcd.setCursor(0,1);
  lcd.print(statusMessage);
  lcd.noBacklight();
  delay(200);
  lcd.backlight();
  delay(200);
}

void printDpfStatus() {
  int32_t kmsSinceDpf = getKmsSinceDpf();
  int32_t dirtLevel = getDpfDirtLevel();
  lcd.clear();
  lcd.setCursor(0,0);
  String message = "LAST: ";
  message = message + kmsSinceDpf + "KM";
  Serial.println(message);
  lcd.print(message);
  lcd.setCursor(0,1);
  message = "FILL: ";
  message = message + dirtLevel + "%";
  lcd.print(message);
}

int32_t getRegenerationStatus() {
  return queryVgate(0x22, 0x3274);
}

int32_t getKmsSinceDpf() {
  return queryVgate(0x22, 0x3277);
}

int32_t getDpfDirtLevel() {
  return queryVgate(0x22, 0x3275);
}

int32_t queryVgate(uint8_t service, uint16_t pid) {
  Serial.print("Response for ");
  Serial.print(pid);
  Serial.print(": ");
  if (vgate.queryPID(service, pid))
  {
    int32_t response = vgate.findResponse();
    if (vgate.status == ELM_SUCCESS)
    {
      Serial.print("SUCCESS:"); Serial.println(response);
      return response;
    }
    else if (vgate.status == ELM_NO_RESPONSE)
      Serial.println("ERROR: ELM_NO_RESPONSE");
    else if (vgate.status == ELM_BUFFER_OVERFLOW)
      Serial.println("ERROR: ELM_BUFFER_OVERFLOW");
    else if (vgate.status == ELM_GARBAGE)
      Serial.println("ERROR: ELM_GARBAGE");
    else if (vgate.status == ELM_UNABLE_TO_CONNECT)
      Serial.println("ERROR: ELM_UNABLE_TO_CONNECT");
    else if (vgate.status == ELM_NO_DATA)
      Serial.println("ERROR: ELM_NO_DATA");
    else if (vgate.status == ELM_STOPPED)
      Serial.println("ERROR: ELM_STOPPED");
    else if (vgate.status == ELM_TIMEOUT)
      Serial.println("ERROR: ELM_TIMEOUT");
    else if (vgate.status == ELM_GENERAL_ERROR)
      Serial.println("ERROR: ELM_GENERAL_ERROR");
    else {
      Serial.print("ERROR: UNKNOWN ELM STATUS:");
      Serial.println(vgate.status);
    }
  }
  else {
    Serial.println("NOT CONNECTED");
  }
  return -1;
}

//##### BLUETOOTH AUTOCONFIGURATION SECTION #########
const int btConfPin = 3 ;
const int btConfBtn = 4;
const int btPowerPin = 5;
const long vgateRate = 9600;
const long confRate = 38400;

void setupBt() {
  vgateSerial.begin(vgateRate);
  pinMode(btPowerPin, OUTPUT);
  pinMode(btConfPin, OUTPUT);
  pinMode(btConfBtn, INPUT_PULLUP);
  digitalWrite(btPowerPin, HIGH);
  digitalWrite(btConfPin, LOW);
}

boolean isBtConfBtnPressed() {
  Serial.println(digitalRead(btConfBtn));
  return digitalRead(btConfBtn) == LOW;
}

//##### AUTOCONFIGURATION SECTION #########
void pairWithVgate()
{
    lcd.clear();
    vgateSerial.end();
    vgateSerial.begin(confRate);

    digitalWrite(btPowerPin, LOW);
    digitalWrite(btConfPin, HIGH);
    delay(1000);
    digitalWrite(btPowerPin, HIGH);
    delay(4000);

    sendBtConfCommand("AT+RMAAD");
    sendBtConfCommand("AT+ORGL");
    sendBtConfCommand("AT+ROLE=1");
    sendBtConfCommand("AT+RESET",4000);
    sendBtConfCommand("AT+CMODE=0");
    sendBtConfCommand("AT+BIND=86DC,3D,ABF7F1");
    sendBtConfCommand("AT+INIT");
    sendBtConfCommand("AT+PAIR=86DC,3D,ABF7F1,20",10000L);
    sendBtConfCommand("AT+LINK=86DC,3D,ABF7F1",10000L);
    Serial.println("Done");
    lcd.clear();
    lcd.setCursor(0,0);
    lcd.print("BLUETOOTH CONF");
    lcd.setCursor(0,1);
    lcd.print("DONE");
    delay(3000);
    vgateSerial.end();
    vgateSerial.begin(vgateRate);

    digitalWrite(btPowerPin, LOW);
    digitalWrite(btConfPin, LOW);
    delay(1000);
    digitalWrite(btPowerPin, HIGH);
    lcd.clear();
}

void sendBtConfCommand(char* cmd) {
  sendBtConfCommand(cmd, 2000);
}

void sendBtConfCommand(char* cmd, long timeout) {
  lcd.clear();
  lcd.setCursor(0,0);
  lcd.print(cmd);
  lcd.setCursor(0,1);
  Serial.println(cmd);
  long start = millis();
  vgateSerial.println(cmd);

  while((millis() - start) < timeout) {
    if (vgateSerial.available()) {
      char c = vgateSerial.read();
      lcd.print(c);
      Serial.write(c);
    }
    delay(5);
  }
  lcd.clear();
}

What to do?

PowerBroker2, 2023-11-04:
That sketch is using an old version of the library. You can try either updating the sketch code to get things to work (see the current examples) or you can try to find an older version of the library that allows your code to compile.

MadaStanescu, 2023-11-04:
> That sketch is using an old version of the library. You can try either updating the sketch code to get things to work (see the current examples) or you can try to find an older version of the library that allows your code to compile.

I'm not an expert in arduino, I only know some elementary things. Can you guide me step by step in solving this problem?  Thanks in advance.

PowerBroker2, 2023-11-04:
In the Arduino IDE, you can go to the libraries manager, search for ELMDuino, select the next lower version, select install/update, and recompile. Continue choosing a next lower version until the recompile is successful. That would be the easiest fix.

MadaStanescu, 2023-11-04:
> In the Arduino IDE, you can go to the libraries manager, search for ELMDuino, select the next lower version, select install/update, and recompile. Continue choosing a next lower version until the recompile is successful. That would be the easiest fix.

Thanks for the suggestion. I'll try. If I have problems, I'll come back with details.

MadaStanescu, 2023-11-05:
> In the Arduino IDE, you can go to the libraries manager, search for ELMDuino, select the next lower version, select install/update, and recompile. Continue choosing a next lower version until the recompile is successful. That would be the easiest fix.

Hi, I was able to write the code with the ELMDuino version 2.6.4. Thanks for the suggestion. I'm going to test it on the car to see if it works. Have a nice day.

MadaStanescu, 2023-11-06:
> In the Arduino IDE, you can go to the libraries manager, search for ELMDuino, select the next lower version, select install/update, and recompile. Continue choosing a next lower version until the recompile is successful. That would be the easiest fix.

It works, thanks for the support. Have a nice day.

closed 2023-11-06 by PowerBroker2