PowerBroker2/ELMduino #190: Connection with OBDII adapter not possible

Beutelleufel, 2023-11-09:
I would like to use the library to read out live data from my car (e.g. speed, rpm).
I have the following adapter: [Vgate iCar Pro Bluetooth](https://www.ebay.de/itm/374095470898)

I can neither connect with the name "IOS-Vlink" nor with the address.
I just get the message that the connection failed.
I took the Esp32 example in each case.

Maybe you can help me. Thanks in advance

```
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

//Advertised Device: Name: IOS-Vlink, Address: d2:e0:2f:8d:54:d5, serviceUUID: 000018f0-0000-1000-8000-00805f9b34fb
//String MACadd = "D2:E0:2F:8D:54:D5";                       //enter the ELM327 MAC address
uint8_t address[6]  = {0xD2, 0xE0, 0x2F, 0x8D, 0x54, 0xD5};  //enter the ELM327 MAC address after the 0x

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect(address))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

and

```
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("IOS-Vlink"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

mattoide, 2023-11-09:
https://github.com/PowerBroker2/ELMduino/issues/183#issuecomment-1760304414

PowerBroker2, 2023-11-28:
BLE is not supported

closed 2023-11-28 by PowerBroker2