PowerBroker2/ELMduino #191: connection to Ircama/ELM327-emulator or PowerBroker2/ELM_Simulator

mattoide, 2023-11-09:
Hi, first of all thanks for your amazing work!

I'm tired to go in and out my car for testing, then i found these 2 ELM327 emulator:

[https://github.com/Ircama/ELM327-emulator](url)
[https://github.com/PowerBroker2/ELM_Simulator](url)

For **Ircama/ELM327-emulator** how i can connect via serial usb?

Then for **PowerBroker2/ELM_Simulator** when it ask for arduino serial port, and i give him my arduino serial port, it responds with:

` python3 ELM_Simulator.py 
Specify Arduino Serial Port: /dev/ttyUSB0
[Errno 16] could not open port /dev/ttyUSB0: [Errno 16] Device or resource busy: '/dev/ttyUSB0'
`

any help is appreciated (:

PowerBroker2, 2023-11-09:
> For Ircama/ELM327-emulator how i can connect via serial usb?

Idk, maybe.

> [Errno 16] could not open port /dev/ttyUSB0: [Errno 16] Device or resource busy: '/dev/ttyUSB0'

Looks like you have another app using the Arduino's USB port. You need to stop all processes that use the Arduino USB before running the sim.

mattoide, 2023-11-09:
> Idk, maybe.
>
>
> Looks like you have another app using the Arduino's USB port. You need to stop all processes that use the Arduino USB before running the sim.

I found the problem, it was serial monitor opened! After many tries i understood that the ride is:

- stop any emulator on the arduino port on any serial port
- upload sketch
- start emulator

 But there is a problem with line 58 of emulator:

`                            pid     = int(query[2:],  16)
`

when i call **myELM327.rpm();** the emulator take wrong pid.

This is the log:

```
Query: 010C1
Pid: 193
```

Regarding the /Ircama/ELM327-emulator the problem is the **Serial.print** in ELMduino lib that make the emulator crazy, i'm working on a solution

mattoide, 2023-11-13:
> Regarding the /Ircama/ELM327-emulator the problem is the **Serial.print** in ELMduino lib that make the emulator crazy, i'm working on a solution

Solved by connecting Arduino Uno connected via serial port to emulator, to another Arduino Nano via serial pins, connected to another usb port for showing messages on serial port without send messages to emulator #
:hot_face:

closed 2023-11-13 by PowerBroker2