PowerBroker2/ELMduino #192: Checking Parking Brake Status

edwardcorke, 2023-11-13:
Hi there, does this library provide a way to check the status of the parking brake, please?

Thanks for an awesome library!

PowerBroker2, 2023-11-13:
Thanks! There might be a custom PID for that for your car, but it's not inherently supported. I doubt there are even custom PIDs for this except maybe for very modern cars considering how mechanical the parking brake is.

edwardcorke, 2023-11-13:
Thanks for your quick response. I'll do some digging. There's a chance as an icon appears on the dashboard when the parking brake is applied (some electric connection). It's a 2010 Ford Fiesta. Do you have any suggestions on how I could find this; I think it's proprietary information.

PowerBroker2, 2023-11-14:
Sorry, I have no idea except snooping on Ford forums

closed 2023-11-14 by PowerBroker2