PowerBroker2/ELMduino #198: queryPID never returns

gonzob, 2023-12-09:
Hi.

I have set up a NodMCU322 to query my AUTOHIL Bluetooth adapter on my MG ZS EV.

It connects properly and can return standard PID values (rpm, etc). Form this I'm happy that the code is correct.

However, because my vehicle is an EV I'm after the non-standard PIDs. I have previously set up Torque to read these values, so I'm certain I have the correct PIDs.  The problem is that when I use queryPID() it stays in the ELM_GETTING_MSG loop and never returns with either a value or an error.

Here's an example of trying to get  22b041 (DC Bus Voltage)

DCBUSV = myELM327.queryPID(34,45121,128);

I have no idea what I'm going to get back, so I set the number of responses high (128).

The same happens with
DCBUSV = myELM327.queryPID("22b041");

What am I doing wrong?

Gonzo

PowerBroker2, 2023-12-09:
Code and debug prints?

gonzob, 2023-12-10:
OK, the loop code is here:

```c++
void loop() {
  switch (obd_state) {
    case IDLE:
      // Schedule PIDs to be read at different rates (slow PIDs and fast PIDs)
      if (millis() > slow_data_last_time + 5000) {
        slow_data_last_time = millis();
        obd_state = DC_BUS_V;
        DISP("Chk DC BUS");
        Serial.printf("IDLE, query SLOW PIDs at %.3f s\n", slow_data_last_time / 1000.0);
      } else if (millis() > fast_data_last_time + 1000) {
        fast_data_last_time = millis();
        // Specify first PID here
        Serial.printf("IDLE, query FAST PIDs at %.3f s\n", fast_data_last_time / 1000.0);
      } else {
        // Serial.printf("[%.3f s] IDLE\n", millis() / 1000.0);
      }
      break;

    case DC_BUS_V:
      DCBUSV = myELM327.queryPID(34,45121,128);
      if (myELM327.nb_rx_state == ELM_SUCCESS) {
        DISP("DC BUS OK");

        // retrieve value
        // and do something
        // with it here
 
        obd_state = IDLE;
        
      } else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
        myELM327.printError();
        DISP("DC BUS Fail");
        obd_state = IDLE ;
      }
      break;

  }
}
``` 
The DISP() command is a routine to write info to a small LED display.

The code runs, connects with the BT adapter and sends the query, but the return state remains ELM_GETTING MSG. It never succeeds, times out, or fails.

```
09:45:39.955 -> Clearing input serial buffer
09:45:39.955 -> Sending the following command/query: AT TP A0
09:45:39.988 -> 	Received char: O
09:45:39.988 -> 	Received char: K
09:45:39.988 -> 	Received char: \r
09:45:39.988 -> 	Received char: \r
09:45:39.988 -> 	Received char: >
09:45:39.988 -> Delimiter found.
09:45:39.988 -> All chars received: OK
09:45:39.988 -> Phase 2 ok
09:45:40.022 -> IDLE, query SLOW PIDs at 29.713 s
09:45:40.022 -> Service: 34
09:45:40.055 -> PID: 45121
09:45:40.055 -> Long query detected
09:45:40.055 -> Query string: 22B041⸮
09:45:40.055 -> Clearing input serial buffer
09:45:40.055 -> Sending the following command/query: 22B041⸮
09:45:40.055 -> Service: 34
09:45:40.055 -> PID: 45121
09:45:40.055 -> Long query detected
09:45:40.055 -> Query string: 22B041⸮
09:45:40.055 -> Clearing input serial buffer
09:45:40.055 -> Sending the following command/query: 22B041⸮
09:45:40.055 -> Service: 34
09:45:40.055 -> PID: 45121
09:45:40.055 -> Long query detected
09:45:40.055 -> Query string: 22B041⸮
09:45:40.055 -> Clearing input serial buffer
09:45:40.088 -> Sending the following command/query: 22B041⸮
09:45:40.088 -> Service: 34

```
The query just keeps repeating (as it should)

I haven't attempted to get the data yet, all I'm trying to do is get a positive response. What am I doing wrong?

Gonzo

PowerBroker2, 2023-12-10:
I think there's a bug in `formatQueryArray()` which is called by `queryPID()`:

https://github.com/PowerBroker2/ELMduino/blob/35f8ead2afc96d4167e76a8a6f63905ca0a0ca3b/src/ELMduino.cpp#L215

The (incorrect) assumption was that the number of expected responses would always be less than 10, thus making the decimal to ASCII conversion super easy. I will have to fix when I get the chance, however, if you or anyone else wants to beat me to it with a PR, I'd be happy with that.

gonzob, 2023-12-10:
Thanks.
I can see that would be a problem for 128, but I've tried 1, 2, and 5 for the number of responses (which should have parsed OK), and the same continuous loop happens. So, if the actual number of received responses exceeds the number in the query, what happens? Does it never update?

I'll have a go at updating the library and see what happens.

Gonzo

PowerBroker2, 2023-12-13:
Fixed the query formatting issue in this commit: 7768476

While fixing the above issue I realized you will have more success when using `processPID()` **instead of** `queryPID()`:

https://github.com/PowerBroker2/ELMduino/blob/190145950f3588407303ea9f59ffde745bd7d2ab/src/ELMduino.cpp#L553-L598

gonzob, 2023-12-13:
Hi.

Thanks for that. However, I have since discovered that the number of responses refers to the number of lines received (I think) and that all of my queries result in only one line, so I think your earlier version would have been fine.

The reason the queryPID never returned was that, unlike the other commands, with queryPID if you call it repeatedly it resets the state of the query. You need to use queryPID once, then call get_response() repeatedly to force the state change.

Now I know more about the responses to the queries, I can use processPID with confidence.

I will post results when I have something more concrete.

Gonzo

jimwhitelaw, 2023-12-16:
Looks like you've sorted out that your initial infinite query loop was due to using queryPID instead of processPID. 

Regarding multiple responses, in the case where you actually receive more responses than indicated by num_responses, ELMduino will happily gather them all up, but it's up to you to parse them out of the response. ELMduino only passes that value on to the ELM device to inform the device of the number of expected responses and then ignores it when receiving data.

closed 2023-12-16 by PowerBroker2