PowerBroker2/ELMduino #199: Use own pid

Faceonlook, 2023-12-10:
Hello i try to use my own pid please somebody can help me.

Pid:22028C
Header:7E0
for my e-golf

Faceonlook, 2023-12-10:
Does this look good?

`if (myELM327.queryPID("22028C")) {
       response = myELM327.findResponse();  
       Serial.print("Response: "); Serial.println(response);
`

PowerBroker2, 2023-12-10:
That might work if that's the only PID you're querying. Also note there seems to be a bug in `queryPID()` that is being actively worked (#198)

Faceonlook, 2023-12-10:
Okay, I'll test this, thanks. Can you also please tell me how to set the header correctly?

PowerBroker2, 2023-12-13:
I'm not familiar with setting headers. You might want to check the datasheet in the reference folder in this library.

Also, see this: https://github.com/PowerBroker2/ELMduino/issues/198#issuecomment-1853198890

closed 2023-12-21 by PowerBroker2