PowerBroker2/ELMduino #203: batteryVoltage() returns zero

themelisx, 2023-12-16:
I am testing your library, since now works fine at all PIDs (1_20) that I need, except the battery voltage.
The function returns 0 but at the debug log I can see 13.1 Volts
I am using like this:
int newValue = int(obd->batteryVoltage() * 10);
so I am wating to get 131 

Query ODB value
Clearing input serial buffer
Sending the following command/query: AT RV
        Received char: A
        Received char: T
        Received char: _
        Received char: R
        Received char: V
        Received char: \r
        Received char: 1
        Received char: 3
        Received char: .
        Received char: 1
        Received char: V
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: ATRV13.1V

PowerBroker2, 2023-12-16:
Can you post all debug statements? Can you also add another debug statement that prints the contents of `payload` before being processed by `strtod()`?:

https://github.com/PowerBroker2/ELMduino/blob/190145950f3588407303ea9f59ffde745bd7d2ab/src/ELMduino.cpp#L2474-L2484

@jimwhitelaw Can you take a quick look at this issue? Thanks!

jimwhitelaw, 2023-12-16:
Submitted a PR #204 

Payload has "V" char at the end that causes strtod to return 0.

closed 2023-12-21 by PowerBroker2