PowerBroker2/ELMduino #206: Adapter version

themelisx, 2023-12-17:
It will be great if you add a function that will return a (float) number of the adapter version after the device is connected.
This will allow to automatic enable some PID requests that depends on the adapter version

jimwhitelaw, 2023-12-17:
1. You can get the adapter version with the AT Z and AT I commands. However, many (most?) of the clone adapters return an incorrect version. I have a couple that return V1.5 which never existed! 
2. There aren’t any restrictions on which PIDs are available based on adapter versions. ELMduino/ELM327 will happily send any PID command you like and it’s up to the ECU(s) to determine if the PID is supported. 
3. The best way to check if a PID is supported is by using one of the SUPPORTED_PID commands. But parsing out the response is a pain, so the easy way is just to send the PID command. If it is not supported, you will receive “NO DATA” response.

themelisx, 2023-12-17:
@jimwhitelaw thank you, got it

PowerBroker2, 2023-12-17:
I agree with Jim on this one - closing for now

closed 2023-12-17 by PowerBroker2