PowerBroker2/ELMduino #207: Help with Volkswagen MEB (ID.3, Skoda Enyaq etc.) needed

Waldmensch1, 2023-12-17:
Hello, I try since a few days reading the Battery Temperature out of my Enyac. The ELM init works fine and I have a connection. Unfortunately I get not set the appropriate header and the PIDs doesnt respond as well. Maybe someone else has already "hacked" an ID.3 with ELMDuino and is able to give me a hint

I found the header and PIDs in this project under BMS https://github.com/nickn17/evDash/blob/master/src/CarVWID3.cpp

The Header I try to set with
```
char header[20] = "\0";
    sprintf(header, SET_HEADER, "17FC007B");

    if (myELM327.sendCommand_Blocking(header) == ELM_SUCCESS)
    {
      tft.println("Header 17FC007B okay");
      cmd = SOC;
    }
    else
    {
      myELM327.printError();
    }
```
with this response:

```
Clearing input serial buffer
Sending the following command/query: AT SH 17FC007B
        Received char: ?
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received:
```
My whole loop, where you can see that I try with different approaches (processPid and queryPid) but nothing comes back at all

```
void loop()
{
  button_loop();

  switch (cmd)
  {
  case CMD1:
  {
    tft.println("try set header");
    char header[20] = "\0";
    sprintf(header, SET_HEADER, "17FC007B");

    if (myELM327.sendCommand_Blocking(header) == ELM_SUCCESS)
    {
      tft.println("Header 17FC007B okay");
      cmd = SOC;
    }
    else
    {
      myELM327.printError();
    }
    cmd = TEMP;
  }

  case TEMP:
  {
    tft.println("try process PID 0x2a0b");
    float temp = myELM327.processPID(0x22, 0x2a0b, 1, 2, 1.0, 0.0);

    if (myELM327.nb_rx_state == ELM_SUCCESS)
    {
      DEBUG_PORT.print("Temp: ");
      DEBUG_PORT.println(temp);
      tft.print("Temp: ");
      tft.println(temp);

[... 292 lines cut ...]
Long query detected
Query string: 222A0B1
Clearing input serial buffer
Sending the following command/query: 222A0B1
Received:
No error detected
Service: 34
PID: 652
Long query detected
Query string: 22028C1
Clearing input serial buffer
Sending the following command/query: 22028C1
SOC2: error
Received:
No error detected
Timeout detected with overflow of 101ms
Received:
ERROR: ELM_TIMEOUT
Service: 34
PID: 652
Long query detected
Query string: 22028C1
Clearing input serial buffer
Sending the following command/query: 22028C1
```

jimwhitelaw, 2023-12-17:
The important info is here: 

Clearing input serial buffer
Sending the following command/query: AT SH 17FC007B
        Received char: ?
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received: 

The ? response indicates that the ELM device did not understand the command. See p.39 of the ELM327 datasheet for details on setting headers. As a guess, all the examples use spaces between the header bytes, so maybe try sending

AT SH 17 FC 00 7B 

and see if that works.

Waldmensch1, 2023-12-17:
Nope, Result is the same

```
Connected to ELM327
Clearing input serial buffer
Sending the following command/query: AT SH 17 FC 00 7B
        Received char: ?
        Received char: \r
        Received char: \r
        Received char: >
Delimiter found.
All chars received:
```
in the project where I found the PIDs they fire additional AT commands before set the header. Can this make a different in accepting the header?
Just to mention: The OBDII Adapter works well with Carscanner App in ioS. The Values for PID I try to use are shown without any problem.

```
    "AT Z",    // reset to default settings to be sure what the base setting are
      "AT I",    // Print the version ID
      "AT SP7",  // Set protokoll to 7 - ISO 15765-4 CAN (29 bit ID, 500 kbaud)
      "AT BI",   // Bypass the initialization sequence to make sure it stays at protokoll 7
      "AT CAF0", // Turns off CAN automatic formating so that PCI bytes are not inserted in the messages, or expected in the response
      "AT L0",   // Linefeeds off
      "AT DP",   // xxxxxxxxx
      "AT ST16", // reduced timeout to 1, orig.16
```

jimwhitelaw, 2023-12-18:
If you know that the above sequence of AT commands is working with carscanner then perhaps try sending those first. From the output you've provided, I'm not convinced that the protocol has been successfully detected. The last AT command you send with an OK response is "AT TP A7". That tells the ELM to try protocol 7 and fall back to an auto search if that doesn't work. After that, the next command you send should trigger the protocol detection. The ELM327 datasheet (p.29) recommends sending "0100" command to trigger the protocol detection. You should get back a response that starts with "SEARCHING..." followed by some additional bytes indicating which PIDs are supported in that mode. 

If you use the "AT SP 7" command as above, you won't have to go through the detection part, but it also won't search for any other protocol if 7 is not correct. All OBDII compliant systems are required to support command "0100" so it's a good test to see if you've actually got a valid connection from ELM to ECU before proceeding with other queries.

Waldmensch1, 2023-12-18:
I changed my loop that it plays out the AT command sequence I mentioned one by one. Also I set ATH1 to see what headers maybe come in. I try then to get the PIDs for SoC and Temp without setting a header. After that all I jump into a test console, where I can manueall send commands from serial terminal over to ELM. In Log you can see that the AT commands are all accepted and confirmed by ELM. Also if I send some commannd manually like 0140 I get a response 18 DA F1 0A 03 7F 40 11 AA AA AA AA. (ignore the echo I get from Terminal, have to find a way to switch of in PIO)
If I send a header without the leading 17 (FC007B) I get an "ok" back instead of "?" when I send 17FC007B. Could it be that the header is to long?

The Header and PID I have found in evDash project. What I meant with Carscanner was, that SoC and Temp get shown there as values, without knowledge of magic behind there. What I wanted to tell was that the ELM should be okay if other software can communicate.

This is the link to ID.3 specific code in evDash: https://github.com/nickn17/evDash/blob/master/src/CarVWID3.cpp

code:
```
void loop()
{
  button_loop();

  switch (cmd)
  {
  case CMD1:
  {
    if (myELM327.sendCommand_Blocking("AT Z") == ELM_SUCCESS)
      tft.println("AT Z okay");
    else
      myELM327.printError();
    cmd = CMD2;
  }
  case CMD2:
  {
    if (myELM327.sendCommand_Blocking("AT I") == ELM_SUCCESS)
      tft.println("AT I okay");
    else
      myELM327.printError();
    cmd = CMD3;
  }
  case CMD3:
  {
    if (myELM327.sendCommand_Blocking("AT SP7") == ELM_SUCCESS)
      tft.println("AT SP7 okay");
    else
      myELM327.printError();
    cmd = CMD4;
  }
  case CMD4:
  {
    if (myELM327.sendCommand_Blocking("AT BI") == ELM_SUCCESS)
      tft.println("AT BI okay");
    else
      myELM327.printError();
    cmd = CMD5;
  }
  case CMD5:
  {
    if (myELM327.sendCommand_Blocking("AT CAF0") == ELM_SUCCESS)
      tft.println("AT CAF0 okay");
    else
      myELM327.printError();
    cmd = CMD6;
  }
  case CMD6:
  {
    if (myELM327.sendCommand_Blocking("AT L0") == ELM_SUCCESS)
      tft.println("AT L0 okay");
    else
      myELM327.printError();
    cmd = CMD7;
  }
  case CMD7:
  {
    if (myELM327.sendCommand_Blocking("AT DP") == ELM_SUCCESS)
      tft.println("AT DP okay");
    else
      myELM327.printError();
    cmd = CMD8;
[... 459 lines cut ...]
NO DATA
>[440184][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
1100
NO DATA
>[549142][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
00110000
18 DA F1 0A 03 7F 00 11 AA AA AA AA 
>[596048][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
00112200
18 DA F1 0A 03 7F 20 11 AA AA AA AA 
>[626106][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
00114400
18 DA F1 0A 03 7F 40 11 AA AA AA AA 
>[674289][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
00110055
STOPPED1 0A 03 7F 05 11 AA AA AA AA 
>[753411][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
001100CC
18 DA F1 0A 03 7F 0C 11 AA AA AA AA 
>[790979][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0
001155CC
18 DA F1 0A 03 7F 5C 11 AA AA AA AA 
>[865070][I][BluetoothSerial.cpp:583] esp_bt_gap_cb(): ESP_BT_GAP_MODE_CHG_EVT: mode: 0

```

Waldmensch1, 2023-12-18:
I found another big list of PIDs. If you search on this page for „BMS Temperatur Pack Gesamt“ - how this record can be used in ElmDuino the proper way to get a value? https://www.goingelectric.de/forum/viewtopic.php?p=1758362&sid=f3168951a5a627b2930c870df144f35a#p1758362

This is the record
`
[CCAN=UDS,13708,0x,1011,5000,REG={0x22,0x2A,0x0B},TYPE={+,24,8,MSB},CONV={-40,1,2},EXT] # BMS Temperatur Pack Gesamt`
 
My interpretation would be
Command AT SH 17FC007B
queryPid( „222A0B“)
or 
queryPid(0x22, 0x2A0B)
or
processPid(0x22, 0x2A0B, 1, 1, 1.0, 1.0)

Waldmensch1, 2023-12-18:
I googled a bit around: could it be that UDS need a specific handling which is not supported yet by ELMDuino? I have seen some descriptions that the session have to be changed from OBD2 to UDS by fire a command like 0x1003 or so

jimwhitelaw, 2023-12-18:
I don’t know much about UDS, but since it operates at a layer “above” CAN, it should be possible to use ELMduino to implement the sending and receiving of UDS data. You would need to implement the command formation and data parsing yourself for the PIDS you are interested in. The good news is you have successfully established communication with your ECU and move forward.

Also, I note that your switch/case in your loop isn’t working quite the way it should. After each successful response, you should set the next cmd state and then break. As it is, all of the commands are run without respect to the cmd state, immediately after each other. That works for the blocking commands, but is going to cause problems on the processPIDS() commands as there’s no waiting for a response before the next command is sent. Have a close look at the multiple PIDs example program for the correct pattern.

closed 2025-02-09 by PowerBroker2