PowerBroker2/ELMduino #215: Issue in getting the value of oilTemp()

Talha-Babar, 2024-01-05:
I am using this library for multiple PIDs at a time with Bluetooth and it is working fine, but when i try to read the oil temp using float oilTemp() the sensor is not giving any data.

Would you help me in this issue?

Thanks in advance : )

jimwhitelaw, 2024-01-05:
Are you receiving a “NO DATA” response? If so, that simply means that the ECU does not support that sensor/PID.

jimwhitelaw, 2024-09-26:
@Turksat46 Do you know for certain that those vehicle all have oil temp sensors? In my (limited) experience, it is not common and I've had to add them myself with aftermarket sensors and gauges. Even if they do have, it's still possible that the manufacturers do not expose the data via the OBDII standard PIDs, but may do so with proprietary modes/PIDs. That would need to be researched for each specific vehicle.

closed 2025-02-09 by PowerBroker2