PowerBroker2/ELMduino #218: Custom PIDs

415Robots, 2024-01-24:
Greetings,

I am having trouble pulling a specific PID (220052) . Currently using processPID() to make custom calls, but it only returns 0.0. I've tried the follow:

- Try ethanolPercentage within ELMduino (not working)
- Use processPID() on a few different custom PIDs (working)
- Validate 220052 exists and is enabled, using torque app (working)
- Try different Services for the PID 0x52 (not working)

// example
double var = myELM327.processPID(34, 82, 1, 1, 100.0 / 255.0, 0.0);

My next plan of attack is to use the code in the test.ino to try passing the decimal or hex values over serial to see I can get anything. Is there any specific variables I should watch, that might give me a clue as to what I am doing wrong?

PowerBroker2, 2024-01-24:
Please post your formatted code and debug printouts.

415Robots, 2024-01-26:
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
ELM327 myELM327;

double var = 1.0;

void setup()
{
  Serial.begin(115200);
  WiFi.mode(WIFI_STA);

  SerialBT.begin("ArduHUD", true);
  if (!SerialBT.connect("OBDII"))
  {
    Serial.println(F("Couldn't connect to OBD scanner - Phase 1"));
    while (1);
  }

  if (!myELM327.begin(SerialBT, true, 2000))
  {
    Serial.println(F("Couldn't connect to OBD scanner - Phase 2"));
    while (1);
  }
  
  Serial.println(F("Connected to ELM327"));
}

void loop() {
  var = myELM327.processPID(34, 82, 1, 1, 100.0/255.0,0);

  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    Serial.print("Ethanol (%): ");
    Serial.println(var);
  } else {
    Serial.println("Failed to read Transmission Temperature or PID not supported.");
  }

  delay(200); // Delay between reads
}
[image]
[image]
[image]

jimwhitelaw, 2024-01-26:
@415Robots Your loop needs to handle the case where the code is still getting the data from the ELM device. Try this in your loop():

```C
if (myELM327.nb_rx_state == ELM_SUCCESS)
      {
       Serial.print("Ethanol (%): ");
       Serial.println(var);
      }
      else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
      {
        myELM327.printError();
      }
```
That will allow the full response message to be received before being processed.

Also, the code shown in your two screenshots are both different than the code you posted, so it's not clear what code is actually running...

415Robots, 2024-01-26:
Thanks for the feedback, I'll add the print Error and post the results. I only have limited access to the vehicle, so the testing was a lot of modifying working code to try to get PID 220052 to work, apologies for the confusion. 

- Screenshot 1: Output from processPID with the following inputs: Service = 34 (0x22 = 34), PID = 82 (0x52 =82)
- Screenshot 2: Debug information showing the query string, Service, and PID 
- Screenshot 3: Testing to see if "220052" is a valid PID using powerbroker's test.ino example with the serial input of "220052" and "340082". If you look at the last byte of the highlighted data in the message, you'll see the value C6. Applying the conversion from hex to percentage, the result yields the expected result of 77.6%. 

Sidenote: I definitely will check out the ELMulator, I really could use some kind of bench test for this project.

415Robots, 2024-01-27:
Added the else if to target the non ELM_GETTING_MSG states, but it doesn't seem to generate an error... or I am doing it wrong. Results in the
[image]
 attached:

jimwhitelaw, 2024-01-27:
You are not doing it wrong, and you should not expect an error. What I see happening now is that you’re getting a CAN “Response Pending” message (7Fxxxx). ELMduino doesn’t currently handle this type of response, and instead it returns zero. See p. 45 of the Elm327 datasheet for info.

PowerBroker2, 2024-01-27:
@415Robots Can you either post the debug printouts in the text of your comments (formatted please) or post the screenshots different? It's hard for me to read in the screenshots and when I try to download the hires, github comes up with an empty page for some reason...

Either way, it looks like the response header for service 2 (or whichever service you're using) doesn't behave like response headers for service 1.

thebeardedgarage, 2024-01-28:
21:54:08.931 -> Ethanol (%): 0.00
21:54:09.119 -> Service: 34
21:54:09.119 -> PID: 82
21:54:09.119 -> Normal length query detected
21:54:09.119 -> Query string: 22521
21:54:09.119 -> Clearing input serial buffer
21:54:09.157 -> Sending the following command/query: 22521
21:54:09.321 -> Received char: 7
21:54:09.534 -> Received char: F
21:54:09.755 -> Received char: 2
21:54:09.940 -> Received char: 2
21:54:10.153 -> Received char: 1
21:54:10.321 -> Received char: 2
21:54:10.534 -> Received char: \r
21:54:10.755 -> Received char: \r
21:54:10.932 -> Received char: >
21:54:10.932 -> Delimiter found.
21:54:10.932 -> All chars received: 7F2212
21:54:10.932 -> Expected response header: 6252
21:54:10.932 -> Response not detected
21:54:10.932 -> WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0
21:54:10.984 -> Ethanol (%): 0.00
21:54:11.154 -> Service: 34
21:54:11.154 -> PID: 82
21:54:11.154 -> Normal length query detected
21:54:11.154 -> Query string: 22521
21:54:11.154 -> Clearing input serial buffer
21:54:11.154 -> Sending the following command/query: 22521
21:54:11.368 -> Received char: 7
21:54:11.581 -> Received char: F
21:54:11.772 -> Received char: 2
21:54:11.958 -> Received char: 2
21:54:12.180 -> Received char: 1
21:54:12.354 -> Received char: 2
21:54:12.559 -> Received char: \r
21:54:12.771 -> Received char: \r
21:54:12.970 -> Received char: >
21:54:12.970 -> Delimiter found.
21:54:12.970 -> All chars received: 7F2212
21:54:12.970 -> Expected response header: 6252
21:54:12.970 -> Response not detected
21:54:12.970 -> WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0
21:54:13.024 -> Ethanol (%): 0.00
21:54:13.202 -> Service: 34
21:54:13.202 -> PID: 82
21:54:13.202 -> Normal length query detected
21:54:13.202 -> Query string: 22521
21:54:13.202 -> Clearing input serial buffer
21:54:13.202 -> Sending the following command/query: 22521
21:54:13.418 -> Received char: 7
21:54:13.587 -> Received char: F
21:54:13.801 -> Received char: 2
21:54:14.179 -> Received char: 2
21:54:14.189 -> Received char: 1
21:54:14.404 -> Received char: 2
21:54:14.606 -> Received char: \r
21:54:14.819 -> Received char: \r
21:54:14.988 -> Received char: >
21:54:14.988 -> Delimiter found.
21:54:14.988 -> All chars received: 7F2212
21:54:15.167 -> Expected response header: 6252
21:54:15.167 -> Response not detected
21:54:15.167 -> WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 - returning 0

PowerBroker2, 2024-01-28:
I'm not sure if this is a CAN error/warning as mentioned earlier or if this is how a service 32 PID response header is supposed to look like - can you provide more info on this custom PID? Do you have any documentation on it? I'm conflicted on whether I should make a special case for it in the lib or not.

thebeardedgarage, 2024-01-28:
Hello Powerbro, I am in the same situation as @415Robots. The information I have recieved on this PID is from the torque pro app.

PID: 220052
Min. Value: 0
Max Value:100
Scale Factor: x1
Unit type: %
Equation: A/255*100

When running this pid through the obd2 editor in torque app, I get this response:

[image]

PowerBroker2, 2024-01-28:
I'm still unsure of how the PID is supposed to work and how to calculate what the response header should be. Is there any actual documentation of the PID or PID response?

thebeardedgarage, 2024-01-28:
Using your test.ino sketch, here is the response for 220052. The last byte of the highlighted data in the message, you'll see the value C6. Applying the conversion from hex to percentage, the result yields the expected result of 77.6%.

[image]

PowerBroker2, 2024-01-28:
Idk what was going on in the previous set of debug prints, but this particular response looks similar to what I would expect. Notice how the response header's first byte is +4 compared to the query response header's first byte. The library in `findResponse()` is setup to handle this case, even for 6-byte PID queries:

https://github.com/PowerBroker2/ELMduino/blob/77874c0d13e262e59a15211c49642ecc5eb34e28/src/ELMduino.cpp#L2327-L2335

I think more detailed debugging will be required to figure out what is going wrong.

StoneIsthisticky, 2024-11-10:
Hi Guys,

(While I can speak a bit of English, for clarity, I prefer to write in French and provide a translated version.)

I own a diesel Saab 93 (the same model as the Opel/Vauxhall Vectra or Cadillac BLS), which uses the GM communication protocol (and the Tech2 tool). Like many owners of these diesel models, I've encountered issues with the DPF, EGR, etc., and quickly sought ways to manage these problems.

I started by purchasing an ELM327 Bluetooth adapter paired with the Android app Torque. Unfortunately, I discovered that GM models weren't very 'cooperative' with the standards used by Torque.

I rented a Tech2 to diagnose a problem and discovered that a lot of information was accessible through this tool. I found it a shame that the protocol was so 'locked down' and so little data could be accessed on Torque.

I eventually found (on various forums) custom PIDs to manually add to track DPF clogging and regeneration. At the time, I didn't even know what hexadecimal language was. Even today, I have no knowledge of development or programming in these languages, nor in open source (I created my GitHub account today).

To get back to the point, I continued to try to get the same parameters on the Torque app that I had seen on the Tech2. On the advice of another user, I scanned all the addresses accessible by my ELM327 adapter connected to my car. I began to understand the acquired hexadecimal data. I managed to retrieve about fifteen other custom PIDs (temperatures, actuator positions, etc.), but there are still some I would like to obtain without success.

So, I would like to know if, in developing your project, you have found a solution to access these PIDs? (addresses, but especially the equations used...)

I had the idea of searching in the .bin files of the maps used in the Tech2, but I don't think the addresses are stored there. However, the Global TIS software might contain this information in its source code? Not being skilled in computer science, I'm a bit stuck at this stage.
Do you think this solution could give us some information?

I'm available to discuss this further if you agree. I can share my findings on my vehicle.

PowerBroker2, 2024-11-10:
@StoneIsthisticky Thank you for the interest! I personally don't have experience using custom PIDs, but it shouldn't be too hard to implement if you know the service and PID of the custom query along with the expected output format. I'm also not experienced with Saabs or diesels in general. Do you have links to documentation on the PIDs you want to process?

StoneIsthisticky, 2024-11-10:
Hi, thanks for your answer.
To be honest, i don't think that it's different because it's a diesel... ^^
Sensors, actuators,...work the same.
I would especially want the instant fuel consumption (that there is no on
the sid, where there is only long term trim),
the mass air flow and the lambda.
But I think that it could be usefull to reach all datas... To make some
diag without tech2.
The adresses with responding data are from AA0101 to AA0436.
For exemple:

StoneIsthisticky, 2024-11-10:
[image]

PowerBroker2, 2024-11-11:
I'm confused, are those the PIDs you need help with or the ones you need help with?

closed 2025-02-09 by PowerBroker2