PowerBroker2/ELMduino #222: Problem querying data with ESP32

cankarabey, 2024-02-05:
Hey everyone,

I am creating this issue since I have been having problems trying to get data from ELM327 v2.1 with an ESP32. I am able to connect to the device, but when I send AT commands via the BT_test script, the results look like this:

ATZATZ?
ATE0ATE0?
ATL0AT?0
ATH1AT?1
ATSPOATSP?

0100 NO DATA
010C NO DATA

I have tested the ELM device with an Android diagnostics tool and I also tested it via a bluetooth serial terminal app where I sent the same commands as listed above and received the correct responses.. So the problem seems to lie with the ESP32 connection, I have tried baudrates 9600, 38400 and 115200 without success. Any ideas on what could cause this?

Any help is much appreaciated!

Regards

cankarabey, 2024-02-05:
Hi @jimwhitelaw, thanks for the quick response! The SP0 part was typo on my end since I was typing it out of memory.. Here is the actual output:

[image]

I am using the example script [ESP32_test](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_test/ESP32_test.ino)

For comparison, here are the same commands sent via the Bluetooth Serial Terminal app:

[image]

Thanks!

jimwhitelaw, 2024-02-06:
I'm not sure what's happening there. The only things that I see of note are 1) the '?' chars in the responses. Normally, a ? in ELM327 response means that the device did not understand the command given, but that doesn't appear to be the case. It might be that your terminal program is not correctly interpreting the carriage return char (shown as ^M in the Android app). 2) The NO DATA response typically means that the ECU doesn't support the PID being queried, but PID 0100 is a required PID by OBD2 standard and should always provide a response.

Perhaps try one of the other example programs that doesn't rely on entering manual commands from a terminal, just to take your terminal program out of the loop. Try running [ESP_Check_PID_Support](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Check_PID_Support/ESP32_Check_PID_Support.ino) with a couple PIDs like 0x05 (RPM) and 0x0C (Coolant Temp) and see if you get any responses that way.

cankarabey, 2024-02-24:
I decided to take a different approach, de-soldered the Bluetooth module and tried to communicate directly via UART, this works and is actually preferred in my case, so I will close the issue now. Thanks for the support!

closed 2024-02-24 by cankarabey