PowerBroker2/ELMduino #225: Custom PID for DPF Saturation (22242C)

etnaskynet, 2024-02-21:
Hello,
   I need to monitor my DPF Saturation using your library, with my ESP32 and Bluetooth.
   for the mode = 22 and PID = 242C with this formula (A*256+B)/100.

   In past when I query with 22242C I received the following string from the ECU:
   7E8 05 62 24 2C 01 63
   So,
   7E8 = identifier
   05 = Length (bytes)
   62 = response mode marker
   24 = PID
   2C = PID
   01 = A
   63 = B

   If I convert A and B in decimal (1 and 99 respectively) and applying the formula, the saturation is 3.5 grams, that is correct.

   QUESTION:
   how can I manage this using your library?
   Should I update both .cpp and .h file or directly use processPID in my .ino file?

Thanks in advance for your help.

jimwhitelaw, 2024-02-22:
The library doesn't support that type of query using _processPID()_, but you can use it to get the data you want. Instead, use _sendCommand()_, and then parse the response in your code. The full response from ELM327 will be available to you in _myELM327.payload_ after calling _getResponse()_. 

Include something like this in your loop() method:

```C
    myELM327.sendCommand("22242C");
    if (myELM327.nb_rx_state == ELM_SUCCESS)
    {
        myELM327.getResponse();
        // parse and process the values you want from myELM327.payload
    }
    else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    {
        myELM327.printError();
    }
```

etnaskynet, 2024-02-22:
**IT WORKS.
I used this code:**
DPF_KM = myELM327.processPID(34, 9556, 4, 4, 1, 0); //PID 222554; Equation: ((A*256)+B)
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  { 
    D = myELM327.responseByte_0;
    C = myELM327.responseByte_1;
    B = myELM327.responseByte_2;
    A = myELM327.responseByte_3;
    DPF_KM_calc = ((C*256)+D);
    obd_state= ENG_RPM;
  }

**and I retrieved the correct data (verified) in Arduino Serial Monitor:**
Delimited found.
All chars received: 6225540139
Expected response header: 622554
Single response detected
64-bit response:
   responseByte_0: 57
   responseByte_1: 0
   responseByte_2: 0
   responseByte_3: 0
   responseByte_4: 0
   responseByte_5: 0
   responseByte_6: 0
   responseByte_7: 0
Service: 34
PID: 9556
Long query detected
Query string: 222554
Clearing input serial buffer
Sending the following command/query: 222554
   Received char: 6
   Received char: 2
   Received char: 2
   Received char: 5
   Received char: 5
   Received char: 4
   Received char: 0
   Received char: 1
   Received char: 3
   Received char: 9
   Received char: \r

So, after conversion, the KM from last regeneration are 313.

Thanks a lot for your support!

closed 2024-02-22 by etnaskynet