PowerBroker2/ELMduino #228: Custom PID, how it works

Evgeniy-18, 2024-02-26:
Hi, thanks for the great project. With the help of the project, I connected to the car, the standard pods work, but I have a Renault car, which I have my own custom codes that work in torque, but I do not understand which methods of the library I should use, I will be very grateful for your help.
For example, a special code for water temperature, **ECU - 7E0**, and **PID - 222001**, in response the ECU sends **6220010B86**.
622001 - successful response, **A - 0B, B - 86** - byte value of water temperature.
formula **(1.0*((A*256+B)-2730))/10**.
How to correctly send a request, receive a response and translate its value?

PowerBroker2, 2024-02-26:
You can use `processPID()`. If you use an ESP32, you might be able to do the following:

```C++
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

float waterTemp = 0;

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  //SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  
  if (!ELM_PORT.connect("OBDII"))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempWaterTemp = ((myELM327.processPID(2, 22001, 1, 4, 1, 0) - 2730.0) / 10.0);

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    waterTemp = tempWaterTemp;
    Serial.print("Water Temp: "); Serial.println(waterTemp);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}
```

No guarantee the above will work, but is a good starting point.

jimwhitelaw, 2024-02-26:
Try this change:
```C
float tempWaterTemp = ((myELM327.processPID(22, 2001, 1, 4, 1, 0) - 2730.0) / 10.0);
```

Evgeniy-18, 2024-02-27:
procesPID - changes the custom PID.
I tried dialog with ELM via ELMCHAT and
 6220010B86 was the answer to the request "222001".
I tried queryPID("222001"), log "...Sending the following command/query: 222001...".
What function will give me the ECU response?

jimwhitelaw, 2024-02-27:
When you send request “222001”, the request is made up of two components, the service (mode) “22” and the PID itself “2001”. If you use ELMduino’s processPID() method, you specify those separately, and the library will create the correct command to send to the ECU. As an alternative, you could use the sendCommand() method with value “222001” and then parse the result manually.

If you use the code that PB2 provided with my suggested change, it should work.

Evgeniy-18, 2024-02-27:
> Try this change:
> 
> ```c
> float tempWaterTemp = ((myELM327.processPID(22, 2001, 1, 4, 1, 0) - 2730.0) / 10.0);
> ```

float tempWaterTemp = ((myELM327.processPID(34, 8193, 1, 4, 1, 0) - 2730.0) / 10.0);

Service: 34
20:03:07 PID: 8193
20:03:07 Long query detected
20:03:07 Query string: 2220011
20:03:07 Clearing input serial buffer
20:03:07 Sending the following command/query: 2220011
20:03:08 ^IReceived char: 6
20:03:09 ^IReceived char: 2
20:03:10 ^IReceived char: 2
20:03:11 ^IReceived char: 0
20:03:12 ^IReceived char: 0
20:03:13 ^IReceived char: 1
20:03:14 ^IReceived char: 0
20:03:15 ^IReceived char: B
20:03:16 ^IReceived char: C
20:03:17 ^IReceived char: A
20:03:18 ^IReceived char: \r
20:03:19 ^IReceived char: \r
20:03:20 ^IReceived char: >
20:03:20 Delimiter found.
20:03:20 All chars received: 6220010BCA
20:03:20 Expected response header: 622001
20:03:20 Single response detected
20:03:20 64‑bit response: 
20:03:20 ^IresponseByte_0: 202
20:03:20 ^IresponseByte_1: 11
20:03:20 ^IresponseByte_2: 0
20:03:20 ^IresponseByte_3: 0
20:03:20 ^IresponseByte_4: 0
20:03:20 ^IresponseByte_5: 0
20:03:20 ^IresponseByte_6: 0
20:03:20 ^IresponseByte_7: 0
20:03:20 WARNING: Number of payload chars is less than the number of expected response chars returned by ELM327 ‑ returning 0

Although the response bytes are correct, 28.8C

Evgeniy-18, 2024-02-27:
there is a reconnection function when communication with the ECU is lost?

jimwhitelaw, 2024-02-27:
There isn’t a reconnect function in the library, but you can check the status of _myELM327.connected_ and if it is false, perform a reconnection process like this:

```C
bool elmConnectBT()
{
    if (!serialBT.connected(1000))
    { // We've lost the blutooth connection, try to reconnect

        btConnected = setupBluetooth();
    }

    Serial.println("Starting ELM327 connection to ECU...");
    while (!elmConnected && elmAttempts <= elmRetries)
    {
        elmAttempts++;
        elmConnected = myELM327.begin(serialBT, true);
    }
    if (elmConnected)
    {
        Serial.println("ELM327 connected to ECU");
    }
    else
    {
        Serial.print("Failed to connect to ELM327 after ");
        Serial.print(elmAttempts);
        Serial.println(" attempts. Check configuration.");
    }
    return elmConnected;
}

```
Note it might also be the Bluetooth connection is lost, so you have to check for that and reestablish the BT connection first if needed.

closed 2024-02-28 by Evgeniy-18