PowerBroker2/ELMduino #230: Problem with flashing RPM value on OLED when using latest version of ELMduino

wcjxixi, 2024-02-28:
Hi, thanks to PowerBroker2 for making this ELMduino library!

I am using ESP32 + Vgate iCar2 + SSD1306 monochrome 128x64 OLED.

I was previously using ELMduino v2.6.4 and the RPM value display has been working fine. When I now use the latest version of ELMduino v3.2.5, the RPM value flashes. I guess it may be due to ELMduino switching to non-blocking code from v3.0.0.

My code is as follows:

```C++
#include <BluetoothSerial.h>
#include <ELMduino.h>
#include <Adafruit_SSD1306.h>
#include <U8g2_for_Adafruit_GFX.h>

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

/* SSD1306 */
Adafruit_SSD1306 display = Adafruit_SSD1306(128, 32);
U8G2_FOR_ADAFRUIT_GFX u8g2;

/* define custom color <RGB565> */
#define BLACK   0x0000
#define WHITE   0xFFFF

uint32_t rpm = 0;

//_____________________________________________________________
void setup() {
  Serial.begin(115200);

  /* initialize SSD1306 */
  display.begin(SSD1306_SWITCHCAPVCC);
  display.setRotation(0);
  u8g2.begin(display);
  display.clearDisplay();
  display.display();
  delay(50);

  //ELM_PORT.setPin("1234");
  ELM_PORT.begin("OBDII", true);

  if (!ELM_PORT.connect("V-LINK")) {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while (1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

//_____________________________________________________________
void loop() {
  display.clearDisplay();
  display.fillRect(0, 0, 128, 32, WHITE);
  u8g2.setForegroundColor(BLACK);
  u8g2.setFont(u8g2_font_wqy16_t_gb2312a);
  u8g2.setFontMode(1);
  u8g2.setCursor(5, 22);
  u8g2.print("RPM: ");

  u8g2.setBackgroundColor(WHITE);
  u8g2.setFont(u8g2_font_freedoomr25_mn);
  u8g2.setCursor(50, 30);

  float tempRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    u8g2.print(rpm);
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    u8g2.print("-");
    myELM327.printError();
  }
  display.display();
}
```
For using ELMduino v2.6.4 before and now using ELMduino v3.2.5, mainly the following code is different:

When using ELMduino v2.6.4:
```C++
  if (myELM327.status == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    u8g2.print(rpm);
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else {
    u8g2.print("-");
    myELM327.printError();
  }
```
When using ELMduino v3.2.5:
```C++
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    u8g2.print(rpm);
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    u8g2.print("-");
    myELM327.printError();
  }
```
Any suggestions please? So that I can simply use the latest version of the ELMduino library, because the latest version of ELMduino has new features such as getting the VIN, and I basically have to use all the PIDs in my project. also, I'm just a hobbyist!

DerKleinePunk, 2024-02-28:
@wcjxixi I think it is the display.clearDisplay(); Because non Blocking from new Lib version. 

`void UpdateDisplay(uint32_t rpm, bool error)
{
display.clearDisplay();
  display.fillRect(0, 0, 128, 32, WHITE);
  u8g2.setForegroundColor(BLACK);
  u8g2.setFont(u8g2_font_wqy16_t_gb2312a);
  u8g2.setFontMode(1);
  u8g2.setCursor(5, 22);
  u8g2.print("RPM: ");

  u8g2.setBackgroundColor(WHITE);
  u8g2.setFont(u8g2_font_freedoomr25_mn);
  u8g2.setCursor(50, 30);
  if(error) {
u8g2.print("-");
  } else {
  u8g2.print(rpm);
 }
 display.display();
}

void loop() {

  float tempRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    UpdateDisplay(rpm, false);
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    UpdateDisplay(0, true);
    myELM327.printError();
  }
}
`

The Code is written hier not Tested. I hope it helps

wcjxixi, 2024-02-29:
How can I get the values of multiple PIDs and display them at the same time?

For example, I want to display two PIDs (e.g. rpm and coolant) on my SSD1306 LCD at the same time, but the correct value and the error value will be displayed alternately.

My code is as follows:
```C++
void show_canvas_on_screen() {
  screen.drawRGBBitmap(0, 0, canvas.getBuffer(), canvas.width(), canvas.height());
}

//_____________________________________________________________
void updateRpm(uint32_t rpm, bool error) {
  canvas.fillScreen(BLACK);
  gfx.setFont(u8g2_font_wqy15_t_gb2312b);
  gfx.setForegroundColor(LIME);
  gfx.setCursor(2, 14);  gfx.println(F("发动机转速 rpm"));

  gfx.setFont(dsdigital_42);
  gfx.setForegroundColor(WHITE);
  gfx.setCursor(0, 62);
  if (error) {
    gfx.println(F("-"));
  } else {
    gfx.println(rpm);
  }
  //show_canvas_on_screen();
}

void updateCoolant(uint32_t coolant, bool error) {
  //canvas.fillScreen(BLACK);
  gfx.setFont(u8g2_font_wqy15_t_gb2312b);
  gfx.setForegroundColor(LIME);
  gfx.setCursor(2, 80);  gfx.println(F("冷却液温度 °C"));

  gfx.setForegroundColor(WHITE);
  gfx.setFont(dsdigital_47);
  gfx.setCursor(0, 128);
  if (error) {
    gfx.println(F("-"));
  } else {
    gfx.println(coolant);
  }
  show_canvas_on_screen();
}

//_____________________________________________________________
void loop() {
  float tempRPM = myELM327.rpm();
  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    rpm = (uint32_t)tempRPM;
    updateRpm(rpm, false);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    updateRpm(0, true);
  }

  float tempCoolant  = myELM327.engineCoolantTemp();
  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    coolant = (uint32_t)tempCoolant;
    updateCoolant(coolant, false);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    updateCoolant(0, true);
  }
}
```

The debug log is attached:
[debug_log.txt](https://github.com/PowerBroker2/ELMduino/files/14448171/debug_log.txt)

PowerBroker2, 2024-02-29:
You'll want to look at [the multiple PIDs example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/multiple_pids/multiple_pids.ino)

DerKleinePunk, 2024-03-01:
[More Complex Sample](https://github.com/DerKleinePunk/ODB-ESP32-Reader)

jimwhitelaw, 2024-03-01:
@DerKleinePunk thanks for sharing that link. There's some interesting code there, I like how you've integrated BLE/classic BT. Nice work.

DerKleinePunk, 2024-03-03:
@jimwhitelaw Thanks

wcjxixi, 2024-03-03:
@PowerBroker2 @DerKleinePunk THX!

According to the sample code, It is supposed to use the switch case. I modified the code template as follows: 
``` C++
int    switchNum = 1;

uint32_t     rpm = 0;
......
uint32_t coolant = 0;

//_____________________________________________________________
void updateRpm(uint32_t rpm, bool error) {
  canvas.fillScreen(BLACK);
  gfx.setFont(u8g2_font_wqy15_t_gb2312b);
  gfx.setForegroundColor(LIME);
  gfx.setCursor(2, 14);  gfx.println(F("发动机转速 rpm"));

  gfx.setFont(dsdigital_42);
  gfx.setForegroundColor(WHITE);
  gfx.setCursor(0, 62);
  if (error) {
    gfx.println(F("-"));
  } else {
    gfx.println(rpm);
  }
  //show_canvas_on_screen();
}

......

void updateCoolant(uint32_t coolant, bool error) {
  //canvas.fillScreen(BLACK);
  gfx.setFont(u8g2_font_wqy15_t_gb2312b);
  gfx.setForegroundColor(LIME);
  gfx.setCursor(2, 80);  gfx.println(F("冷却液温度 °C"));

  gfx.setForegroundColor(WHITE);
  gfx.setFont(dsdigital_47);
  gfx.setCursor(0, 128);
  if (error) {
    gfx.println(F("-"));
  } else {
    gfx.println(coolant);
  }
  show_canvas_on_screen();
}

//_____________________________________________________________
void loop() {
  switch (switchNum) {
    case 1: {
        float tempRPM = myELM327.rpm();
        if (myELM327.nb_rx_state == ELM_SUCCESS) {
          rpm = (uint32_t)tempRPM;
          updateRpm(rpm, false);
          switchNum == 2;
        }
        else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
          updateRpm(0, true);
          switchNum == 2;
        }
        break;
      }

      ......

    case n: {
        float tempCoolant  = myELM327.engineCoolantTemp();
        if (myELM327.nb_rx_state == ELM_SUCCESS) {
          coolant = (uint32_t)tempCoolant;
          updateCoolant(coolant, false);
          switchNum == 1;
        }
        else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
          updateCoolant(0, true);
          switchNum == 1;
        }
        break;
      }
  }
}

```

Now I can get the PIDs value. 

However, there is a delay in displaying multiple PID queries, and the more PID queries there are, the greater the delay in displaying them, whereas ELMduino v2.6.4 has no delay at all.

Overall this delay seems acceptable though, so if there's nothing better to do, that's it. Thanks again guys!

closed 2024-03-03 by PowerBroker2