PowerBroker2/ELMduino #231: ESP32 not connecting to ELM327

saleh-alobaidi, 2024-02-29:
Hello 
I'm working on a vehicle black box system for my end-of-year project and i cant connect the esp32 to elm327

i ran the ESP32 test sketch and got the same output: "Couldn't connect to OBD scanner." 

Is there a way to initiate communication first? Do I need to force the esp32 to be in master mode in order to communicate?

jimwhitelaw, 2024-02-29:
The ESP32_test.ino sketch uses a Bluetooth device name of "OBDII" with no PIN. Is that correct for your ELM327 device?

saleh-alobaidi, 2024-02-29:
OBDII has default pin 1234

i managed to connect it with some other code but i had to change boards for some reason for it to work

but the problem is i cant send any commands to ELM, i tried AT commands and OBD commands but no feedback

`
#include "BluetoothSerial.h"

BluetoothSerial SerialBT;

String MACadd = "00:10:CC:4F:36:03";
uint8_t address[6] = { 0x00, 0x10, 0xCC, 0x4F, 0x36, 0x03 };
String name = "OBDII";
char *pin = "1234";  //<- standard pin would be provided by default
bool connected;

void setup() {
  Serial.begin(115200);
  SerialBT.setPin(pin);
  SerialBT.begin("ESP32", true);
  Serial.println("The device started in master mode, make sure remote BT device is on!");
  connected = SerialBT.connect(address);

  if (connected) {
    Serial.println("Connected Succesfully!");
  } else {
    while (!SerialBT.connected(10000)) {
Serial.println("Failed to connect. Make sure remote device is available and in range, then restart app.");
    }
  }
}

void loop() {
  if (Serial.available()) {
    SerialBT.write(Serial.read());
  }
  if (SerialBT.available()) {
    Serial.write(SerialBT.read());
  }
  delay(20);
}
```

________________________________
من: Jim Whitelaw ***@***.***>
‏‏تم الإرسال: 19/شعبان/1445 03:40 ص
إلى: PowerBroker2/ELMduino ***@***.***>
نسخة: saleh-alobaidi ***@***.***>; Author ***@***.***>
‏‏الموضوع: Re: [PowerBroker2/ELMduino] ESP32 not connecting to ELM327 (Issue #231)

The ESP32_test.ino sketch uses a Bluetooth device name of "OBDII" with no PIN. Is that correct for your ELM327 device?

—
Reply to this email directly, view it on GitHub<https://github.com/PowerBroker2/ELMduino/issues/231#issuecomment-1970171499>, or unsubscribe<https://github.com/notifications/unsubscribe-auth/BEAGSSM3KYPKYVO2BDDODDLYV7E6LAVCNFSM6AAAAABD67THACVHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMYTSNZQGE3TCNBZHE>.
You are receiving this because you authored the thread.Message ID: ***@***.***>

jimwhitelaw, 2024-03-01:
It's hard to know exactly what's happening without the serial/debug output. Perhaps try the _examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino_ sketch and post the serial output here.

saleh-alobaidi, 2024-03-02:
This is the output

[cid:e5b771a8-950d-457b-8d35-42327304d5af]
________________________________
من: Jim Whitelaw ***@***.***>
‏‏تم الإرسال: 20/شعبان/1445 03:59 ص
إلى: PowerBroker2/ELMduino ***@***.***>
نسخة: saleh-alobaidi ***@***.***>; Author ***@***.***>
‏‏الموضوع: Re: [PowerBroker2/ELMduino] ESP32 not connecting to ELM327 (Issue #231)

It's hard to know exactly what's happening without the serial/debug output. Perhaps try the examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino sketch and post the serial output here.

saleh-alobaidi, 2024-03-22:
Hello
I'm still stuck with the same problem, I can't connect the ESP32 to the ELM327.

I also tried with HC-05, and it still won't connect.

I appreciate your assistance in any way you can.

________________________________
من: Jim Whitelaw ***@***.***>
‏‏تم الإرسال: 20/شعبان/1445 03:59 ص
إلى: PowerBroker2/ELMduino ***@***.***>
نسخة: saleh-alobaidi ***@***.***>; Author ***@***.***>
‏‏الموضوع: Re: [PowerBroker2/ELMduino] ESP32 not connecting to ELM327 (Issue #231)

It's hard to know exactly what's happening without the serial/debug output. Perhaps try the examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino sketch and post the serial output here.

closed 2025-02-09 by PowerBroker2