PowerBroker2/ELMduino #232: Protocol Data Point

Fichte79, 2024-03-04:
Hello everyone

With which data point can I see which protocol the ELM327 communicates with the vehicle?

I would like to see which protocol is used.

Best regards

jimwhitelaw, 2024-03-04:
use `sendCommand("ATDP")` to get the name of the current protocol. AT command "DPN" will return the protocol number.

wcjxixi, 2024-03-06:
I would also like to know which protocol is actually used by the ELM327 to communicate with the vehicle.

I am using ESP32 + Vgate iCar2.

When using `sendCommand("ATDP")`, it seems to only show `AUTO`: 

```C++
ELM responded with error "UNABLE TO CONNECT"
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Setting protocol via AT SP %c did not work
Connected to ELM327
Clearing input serial buffer
Sending the following command/query: AT DP
	Received char: A
	Received char: U
	Received char: T
	Received char: O
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: AUTO
```

When using `sendCommand("ATDPN")`, only `A0` is displayed:
```C++
ELM responded with error "UNABLE TO CONNECT"
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Setting protocol via AT SP %c did not work
Connected to ELM327
Clearing input serial buffer
Sending the following command/query: AT DPN
	Received char: A
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: A0
```

jimwhitelaw, 2024-03-06:
After setting the auto protocol (ATSP0 or ATTP0), you also need to make a request to initiate the protocol search and then wait for the protocol negotiation to complete. Easiest way to do that is to send command "O100". 

Once that returns, you can check which protocol was negotiated.

wcjxixi, 2024-03-06:
> After setting the auto protocol (ATSP0 or ATTP0), you also need to make a request to initiate the protocol search and then wait for the protocol negotiation to complete. Easiest way to do that is to send command "0100".
> 
> Once that returns, you can check which protocol was negotiated.

@jimwhitelaw THX! I also learnt the way to get the data returned by sendCommand_Blocking, the way of sendCommand should be similar, right?

The example code is as follows, others can refer to:

```C++
unsigned char obdProtocolCode = 0;
......
myELM327.sendCommand_Blocking("0100");
if (myELM327.nb_rx_state == ELM_SUCCESS) {
  myELM327.sendCommand_Blocking("ATDPN");
  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    myELM327.get_response();
    obdProtocolCode = myELM327.payload[1];
  }
}
Serial.println("OBD Protocol: " + String(obdProtocolCode));
```

closed 2025-02-09 by PowerBroker2