PowerBroker2/ELMduino #233: Couldn't connect to OBD scanner (stucked)

gianluca-tursi, 2024-03-06:
hello thanks for your work! i'm currently stucked. I have uploaded this sketch:
ESP32_Bluetooth_Serial
ESP32_test

but i receive the message "Couldn't connect to OBD scanner"

i have this hardware:
esp32 titigo with display .> https://www.amazon.it/ICQUANZX-T-Display-Sviluppo-Bluetooth-Arduino/dp/B07VNG9D52
odbII -> https://www.amazon.it/bluetooth-Diagnostic-diagnostica-Problema-guastiCompatibile/dp/B0CKHCSQ5C/

i saw that the pin is 1234, and the name os OBDII

but i receive this message "Couldn't connect to OBD scanner"

i don't know how to fix it.
thanks a lot

jimwhitelaw, 2024-03-06:
Note that in both the examples you list, the code to set the Blutetooth PIN is commented out by default. Did you enable it?

gianluca-tursi, 2024-03-06:
> Note that in both the examples you list, the code to set the Blutetooth PIN is commented out by default. Did you enable it?

yes thanks, i have enabled it

jimwhitelaw, 2024-03-06:
OK, in that case, it will be helpful if you can post the full serial output from the device so we can see everything that is happening. Are you able to connect to the ELM device with another app, on a phone for instance?

gianluca-tursi, 2024-03-06:
> OK, in that case, it will be helpful if you can post the full serial output from the device so we can see everything that is happening. Are you able to connect to the ELM device with another app, on a phone for instance?

i think i can connect the odb with my phone, but how i can get the "full serial output from the device" ? what is it exactly?

jimwhitelaw, 2024-03-06:
Assuming you're using Arduino IDE: https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-monitor/

gianluca-tursi, 2024-03-06:
> Assuming you're using Arduino IDE: https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-serial-monitor/

yes i currently using arduino IDE for mac

gianluca-tursi, 2024-03-07:
> i think i can connect the odb with my phone, but how i can get the "full serial output from the device" ? what is it exactly?

some help with this?

PowerBroker2, 2024-03-07:
There's a chance you already have a phone or laptop that is already connected to the OBD2 scanner and won't let the ESP connect. Try turning off all other BT devices when using the ESP. Also, you might not need the PIN (my scanner(s) never needed a PIN, while others do). Lastly, you might try different bauds. Other than that, you might need to buy another OBD2 scanner.

gianluca-tursi, 2024-03-07:
> There's a chance you already have a phone or laptop that is already connected to the OBD2 scanner and won't let the ESP connect. Try turning off all other BT devices when using the ESP. Also, you [...]

i had tried the odb2 with my phone and it request the 1234 pin, then i cancelled the connection between phone and odb2. i tried different bauds (not all ).  do you know a link of a specific odb2 that works? 
other two question, there is a sketch to connect the odb2 with macaddress ? (i found only with odb2 name in your examples).

otherwise i have other odb2 with wifi connect..maybe i can try with that one

PowerBroker2, 2024-03-07:
I use [this one](https://www.amazon.com/gp/product/B005NLQAHS/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1) and it works like a charm

PowerBroker2, 2024-03-08:
If your OBD2 scanner is BLE, then it isn't supported by this lib, unfortunately

gianluca-tursi, 2024-03-10:
@PowerBroker2 i think that i'am connected, but i tried this example sketch: ESP32_Check_Voltage

and i see that warning in the serial monitor:

	Received char: >
Delimiter found.
All chars received: 11.3V
Battery Voltage: 11.30
ASSERT_WARN(1 8), in lc_task.c at line 1410ASSERT_WARN(1 8), in lc_task.c at line 1410Clearing input serial buffer
Sending the following command/query: AT RV

why that? 
thanks for your help

jimwhitelaw, 2024-03-10:
We’ve seen instances of that before. The ELMduino code is ok, but unfortunately it triggers a bug in an older version of the ESP-IDF (the framework library). See issue #226.

jimwhitelaw, 2024-03-27:
@gianluca-tursi I recently made a change that attempts to fix that ASSERT_WARN bug. I can't test it myself as I'm unable to reproduce that error with any of the ESP32 devices I have. Are you able to get the latest update and run your code again to see if the issue is resolved? (Note, the change is not in the latest release 3.2.5 as it's untested, so don't download that, clone or [download](https://github.com/PowerBroker2/ELMduino/archive/refs/heads/master.zip) the latest from master branch).

gianluca-tursi, 2024-03-28:
> @gianluca-tursi I recently made a change that attempts to fix that ASSERT_WARN bug. I can't test it myself as I'm unable to reproduce that error with any of the ESP32 devices I have. Are you able to [...]

yes i can do it, i need to download the branch from your link https://github.com/PowerBroker2/ELMduino/archive/refs/heads/master.zip then i need to substitute the file in my arduino library and compile the sketch?

jimwhitelaw, 2024-04-11:
@gianluca-tursi I meant to confirm this and forgot about it, sorry. Yes, your test plan is correct.

closed 2025-02-09 by PowerBroker2