PowerBroker2/ELMduino #235: custom pid for DPF

gianluca-tursi, 2024-03-10:
hello! thank for your work,

can you help me to create a sketch to read de DPF  Clogging?, i know that the information to obtain are the following but i don't know exactly to create the sketch
any help?

• DPF clogging:
-> PID = 2218E4
-> equation = ((A*256)+B)*(1000/65535)
-> OBD Header = DA10F1

gianluca-tursi, 2024-03-11:
hello i'm traing this code but i receive error on sending the headr:

`#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <TFT_eSPI.h> // Include the TFT_eSPI library for the display

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT
// #define DEBUG_PORT Serial // Comment or remove after debugging

ELM327 myELM327;
TFT_eSPI tft = TFT_eSPI();  // Create a tft object for the display
String obdName = "V-LINK";  // Variable for the OBD Bluetooth name

void setup() {
    Serial.begin(115200); // Comment or remove after debugging
    
    tft.init();  // Initialize the TFT display
    tft.setRotation(1);  // Set display orientation if needed
    tft.fillScreen(TFT_BLACK);   // Clear the screen
    tft.setTextFont(4);          // Set a larger font size for better readability
    tft.setTextColor(TFT_WHITE, TFT_BLACK);  // Set text and background color

    // Initial Bluetooth connection message
    tft.setCursor(0, 0);  // Reposition the cursor to the top left corner
    tft.println("Connecting to OBD-II");
    delay(3000);  // Wait for 3 seconds

    SerialBT.setPin("1234");
    ELM_PORT.begin("ArduHUD", true);

    if (!ELM_PORT.connect(obdName.c_str())) { // Use obdName variable
        tft.fillScreen(TFT_BLACK);
        tft.println("Couldn't connect to OBD scanner");
        while (1) delay(1000); // Stop the loop if unable to connect
    }

    if (!myELM327.begin(ELM_PORT, true, 2000)) {
        tft.fillScreen(TFT_BLACK);
        tft.println("OBD scanner connect failed");
        while (1) delay(1000); // Stop the loop if unable to connect
    }

    // Connected to Bluetooth
    tft.fillScreen(TFT_BLACK);
    tft.println("Connected to OBD-II");
    delay(3000);  // Wait for 3 seconds before proceeding

    // Clear screen for subsequent data display
    tft.fillScreen(TFT_BLACK);

  // Imposta l'intestazione OBD specifica per Alfa Romeo Stelvio
myELM327.sendCommand("AT SH DA10F1");

}

void loop() {
    // Verifica se l'intestazione è stata impostata correttamente
    if (myELM327.queryPID(0x01, 0x00)) { // Questa è una richiesta di prova per vedere se otteniamo una risposta
        Serial.println("Intestazione impostata, inviando comando custom...");
    } else {
        Serial.println("Errore nell'impostazione dell'intestazione OBD.");
    }

    // Invia il comando custom per ottenere i dati di interesse
    myELM327.sendCommand("2218E4");
    Serial.println("Comando inviato, attendendo la risposta...");

    delay(100); // Dà tempo alla risposta di arrivare

    // Controlla lo stato della risposta ricevuta
    if (myELM327.nb_rx_state == ELM_SUCCESS) {
        Serial.println("Risposta ricevuta con successo. Dati:");
        for (byte i = 0; i < myELM327.recBytes; i++) {
            Serial.print(myELM327.payload[i], HEX);
            Serial.print(" ");
        }
        Serial.println();
    } else if (myELM327.nb_rx_state == ELM_GETTING_MSG) {
        Serial.println("In attesa della risposta...");
    } else {
        Serial.println("Nessuna risposta valida ricevuta.");
        // Mostra il codice di errore per il debugging
        Serial.print("Codice di errore: ");
        Serial.println(myELM327.nb_rx_state);
    }

    delay(5000); // Attendi prima di inviare una nuova richiesta
}
`

gianluca-tursi, 2024-03-12:
i'm traing different code without success...

`void setup {
....
myELM327.sendCommand("AT SH DA10F1");
....
}

void loop() {
  myELM327.sendCommand("2218E4"); 
  delay(100); // Questo valore può variare a seconda del modulo OBD-II

  // Controlla lo stato della risposta
  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    Serial.println("Risposta ricevuta. Dati grezzi:");

    // Itera attraverso i byte ricevuti e li stampa sul monitor seriale
    for (byte i = 0; i < myELM327.recBytes; i++) {
      Serial.print(myELM327.payload[i], HEX);
      Serial.print(" ");
    }
    Serial.println(); // Vai a capo dopo l'ultimo byte
  } else if (myELM327.nb_rx_state == ELM_GETTING_MSG) {
    Serial.println("In attesa della risposta...");
  } else {
    Serial.println("Errore nella ricezione della risposta o nessun dato ricevuto.");
  }

  // Attendi prima di inviare una nuova richiesta
  delay(5000); // Aumenta o diminuisci secondo le necessità
}
`

without success...if someone can help

gianluca-tursi, 2024-03-13:
hello! i'm going ahaed with my test waiting for your help...

i have tested a custom pid with an ios app with confirmed result, so i know that works well with my car.
this is the pid
#07 Gearbox oil temperature
OBD Mode&PID -- 2204fe
Unit Type -- ℃
Equation -- A-40
OBD Header -- DA18F1

on the app i see this result:
ELM: 18DAF118046204FE52 
A:52
result: 42°

i'm trying to read data from odb with two strategy: processpid and sendCommand, with no success, here is the loop code, 

void loop() {
      myELM327.sendCommand("ATSH DA18F1");

  // Imposta l'header OBD, se necessario
  delay(100); // Dà tempo per elaborare il comando

  // Tentativo con processPID
  float gearboxOilTemp = myELM327.processPID(0x22, 0x04FE, 1, 7, 1.0, 0.0);
    delay(2000); // Dà tempo alla risposta di arrivare

  if (myELM327.nb_rx_state == ELM_SUCCESS) {
   int A = myELM327.payload[6]; // Prende il valore A dalla risposta
    float temperature = A - 40; // Applica la formula per ottenere la temperatura
        displayTemperature(temperature);

  } else {
    // Se processPID fallisce, prova con sendCommand
    myELM327.sendCommand("2204FE");
    delay(2000); // Dà tempo alla risposta di arrivare

    if (myELM327.nb_rx_state == ELM_SUCCESS) {
      // Estrai il byte dei dati dalla risposta
      int dataByteIndex = myELM327.recBytes - 1; // L'ultimo byte dei dati
      int A = myELM327.payload[dataByteIndex];
      float temperature2 = A - 40; // Applica la formula per ottenere la temperatura

      // Mostra la temperatura sul TFT
      displayTemperature(temperature2);
      delay(100); // Dà tempo alla risposta di arrivare

    } else {
      // Se anche sendCommand fallisce, mostra l'errore sul TFT
      displayError();
      myELM327.printError(); // Stampa l'errore sul monitor seriale per il debug
    }
  }

  delay(5000); // Attendere 5 secondi prima della prossima query
}

void displayTemperature(float temperature) {
  tft.fillScreen(TFT_BLACK);
  tft.setCursor(0, 0);
  tft.setTextColor(TFT_WHITE, TFT_BLACK);
  tft.setTextSize(2);
  tft.print("Temperatura: ");
  tft.print(temperature);
  tft.println(" °C");
}

void displayError() {
  tft.fillScreen(TFT_BLACK);
  tft.setCursor(0, 0);
  tft.setTextColor(TFT_RED, TFT_BLACK);
  tft.setTextSize(2);
  tft.println("Errore nella lettura del PID");
}

jimwhitelaw, 2024-03-18:
Ok, that's not quite the correct way to query in a loop. First, for a simple sketch, I'd move the header setting into your _setup()_ method and use _sendCommandBlocking()_:

```C
sendCommandBlocking("ATSH DA18F1");
```
Then remove the delays from the loop and restructure it:
```C
void loop() {
if (nb_query_state == SEND_COMMAND)
{
    myELM327.sendCommand("2204FE");
    nb_query_state = WAITING_RESP;
}
else if (nb_query_state == WAITING_RESP)
{
    get_response();
}

if (myELM327.nb_rx_state == ELM_SUCCESS) 
{ 
    nb_query_state = SEND_COMMAND;
    int A = myELM327.payload[6]; // Prende il valore A dalla risposta
    float temperature = A - 40; // Applica la formula per ottenere la temperatura
    displayTemperature(temperature);
}
 
else if (nb_rx_state != ELM_GETTING_MSG)
{
    nb_query_state = SEND_COMMAND; 
    displayError();
    myELM327.printError();
}
```

jimwhitelaw, 2024-03-19:
You are right on item 1, that's the correct way to call the method (same for other ELMduino methods used). Note that the code I posted is just a snippet to get you going, not the complete sketch. It looks like your sketch has not included the Can you post the full sketch that you're using ([formatted please](https://github.com/meshtastic/firmware/blob/711b85cfe80c10ac79b16864c59956552ca8321e/src/mesh/MeshService.cpp#L6))?

gianluca-tursi, 2024-03-19:
hello, i'm not a developer i'm just had a arduino training some years ago...so i found so hard to get some custom pid because there is not example in the library, instead for the standard pid starting fromthe exempla i did it well.

so thank you for your help, here is the full code:

``
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <TFT_eSPI.h> // Include the TFT_eSPI library for the display

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT

ELM327 myELM327;
TFT_eSPI tft = TFT_eSPI();  // Create a tft object for the display
String obdName = "V-LINK";  // Variable for the OBD Bluetooth name

int nb_query_state = SEND_COMMAND; // Variabile di stato inizializzata a SEND_COMMAND

void setup() {
    Serial.begin(115200); // Comment or remove after debugging
    
    tft.init();  // Initialize the TFT display
    tft.setRotation(1);  // Set display orientation if needed
    tft.fillScreen(TFT_BLACK);   // Clear the screen
    tft.setTextFont(4);          // Set a larger font size for better readability
    tft.setTextColor(TFT_WHITE, TFT_BLACK);  // Set text and background color

    // Initial Bluetooth connection message
    tft.setCursor(0, 0);  // Reposition the cursor to the top left corner
    tft.println("Connecting to OBD-II");
    delay(3000);  // Wait for 3 seconds

    SerialBT.setPin("1234");
    ELM_PORT.begin("ArduHUD", true);

    if (!ELM_PORT.connect(obdName.c_str())) { // Use obdName variable
        tft.fillScreen(TFT_BLACK);
        tft.println("Couldn't connect to OBD scanner");
        while (1) delay(1000); // Stop the loop if unable to connect
    }

    if (!myELM327.begin(ELM_PORT, true, 2000)) {
        tft.fillScreen(TFT_BLACK);
        tft.println("OBD scanner connect failed");
        while (1) delay(1000); // Stop the loop if unable to connect
    }

    // Connected to Bluetooth
    tft.fillScreen(TFT_BLACK);
    tft.println("Connected to OBD-II");
    delay(3000);  // Wait for 3 seconds before proceeding

    // Clear screen for subsequent data display
    tft.fillScreen(TFT_BLACK);

  // Imposta l'intestazione OBD specifica per Alfa Romeo Stelvio
    myELM327.sendCommand_Blocking("ATSH DA18F1");

}

void loop() {
if (nb_query_state == SEND_COMMAND)
{
    myELM327.sendCommand("2204FE");
    nb_query_state = WAITING_RESP;
}
else if (nb_query_state == WAITING_RESP)
{
  get_response();
}
  // Clear screen for subsequent data display
    //tft.fillScreen(TFT_BLACK);
if (myELM327.nb_rx_state == ELM_SUCCESS) 
{ 
    Serial.println("success");
    nb_query_state = SEND_COMMAND;
    int A = myELM327.payload[6]; // Prende il valore A dalla risposta
    float temperature = A - 40; // Applica la formula per ottenere la temperatura
    Serial.println(temperature);
}
 
else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
{
    nb_query_state = SEND_COMMAND; 
    //displayError();
    myELM327.printError();
}
}
``

jimwhitelaw, 2024-03-26:
OK, I took a quick look at this for you and made a couple changes. It compiles without error but I can't test it as I currently don't have access to a TFT panel. 

```C
#include "BluetoothSerial.h"
#include "ELMduino.h"
#include <TFT_eSPI.h> // Include the TFT_eSPI library for the display

BluetoothSerial SerialBT;
#define ELM_PORT SerialBT

ELM327 myELM327;
TFT_eSPI tft = TFT_eSPI();  // Create a tft object for the display
String obdName = "V-LINK";  // Variable for the OBD Bluetooth name

int nb_query_state = SEND_COMMAND; // Variabile di stato inizializzata a SEND_COMMAND

uint8_t ctoi(uint8_t value)
{
    if (value >= 'A')
        return value - 'A' + 10;
    else
        return value - '0';
};

void setup() {
    Serial.begin(115200); // Comment or remove after debugging
    
    tft.init();  // Initialize the TFT display
    tft.setRotation(1);  // Set display orientation if needed
    tft.fillScreen(TFT_BLACK);   // Clear the screen
    tft.setTextFont(4);          // Set a larger font size for better readability
    tft.setTextColor(TFT_WHITE, TFT_BLACK);  // Set text and background color

    // Initial Bluetooth connection message
    tft.setCursor(0, 0);  // Reposition the cursor to the top left corner
    tft.println("Connecting to OBD-II");
    delay(3000);  // Wait for 3 seconds

    SerialBT.setPin("1234");
    ELM_PORT.begin("ArduHUD", true);

    if (!ELM_PORT.connect(obdName)) { // Use obdName variable
        tft.fillScreen(TFT_BLACK);
        tft.println("Couldn't connect to OBD scanner");
        while (1) delay(1000); // Stop the loop if unable to connect
    }

    if (!myELM327.begin(ELM_PORT, true, 2000)) {
        tft.fillScreen(TFT_BLACK);
        tft.println("OBD scanner connect failed");
        while (1) delay(1000); // Stop the loop if unable to connect
    }

    // Connected to Bluetooth
    tft.fillScreen(TFT_BLACK);
    tft.println("Connected to OBD-II");
    delay(3000);  // Wait for 3 seconds before proceeding

    // Clear screen for subsequent data display
    tft.fillScreen(TFT_BLACK);

  // Imposta l'intestazione OBD specifica per Alfa Romeo Stelvio
    myELM327.sendCommand_Blocking("ATSH DA18F1");

}

void loop() {
    if (nb_query_state == SEND_COMMAND)
    {
        myELM327.sendCommand("2204FE");
        nb_query_state = WAITING_RESP;
    }
    else if (nb_query_state == WAITING_RESP)
    {
        myELM327.get_response();
        if (myELM327.nb_rx_state == ELM_SUCCESS) 
        { 
            Serial.println("success");
            nb_query_state = SEND_COMMAND;
            int A = (ctoi(myELM327.payload[6]) << 4) | ctoi(myELM327.payload[7]); 
            float temperature = float (A - 40); // Applica la formula per ottenere la temperatura
            Serial.println(temperature);
        }
        
        else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
        {
            nb_query_state = SEND_COMMAND; 
            //displayError();
            myELM327.printError();
        }
    }
}
```

jimwhitelaw, 2024-03-28:
Hey! I'm glad to hear it's working for you now. For the sketch above, I have a few observations: 

- I don't think adding a delay in every iteration of your loop will be helpful. If it's needed at all, it would make more sense to do so after a query is complete before starting the next one.
- The issue seen now could be related to how the header is being set with _myELM327.sendCommand_Blocking("ATSH DA10F1")_. That header requests a specific ECU in the system to respond. Are you sure that is the correct setting for each of the commands being sent? 
- If the second and third queries do need to have a different header set then the best place to do so would be when the _queryState_ is updated, so it's ready for the next query. Either set it to defaults, or the new required header. Example:

```C
Serial.println(lastDpfClogging);
displayData(); // Aggiorna i dati sul display
queryState = SEND_DISTANCE_SINCE_DPF;
sendCommandBlocking("ATD"); //reset to defaults
```

gianluca-tursi, 2024-03-28:
> * That header requests a specific ECU in the system to respond. Are you sure that is the correct settin

is the three pids have the same header:

• DPF clogging:
-> PID = 2218E4
-> equation = ((A*256)+B)*(1000/65535)
-> OBD Header = DA10F1

• DPF regeneration process:
-> PID = 22380B
-> equation = ((A*256)+B)*(100/65535)
-> OBD Header = DA10F1

• Distance covered from last DPF regeneration:
-> PID = 223807
-> equation = ((A*65536)+(B*256)+C)*0.1
-> OBD Header = DA10F1

the strange fact is that for the first two iterations the pid responds well, on the 3rd with timeout... also the sketch works well of each iteration with the bluetooth obd2 :|

what can i test?

gianluca-tursi, 2024-03-28:
here is the serial monitor

```
Sending command for Distance Since DPF Regeneration...
Clearing input serial buffer
Sending the following command/query: 223807
Timeout detected with overflow of 0ms
Error receiving Distance Since DPF Regeneration data.
Received: 
ERROR: ELM_TIMEOUT
Sending command for Distance Since DPF Regeneration...
Clearing input serial buffer
Sending the following command/query: 223807
```

jimwhitelaw, 2024-03-29:
To help people with similar queries, I have added a new example sketch to show how to use custom headers and PID. [e81f308](https://github.com/PowerBroker2/ELMduino/commit/e81f308098188baf2e51f1f32a94cd3783377c7f)

It is now on the latest master branch.

jimwhitelaw, 2024-03-29:
> the strange fact is that for the first two iterations the pid responds well, on the 3rd with timeout... also the sketch works well of each iteration with the bluetooth obd2 :|

Can you post the full serial output with debugging on so we can see the context of what happens?

gianluca-tursi, 2024-03-29:
> > the strange fact is that for the first two iterations the pid responds well, on the 3rd with timeout... also the sketch works well of each iteration with the bluetooth obd2 :|
> 
> Can you post the full serial output with debugging on so we can see the context of what happens?

yes, here the full serial monitor:

```
Connecting to V-LINK
.
Connected to Wifi
IP address: 
192.168.0.150
Clearing input serial buffer
Sending the following command/query: AT D
	Received char: A
	Received char: T
	Received char: _
	Received char: D
	Received char: \r
	Received charClearing input serial buffer
Sending the following command/query: AT AL
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT ST 00
Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: AT SP A0
	Received char: O
	Received char: K
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: OK
Clearing input serial buffer
Sending the following command/query: 0100
	Received char: S
	Received char: E
Received char: A
	Received char: R
	Received char: C
	Received char: H
	Received char: I
	Received char: N
	Received char: G
	Received char: .
	Received char: .
	Received char: .
	Received char: \r
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: 0
	Received char: 9
	Received char: 8
	Received char: 1
	Received char: 8
	Received char: 0
[... 182 lines cut ...]
	Received char: 0
	Received char: 0
	Received char: 0
	Received char: \r
	Received char: \r
	Received char: >
Delimiter found.
All chars received: 62380B0000
DPF Regeneration Process response received.
0.00
Sending command for DPF Clogging...
Clearing input serial buffer
Sending the following command/query: 2218E4
Timeout detected with overflow of 0ms
Error receiving DPF Clogging data.
Received: 
ERROR: ELM_TIMEOUT
errore conto2Sending command for DPF Clogging...
Clearing input serial buffer
Sending the following command/query: 2218E4
Timeout detected with overflow of 0ms
Error receiving DPF Clogging data.
Received: 
ERROR: ELM_TIMEOUT
```

gianluca-tursi, 2024-04-01:
i also tested the simple custom pid with 1 pid and without tft library, with bluetooth connection and odb vgate pro bluetooth classic works well, but wgate wifi and wifi connection do this error described above...i don't know what else to test...

jimwhitelaw, 2024-04-03:
I don't have a lot of experience with ELM327 via wifi - enough to know that a few of my tests work, but that's about it. It looks like there's some kind of timing issues that are occurring. You could try adding a small delay (100-200ms) after you change query_state in each switch case. The issue you're seeing here clearly hardware specific and not directly related to the ELMduino library itself.

closed 2025-02-09 by PowerBroker2