PowerBroker2/ELMduino #238: BT Pairing problem between ESP-32 and ELM327

paulhth, 2024-03-26:
Hello, 

At the moment I'm trying to implement a car monitoring system which would take the usual car parameters like current speed, consumption and rpm (or others) and display them on a webserver hosted by the ESP-32 through SPIFFS. The link between the ESP-32 and ELM327 (OBDII to BT) is made via Bluetooth, so naturally I wanted to test only the connection between the ESP-32 and the ELM using the example code you provided (as the webserver is running fine).

So far I have tried flashing the ESP with the examples from **"examples/ESP32_test"** and **"examples/ESP32_Bluetooth_Serial"** but none of them are connecting to the ELM327. I also bought 2 other ESP-32 boards thinking it may be a hardware problem but it doesn't work with them either. So the outcome is the same on 3 different ESP-32s. Definitely not an ESP problem.

I am attaching the output on both of the files tested:

//////////////////////////////////ESP32_Bluetooth_Serial.ino:
--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
etsJn 821 00:22:5
confgsip: 0 PIP0ebot01 (SI_FAST_FAHBOT
clkdrv:0x0,_r:x0,d_drv0x00,s_r:x0,hd_drv0x00w_r:x0
load:0x3ff00ln1184
load:0x0700ln:1322
load04000,len:028 
enr x085e4        
**Couldn't connect to OBD scanner - Phase 1**

////////////////////////////////Esp32_test.ino:
--- Terminal on COM5 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
etsJn 8216 00:22:57

s:01(POWERON_RSET)bo:01 (SPI_FAST_FLS_OO)
configsip: ,SPW:xe
clk_drv:0x0q_r:000,d_drv:0x0,s_dv0x00,h_drv:0x0,p_r:000
mode:DIO lokdv:2
load:0x3f030ln:184
load:0x0700,en:13232
lod0x0840,len:3028
ety0x0805e4
Attempting to connect to ELM327...
**Couldn't connect to OBD scanner**

I also am adding photos with what ELM327 I'm using. One odd behaviour I found was that whenever I scanned using my laptop for Bluetooth devices, I always saw two instances of "OBDII" devices available (I suspect one is for input and one for output). They are named the same so I couldn't tell at first, but when I try to connect to one instance, it may either ask me to put a password (which is "1234" for the ELM) **and no less flash on the elm** OR if I connect to another one, it doesn't request the password but in a blink of an eye the green and blue leds flash on the ELM (the red one for power stays lit at all times). 

I'm afraid this messes with the connection of the ESP. I haven't tried yet to implement the [removal of paired devices](https://github.com/espressif/arduino-esp32/blob/master/libraries/BluetoothSerial/examples/bt_remove_paired_devices/bt_remove_paired_devices.ino) but this will be my next attempt. 

I would like to make some clarifications:

- The engine is running at every attempt (I tried it with only the ignition on, the same result)
- The ELM327 has the red LED on at all times (the green and blue are only flashing sometimes if I connect to it using my laptop - and there is a 50/50 chance if I get it right)
- If it matters, the car is a 2009 BMW 1 Series E87.
- At first I plug in the ELM, then start the car, then flash the ESP-32. After these, I look in the serial monitor hoping to see the rpm, but it says Couldn't Connect as you can see above.
- While flashing, I hold down the BOOT button (I tried without holding it too).
- The ELM doesn't have any kind of RST buttons.
- I specifically removed any BT devices from proximity, even the wireless mouse and kept them away while trying, so the band should be available.
- I also tried this in another place where let's say the 2.4 GHz wouldn't be overloaded with other networks.
- The next step in my attempt would be to try to remove the previously paired devices and if that doesn't work either, buy another ELM327.

_I am kindly asking for instructions or advice on what to do next, as this is a pretty important project for my finals. 
Thank you for reading this far and thank you for your attention!_ 

This is the ELM327:
[image]
[image]

jimwhitelaw, 2024-03-27:
It's possible that the two "OBDII" BT devices you see on your computer are one each for Classic Bluetooth and BLE. The example sketches you're trying will only work with Classic Bluetooth. Note that those sketches also have the setting of the BT PIN commented out by default, have you uncommented that code? (Have to ask, just in case...) 

https://github.com/PowerBroker2/ELMduino/blob/2c242bcf28a05ebc46c48f650b16611452b04695/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino#L23-L25

From the info you've provided, it appears you are building and flashing the EPS32 correctly and that the firmware is running and outputting serial debug info properly. Have you been able to establish a BT connection to the ELM327 using another device (laptop, phone, tablet)? That's a good next step, see if you can connect with an OBD app on a phone, like Torque(Android) or Car Scanner(iOS) and verify that the ELM device is good.

paulhth, 2024-03-28:
Yes, forgot to mention that I indeed tried the code with and without line 24 commented, with no success. I tried the Car Scanner app for iOS and to my surprise the ELM is good, the app successfully showed the rpm, speed and other parameters. It also had the green and blue LEDs blinking which is good. 

So the ELM327 is not the problem as I expected. What I was thinking about is the potential difference between the BT versions of the devices. When I paired the ELM to my phone I chose the Bluetooth 4.0+ version. To my knowledge, the ESP has BT classic version 4.2 so I don't see why should there be a problem between these. 

The next step I will be doing is test all of the combinations again, commenting and uncommenting line 24, and activating the removal of paired devices. Other than that, I have no other idea why it might be failing to connect. If you have any other ideas I can try please let me know 😊

wcjxixi, 2024-03-29:
It is recommended not to use this "bad" ELM327 adapter.

- [Choosing OBDII adapter ](https://www.carscanner.info/choosing-obdii-adapter/)
- [ How to buy proper ELM327 v1.5 OBD diagnostic interface](https://cvtz50.info/en/elm327/)

I've used many types of these ELM327s before with poor compatibility. You might get lucky and find one that works, but it's not stable either.

I'm now using the Vgate iCar Pro BLE, which I think is the best ELM327 adapter for the money!

weeblebiker, 2024-03-29:
or just get a macchina A0 and have an OBD dongle that is actually a CAN to ESP32 that you can code directly, so elmduino is not necessary. and no relying on some black box Elm327 dongle for a go between.

paulhth, 2024-03-31:
> It is recommended not to use this "bad" ELM327 adapter.
>
> * [Choosing OBDII adapter ](https://www.carscanner.info/choosing-obdii-adapter/)
> * [ How to buy proper ELM327 v1.5 OBD diagnostic [...]

But isn't the BLE not compatible with the ELMduino library? I'm assuming you're using something like this:
[image]

wcjxixi, 2024-03-31:
> But isn't the BLE not compatible with the ELMduino library? I'm assuming you're using something like this: [...]

Yes, that's what I'm using.

Yes, the ELMduino is not BLE compatible, but the Vgate iCar Pro BLE supports both BLE and classic Bluetooth.
Of course, you can also use the Vgate iCar Pro BT3.0 (which looks exactly like the Vgate iCar Pro BLE).

Also, if you're looking for a cheaper adapter, the KINGBOLEN® ELM327 Bluetooth OBD2 Scanner (which is supposed to be from the same manufacturer as the Vgate) is also fine according to my tests.

Maybe you have an Apple device, if you occasionally need to use it for something else. Or you also require it to have auto-sleep and auto-wake functions, so, Vgate iCar Pro BLE is the best value for money.

paulhth, 2024-04-05:
Hey, so I bought the one you recommended and thankfully it works, it successfully sends and receives data from the car. Thank you a lot for recommending this to me! The issue was that bad ELM I purchased at first.

closed 2024-04-05 by paulhth