PowerBroker2/ELMduino #240: How to change OBD protocol

reflexsa, 2024-04-09:
Firstly thanks for this, it was exactly what I was searching for.
I just wanted to ask how/where do I change the OBD protocol? I need to select "const char ISO_15765_11_BIT_500_KBAUD = '6';" to work reliably with my vehicle.
Apologies for the silly question!

PowerBroker2, 2024-04-10:
The elm should automatically find and use the best protocol, but if you want to specify it yourself, you can do so in the `begin()` function:

https://github.com/PowerBroker2/ELMduino/blob/e81f308098188baf2e51f1f32a94cd3783377c7f/src/ELMduino.cpp#L3-L23

reflexsa, 2024-04-10:
Thank you. For some reason when using my adapter with Torque for example I will get an 'engine check light' unless I specify the protocol. The actual error is communication to the gearbox so it seems that is somehow interfered with.
So would I just change the "const char &protocol" to 6 in the below?
`bool ELM327::begin(Stream &stream, const bool &debug, const uint16_t &timeout, const char &protocol, const uint16_t &payloadLen, const byte &dataTimeout)`

PowerBroker2, 2024-04-11:
You would have to add the argument in the sketch kind of like this:

```C++
myELM327.begin(ELM_PORT, true, 2000, ISO_15765_11_BIT_500_KBAUD)
```

reflexsa, 2024-04-11:
Thank you. I see, that does make sense. I'm hoping this will resolve my other issue where I just have a recurring response after the initial AT commands ([https://github.com/PowerBroker2/ELMduino/issues/241#issue-2236292600](url))

jimwhitelaw, 2024-04-18:
Fixed in [version 3.3.0](https://github.com/PowerBroker2/ELMduino/releases/tag/3.3.0).

closed 2024-04-18 by jimwhitelaw