PowerBroker2/ELMduino #243: Error in getting Multiple PID's

Shaikh010, 2024-04-14:
**Setup**
I have used esp 32 wroom and car is basically suzuki vitara.

**Error:**
I have used single PID example and it works perfectly but when I use Multiple PID's example I am getting error ElmTimout error 
screenshot is attached help me fix it kindly have studied all the issues but can't get my required one. 
Anybody kindly help me out.
[image]

jimwhitelaw, 2024-04-16:
Which specific examples are you using? Note that "ESP32_Bluetooth_Serial.ino" uses Bluetooth for a single PID test and "multiple_pids.ino" uses HW Serial for the multiple PID test. So they aren't directly interchangeable with the same adapter.

jimwhitelaw, 2024-04-18:
@Shaikh010, I have added a [new example sketch](https://github.com/PowerBroker2/ELMduino/commit/5d041617e79bad7487e69f722f8719f9f6ef4155) that queries multiple PIDs using the same type of BT connection as the "ESP32_Bluetooth_Serial.ino" sketch. Not sure if that's the issue you're seeing, but it may help others in the future.

Bakar002, 2024-04-21:
@jimwhitelaw I have esp 32 module and obd 2 sensor it is showing correct data when using single Pid function but it shows time out error when i use multiple pid example code .
i just simply need multiple pids like coolant temp, engine temp, speed, rpm, abs load etc how may i get this data ?

jimwhitelaw, 2024-04-21:
@Bakar002 Are you using the multiple PID sketch I linked to above? If not, please post (with formatting) the sketch you are using and post the full serial/debug output to help us see what is happening.

Bakar002, 2024-05-17:
@jimwhitelaw Have tried you new code why it is giving me wrong data and also give data in a very fast manner.Like when i connect it with esp 32 it connects but when it connects it gives serial data in a very fast manner and give wrong data

jimwhitelaw, 2024-05-18:
@Bakar002 please post the serial output so we can see what's happening. A fast response isn't an indication that something is wrong, unless every query results in '?' or 'NO DATA'.

Bakar002, 2024-06-21:
@jimwhitelaw  I am testing in Aqua and it mostly give me wrong readings why and its speed of response is also very fast. and also guide me that I am getting data through bluetooth and now how to get this data to any server or cloud because i have heard that both antennas dont work

jimwhitelaw, 2024-06-22:
I don't have any experience with Aqua. I'm aware that esp32 shares a radio between Bluetooth and WiFi, but I haven't had any issues with using both at the same time in any of my projects.

Bakar002, 2024-06-22:
@jimwhitelaw 
Thank u very much 
Can you kindly share a code file or repository with me which in which we uses both radio at a same time 
My Email: abubakaramjad78600@gmail.com

closed 2025-02-09 by PowerBroker2