PowerBroker2/ELMduino #246: ESP32 s3 BLE compatibility?

weeblebiker, 2024-04-25:
Classic bluetooth isn't supported with the newer ESP32 S3s. they don't have the hardware.

Is Elmduino compatible with ESP32 BLE libraries?
BLEDevice.h
BLEUtils.h
BLEScan.h
BLEAdvertisedDevice.h

or does it work only with older ESP32s and BluetoothSerial.h

I have a ELM327 BLE dongle and ESP32 S3 based lilygo T-display S3 I'd like to work together 

I didn't find any BLE examples

PowerBroker2, 2024-04-25:
Sorry, but BLE is not supported. You could opt for a WiFi OBD2 scanner and take advantage of the ESP's WiFi interface instead.

closed 2024-04-25 by PowerBroker2

jimwhitelaw, 2024-04-25:
@weeblebiker, user @DerKleinePunk has [created an app](https://github.com/DerKleinePunk/ODB-ESP32-Reader/tree/main) using ELMduino with BLE. I haven't tried it myself, but perhaps there is something there may be of some use for you.

DerKleinePunk, 2024-04-26:
No the BLE ist also not Working Currently