PowerBroker2/ELMduino #247: Attempting to read non-standard PID

Macko92, 2024-04-27:
Hello,
First of all, congratulations to the author - great job with this library. I was able to connect my esp32 project to the car without any major problems. I can read many standard PIDs such as engine speed, engine temperature, etc.

I am wondering how I could read several parameters that are specific to my car: 2003 MG ZR 160, communication via k-line (ISO 9141& ISO 14230).

For example: oil temperature, which in my case is not available under the standard PID 5C.
I managed to get some information about communication with my ECU to read such parameters, the question is how can I convert this structure to be able to read it using e.g. processPID()?

Some data:
```
service $61 data $00 set word 1 3132  / Engine Coolant Temperature (In 0.1°K) 
service $61 data $0c set word 5 3100  / Injector 1 Pulse Width (In μs) 
service $61 data $25 set word 1 2250 / VVC Inlet Cam Duration Requested (In 0.1°) 
service $61 data $25 set word 3 2370 / VVC Inlet Cam Duration Measured (In 0.1°)                
```

Can someone help me with this case?

jimwhitelaw, 2024-04-28:
I don't any specific knowledge of that system, but I can offer some suggestions that might help you get started. 

The example sketch [_ESP32\_CustomHeader.ino_](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_CustomHeader/ESP32_CustomHeader.ino) shows how to create a query using a custom PID and extracting the desired data from the raw response. It's hard to know exactly how your query should be structured from that info, but I bet a bit of experimentation would get you there.

Perhaps try combining some variants of the "data" and set word values provided with service 61 and go from there. 
IE, "6100" , "610C", "613132" etc and see what kind of responses you get.

The 00, 0C, 25, 25 values might indicate a specific controller in the system. You'd address those with a custom header as in the same example above - with the "ATSH" command before your query.

I'd try running a few test queries and look at the data (if any) that gets returned and see if I could make anything of that.

closed 2025-02-09 by PowerBroker2

jremm15, 2025-05-03:
Hello Macko92,

Any chance you could share your code?  I have been trying to connect to a Kline vehicle with no success.  I have been using a hardwired connection between the ELM327 and Arduino rather than Bluetooth.

Thanks and hopefully you see this!