PowerBroker2/ELMduino #250: How to read data if I know TXD RXF RXD and MTH

BCProgrammingTH, 2024-05-25:
I want to read the transmission fluid temperature in my Subaru XV, but I don't know how to take the values. How to put it in the reading process to be complete?

Hardware
- ESP32
- ELM327

[
[image]
](url)

PowerBroker2, 2024-05-25:
Here are a few links that might help:

- [This shows what the PID is](https://www.subaruoutback.org/posts/4345441/)
- [This shows how to do the response processing](https://www.subaruoutback.org/posts/2867993/)

BCProgrammingTH, 2024-05-26:
I am a beginner I'm not sure if my use of processPID is correct. I don't quite understand the RXD part of how to use it with processPID.

RXD: 3008

#include "ELMduino.h"
#define DEBUG_PORT Serial
ELM327 myELM327;
int nb_query_state = SEND_COMMAND; // Set the inital query state ready to send a command

void setup()
{
  DEBUG_PORT.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, 16, 17);

  if (!myELM327.begin(Serial2, true, 3000))
  {
    Serial.println("ELM327 device couldn't connect to ECU.");
    while (1);
  }

  Serial.println("Connected to ELM327");
  // Set a custom header using ATSH command.
  myELM327.sendCommand_Blocking("AT SH 7E1");
}
void loop()
{
  float temp = myELM327.processPID(34, 4119, 1, 1, 9 / 5, -58);
  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    Serial.print("Temp: "); Serial.println(temp);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}

[image]

closed 2024-05-30 by BCProgrammingTH