PowerBroker2/ELMduino #251: ERROR: ELM_BUFFER_OVERFLOW on Honda Vezel 2023

sbudiman12, 2024-05-26:
Im building a LED Shiftlight for my car (2023 Honda Vezel/ HRV) using ESP32 and ELM327. The device works great on my other car (2023 Toyota Fortuner) but when tested on my Honda it miscalculated the RPM received. 

HRV Vezel protocol from https://play.google.com/store/apps/details?id=com.ovz.carscanner
[image]

Tested with modified
https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_Serial/ESP32_Bluetooth_Serial.ino

```c
#include "BluetoothSerial.h"
#include "ELMduino.h"

BluetoothSerial SerialBT;
#define ELM_PORT   SerialBT
#define DEBUG_PORT Serial

ELM327 myELM327;

uint32_t rpm = 0;
//MAC address
uint8_t address[6] = {0x00, 0x10, 0xCC, 0x4F, 0x36, 0x03}

void setup()
{
#if LED_BUILTIN
  pinMode(LED_BUILTIN, OUTPUT);
  digitalWrite(LED_BUILTIN, LOW);
#endif

  DEBUG_PORT.begin(115200);
  SerialBT.setPin("1234");
  ELM_PORT.begin("ArduHUD", true);
  //connect with MAC
  if (!ELM_PORT.connect(address))
  {
    DEBUG_PORT.println("Couldn't connect to OBD scanner - Phase 1");
    while(1);
  }

  if (!myELM327.begin(ELM_PORT, true, 2000))
  {
    Serial.println("Couldn't connect to OBD scanner - Phase 2");
    while (1);
  }

  Serial.println("Connected to ELM327");
}

void loop()
{
  float tempRPM = myELM327.rpm();

  if (myELM327.nb_rx_state == ELM_SUCCESS)
  {
    rpm = (uint32_t)tempRPM;
    Serial.print("RPM: "); Serial.println(rpm);
  }
  else if (myELM327.nb_rx_state != ELM_GETTING_MSG)
    myELM327.printError();
}

``` 

Output in Serial Monitor:

```
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
        Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: A
	Received char: F
	Received char: C
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: \r
	Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: B
	Received char: 0
	Received char: 0
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: \r
        Received char: 4
	Received char: 1
	Received char: 0
	Received char: C
	Received char: 0
	Received char: B
	Received char: 0
	Received char: 0
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
	Received char: 5
OBD receive buffer overflow (> 40 bytes)
Received: 410C0AFC555555410C0B00555555410C0B005555
ERROR: ELM_BUFFER_OVERFLOW
``` 

It is supposed to be an RPM value, and it doesn't just affect my Honda Vezel but it affects every Honda vehicle I've tried so far. Why is this and can someone help me resolve this so my device will work on Honda vehicles? Thank you.

PowerBroker2, 2024-05-26:
Instead of

```C++
myELM327.begin(ELM_PORT, true, 2000)
```

Try increasing the payload buffer length by doing the following:

```C++
myELM327.begin(ELM_PORT, true, 2000, '0', 100)
```

PowerBroker2, 2024-05-28:
For whatever reason, your car reports the RPM in bursts of 3 with trailing 0x55s. I have no clue why it's doing that and nobody has had this issue before.

You might need to do the following as a workaround:

```C++
int32_t A = myELM327.responseByte_4;
int32_t B = myELM327.responseByte_3;
float rpm = 0;

rpm = ((256 * A) + B) / 4.0;
```

I may have mixed up which response byte is A and which is B. If you don't get accurate RPM values, try flipping A and B.

sbudiman12, 2024-06-06:
Dear @PowerBroker2 
I tried your method and made a few tweaks of my own, and it worked on Honda vehicles. Thank you so much for your time and help. Without you mentioning the 'responseByte' method, my project wouldn't have been a success.

closed 2024-06-06 by sbudiman12