PowerBroker2/ELMduino #252: Problems in phase 1, can't connect with ELM327

smesquita08, 2024-05-27:
Hi, I noticed that all the examples start with bluetooth communication and then communication with the ELM327. My problem arises right at the initial Bluetooth connection stage. I have already tried connecting using the name and MAC Address but without success. Is it necessary for the ELM327 to receive information within a certain time, otherwise it disconnects? When I try to connect with the MAC Address, the ELM327 flashes green and blue lights, do you know what this could mean?

PowerBroker2, 2024-05-28:
Ensure that no other device is connected wirelessly to the ELM327, the car is on and running, that you are using the correct baud, and that the ELM327 you're using is standard bluetooth, not BLE. Double checking your ELM327's bluetooth device name and/or MAC would be a good idea, too. Not sure about your other questions.

smesquita08, 2024-05-31:
Ok, I did some tests and my ELM327 is BLE. So, is the possibility of using your library ruled out? Is it possible to connect ESP32-ELM327 via BLE and then use commands from your library to retrieve the values ​​of certain parameters (rpm, speed, ...)?

PowerBroker2, 2024-05-31:
Sorry, this lib does not support BLE. I would suggest getting a classic bluetooth obd2 scanner and trying with that device