PowerBroker2/ELMduino #253: couldn't connect to OBD scanner

mohamedsadok7, 2024-06-15:
Hi! great work here. for my project, i'm using the OBDLink SX as my OBD2 cable ( i'm pretty sure it does support ELM327 and work with same protocols) and i'm connecting it to USB host shield that is mounted on my arduino UNO (i need the USB host because the arduino board doesn't have a USB port). Now when i upload the usb host shield code it works perfectly but when i upload the slightly altered multiple PIDs code, this is what the serial monitor displays:
`Attempting to connect to ELM327...
Clearing input serial buffer
Sending the following command/query: AT D
AT D
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT Z
AT Z
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT E0
AT E0
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT S0
AT S0
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT AL
AT AL
Timeout detected with overflow of 1ms
Clearing input serial buffer
Sending the following command/query: AT ST 00
AT ST 00
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT SP A0
AT SP A0
Timeout detected with overflow of 0ms
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
AT SP 0
Timeout detected with overflow of 0ms
Setting protocol via AT SP %c did not work
Couldn't connect to OBD scanner
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
010C1
Timeout detected with overflow of 1000ms
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
010C1
Timeout detected with overflow of 0ms
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
010C1
Timeout detected with overflow of 1ms
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
`
this is my code : 
`#include "ELMduino.h"

#define ELM_PORT Serial

const bool DEBUG = true;
const int TIMEOUT = 2000;
const bool HALT_ON_FAIL = false;

ELM327 myELM327;

void setup() {
  Serial.begin(115200);
  ELM_PORT.begin(115200);

  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT, DEBUG, TIMEOUT)) {
    Serial.println("Couldn't connect to OBD scanner");

    if (HALT_ON_FAIL) {
      while (1);
    }
  }

  Serial.println("Connected to ELM327");
}

void loop() {
  float coolantTemp = myELM327.engineCoolantTemp();
  float rpm = myELM327.rpm();
  float speed = myELM327.kph();

  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    Serial.print("Temperature: ");
    Serial.println(coolantTemp);
    Serial.print("RPM: ");
    Serial.println(rpm);
    Serial.print("Speed: ");
    Serial.println(speed);
    
    // Send data in a single line for easy parsing
    Serial.print("DATA,");
    Serial.print(coolantTemp);
    Serial.print(",");
    Serial.print(rpm);
    Serial.print(",");
    Serial.println(speed);
  } else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    myELM327.printError();
  }

  delay(1000); // Adjust delay as needed
}
`
i even tried the software serial test and it gave me pretty much the same output . am i missing something here ?

PowerBroker2, 2024-06-16:
Please reformat your post properly

mohamedsadok7, 2024-06-16:
this is what the serial monitor displays:
 
 `Attempting to connect to ELM327...
  Clearing input serial buffer 
  Sending the following command/query: AT D 
  AT D 
  Timeout detected with overflow of 0ms 
  Clearing input serial buffer 
  Sending the following command/query: AT Z 
  AT Z 
  Timeout detected with overflow of 0ms 
  Clearing input serial buffer 
  Sending the following command/query: AT E0 
  AT E0 
Timeout detected with overflow of 0ms 
Clearing input serial buffer 
Sending the following command/query: AT S0 
AT S0 
Timeout detected with overflow of 0ms 
Clearing input serial buffer 
Sending the following command/query: AT AL 
AT AL 
Timeout detected with overflow of 1ms 
Clearing input serial buffer 
Sending the following command/query: AT ST 00 
AT ST 00 
Timeout detected with overflow of 0ms 
Clearing input serial buffer 
Sending the following command/query: AT SP A0 
AT SP A0 Timeout detected with overflow of 0ms 
Setting protocol via AT TP A%c did not work - trying via AT SP %c 
Clearing input serial buffer 
Sending the following command/query: AT SP 0 
AT SP 0 
Timeout detected with overflow of 0ms 
Setting protocol via AT SP %c did not work 
Couldn't connect to OBD scanner 
Query string: 010C1 
Clearing input serial buffer 
Sending the following command/query: 010C1 
010C1 
Timeout detected with overflow of 1000ms 
Service: 1 
PID: 12 
Normal length query detected 
Query string: 010C1 
Clearing input serial buffer 
Sending the following command/query: 010C1 
010C1 
Timeout detected with overflow of 0ms Service: 1 
PID: 12 
Normal length query detected 
Query string: 010C1 
Clearing input serial buffer 
Sending the following command/query: 010C1 
010C1 
Timeout detected with overflow of 1ms 
Service: 1 
PID: 12 
Normal length query detected 
Query string: 010C1 
Clearing input serial buffer 
Sending the following command/query: 010C1`

and this is the code that i'm working on (multiple PIDs but slightly altered):

`#include "ELMduino.h"

#define ELM_PORT Serial

const bool DEBUG = true;
const int TIMEOUT = 2000;
const bool HALT_ON_FAIL = false;

ELM327 myELM327;

void setup() {
Serial.begin(115200);
ELM_PORT.begin(115200);

Serial.println("Attempting to connect to ELM327...");

if (!myELM327.begin(ELM_PORT, DEBUG, TIMEOUT)) {
Serial.println("Couldn't connect to OBD scanner");

if (HALT_ON_FAIL) {
  while (1);
}
}

Serial.println("Connected to ELM327");
}

void loop() {
float coolantTemp = myELM327.engineCoolantTemp();
float rpm = myELM327.rpm();
float speed = myELM327.kph();

if (myELM327.nb_rx_state == ELM_SUCCESS) {
Serial.print("Temperature: ");
Serial.println(coolantTemp);
Serial.print("RPM: ");
Serial.println(rpm);
Serial.print("Speed: ");
Serial.println(speed);

// Send data in a single line for easy parsing
Serial.print("DATA,");
Serial.print(coolantTemp);
Serial.print(",");
Serial.print(rpm);
Serial.print(",");
Serial.println(speed);
} else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
myELM327.printError();
}

delay(1000); // Adjust delay as needed
}
`
there is a problem with the connection between the arduino and the ELM327, like it detects that there is an ELM327 device but in can't communicate with it, and i can't reaally point out the problem

jimwhitelaw, 2024-06-16:
Your sketch has the debug port and ELM PORT both designated as Serial, which is the UNO onboard serial port. So the code is trying to connect to the ELM on the same port. 

Your serial shield will likely be defined by its own TX and RX pins. Work that out and then try the simple serial sketch to confirm a connection.

mohamedsadok7, 2024-06-18:
i tried using the software serial example and created digital TX and RX for the arduino (i removed the shield ) and connected my board to an USB to ttl adapter(FT232BL) and connected that to my OBDLink SX . this is the new arduino code :

#include "ELMduino.h"
#include <SoftwareSerial.h>

// Define the pins for SoftwareSerial
#define ELM_TX_PIN 10 // Arduino pin to ELM327 Rx
#define ELM_RX_PIN 11 // Arduino pin to ELM327 Tx

SoftwareSerial elmSerial(ELM_RX_PIN, ELM_TX_PIN); // RX, TX

#define ELM_PORT elmSerial

const bool DEBUG = true;
const int TIMEOUT = 2000;
const bool HALT_ON_FAIL = false;

ELM327 myELM327;

void setup() {
  Serial.begin(115200); // Start the hardware serial for debugging
  ELM_PORT.begin(115200); // Start the SoftwareSerial for ELM327

  Serial.println("Attempting to connect to ELM327...");

  if (!myELM327.begin(ELM_PORT, DEBUG, TIMEOUT)) {
    Serial.println("Couldn't connect to OBD scanner");

    if (HALT_ON_FAIL) {
      while (1);
    }
  }

  Serial.println("Connected to ELM327");
}

void loop() {
  float coolantTemp = myELM327.engineCoolantTemp();
  float rpm = myELM327.rpm();
  float speed = myELM327.kph();

  if (myELM327.nb_rx_state == ELM_SUCCESS) {
    Serial.print("Temperature: ");
    Serial.println(coolantTemp);
    Serial.print("RPM: ");
    Serial.println(rpm);
    Serial.print("Speed: ");
    Serial.println(speed);
    
    // Send data in a single line for easy parsing
    Serial.print("DATA,");
    Serial.print(coolantTemp);
    Serial.print(",");
    Serial.print(rpm);
    Serial.print(",");
    Serial.println(speed);
  } else if (myELM327.nb_rx_state != ELM_GETTING_MSG) {
    myELM327.printError();
  }

  delay(1000); // Adjust delay as needed
}

i'm somehow still getting the same error :
Attempting to connect to ELM327...
Clearing input serial buffer
Sending the following command/query: AT D
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT Z
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT E0
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT S0
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT AL
Timeout detected with overflow of 1ms
Clearing input serial buffer
Sending the following command/query: AT ST 00
Timeout detected with overflow of 0ms
Clearing input serial buffer
Sending the following command/query: AT SP A0
Timeout detected with overflow of 0ms
Setting protocol via AT TP A%c did not work - trying via AT SP %c
Clearing input serial buffer
Sending the following command/query: AT SP 0
Timeout detected with overflow of 0ms
Setting protocol via AT SP %c did not work
Couldn't connect to OBD scanner
Connected to ELM327
Service: 1
PID: 5
Normal length query detected
Query string: 01051
Clearing input serial buffer
Sending the following command/query: 01051
Timeout detected with overflow of 1000ms
Service: 1
PID: 12
Normal length query detected
Query string: 010C1
Clearing input serial buffer
Sending the following command/query: 010C1
Timeout detected with overflow of 0ms
Service: 1
PID: 12
.
.
.
I'm really confused

jimwhitelaw, 2024-06-19:
A few thoughts:

```C
/ Define the pins for SoftwareSerial
#define ELM_TX_PIN 10 // Arduino pin to ELM327 Rx
#define ELM_RX_PIN 11 // Arduino pin to ELM327 Tx
```
This code doesn't match the comments. It could be correct, but worth checking that you don't have RX & TX swapped.

```C
void loop() {
float coolantTemp = myELM327.engineCoolantTemp();
float rpm = myELM327.rpm();
float speed = myELM327.kph();
```
This is not the correct way to use _loop()_ to query multiple PIDs. See the [multiple pids example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/ESP32_Bluetooth_multiple_pids/ESP32_Bluetooth_multiple_pids.ino) for the correct method.

From the looks of it, you are not getting a valid serial connection to the adapter. I'd start there before moving on to specific queries. Start with this [simple software serial example](https://github.com/PowerBroker2/ELMduino/blob/master/examples/Arduino_software_serial_test/Arduino_software_serial_test.ino) and get your connection working, then move on to running the multiple query example.

closed 2025-02-09 by PowerBroker2